FastLED/FastLED #283: ESP8266: Data transmission gets aborted

Lightning303, 2016-04-07:
Hey,
Im using an ESP8266 to drive some WS2812B leds. However, sometimes when i set a new color not all leds actually change their color. I used this code to test:

`#include <FastLED.h>
CRGB leds[27];

void setup() {
  FastLED.addLeds<WS2812B, 13, GRB>(leds, 27);
}

void loop() {
  FastLED.showColor(CRGB(16, 16, 16));
  delay(1000);
  FastLED.showColor(CRGB(0, 0, 0));
  delay(1000);
}`

Every so often just some leds would change their color and the rest would stay in their old color. I dont have a digitial storage oscilloscope so i cant check what exactly is happening. But i do have an analog one. And I can see the bursts on the data line there. Whenever my problem appears i can clearly see that the burst is shorter. Which lets me to believe that something is happening that interrupts/aborts the bit banging.
I tried feeding the watchdog just before the showColor/show command, as I suspected this to be the problem, but it didnt fix it.
Before checking with the oscilloscope i also tried to level shift the data line to 5v, without luck. 

Im using this module: http://www.ebay.de/itm/201438648636
And this FastLED master from mid march (latest doesnt work): https://github.com/FastLED/FastLED/tree/242988ba49d3bdf1663a930517cf242544f82063

ESP and leds both are powered from an external powersupply (3.3V and 5V of course) and not from the USB to TTL module.

Hope somebody can help.
Thanks.

TheAustrian, 2016-04-07:
Try adding a delay after "FastLED.addLeds<WS2812B, 13, GRB>(leds, 27);" - I used to have a similar issue and that fixed it for me. It seems .addLeds needs some time to add all the LEDs and if you call .showcolor too early it only sends data to those it was able to add in the time it had.

Lightning303, 2016-04-07:
I will try that, but cant imagine that this is the problem. If i watch my leds for a minute, then 95% of the time, or even more, all leds do change their color. Just in those 5% of the cases it doesnt. If it wasnt able to add all 27 leds then i would expect to have this issue 100% within this timeframe.
But as i said, i will try :) Thanks.

TheAustrian, 2016-04-07:
Ok, then I misunderstood the issue - maybe try another pin? IO13 has RXD2 on it, maybe something wants to use it.

Lightning303, 2016-04-07:
Sadly no.
Tried pretty much every pin i could use, and on all i have the same issue.

focalintent, 2016-04-07:
1) when you say "latest doesn't work" - what do you mean?  If you don't give me more detail on a problem, I'm not going to be able to even attempt to fix it for 

2) see https://github.com/FastLED/FastLED/wiki/Interrupt-problems - likely what is happening is some interrupt handler is running for too long, which will cut short the writing out of a string of data.  I thought I had put something in place that would re-attempt writing the frame if it had been cut off by an interrupt.

Lightning303, 2016-04-07:
> 1) when you say "latest doesn't work" - what do you mean? If you don't give me more detail on a problem, I'm not going to be able to even attempt to fix it for

I started working on my project mid march, thats when in pulled the master. Today i was trying to fix this issue again. I checked the latest commits to master and saw you worked on esp8266 since then (https://github.com/FastLED/FastLED/commit/dcb973e42bb6bed5a9c120718884e0259a0302bb). So i pulled again and tryied with the latest master. However my leds stay off with the it, nothing is happening. I figgured your latest changes are still work in progress and therefore its currently not working. So i reverted back to the mid march master i linked above.

focalintent, 2016-04-08:
Ok - I think i've fixed the issue with pins.  I've also changed the pin ordering to match what's on things like the NodeMCU/Weemos boards - of course, this means that things won't line up right if you're using a huzzah board - for now you'll have to add `#define FASTLED_ESP8266_RAW_PIN_ORDER` to make it happy.  I've filed a bug with the esp8266-arduino folks to make it possible to differentiate between which esp8266 board is being built.

Also - I put something in place that will help with fixing this - if the controller code detects that an interrupt ran too long while writing out ws2812 data, it will immediately try re-writing the frame.  Of course - if there are long interrupts that are occurring on regular intervals, there will be a maximum number of leds that you'll be able to write on a single pin reliably (this is mostly just simple math - let's say you have a long interrupt that fires once every 10ms.  Then you can't have a strand with more than 333 leds on it (10ms / 30µs) because it will _always_ get interrupted.

closed 2016-04-08 by focalintent

Lightning303, 2016-04-08:
Hey,
with https://github.com/FastLED/FastLED/commit/0cdf707c1e86743121f93b449344e15d314a1342 my ESP resets all the time. I narrowed it down to the addLeds command. When its called in the setup the watchdog resets the chip.

```
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld
```

focalintent, 2016-04-08:
What pin are you using when it does that?

Lightning303, 2016-04-08:
I was using pin 13. `FastLED.addLeds<WS2812B, 13, GRB>(leds, 27);`
With and without `#define FASTLED_ESP8266_RAW_PIN_ORDER`. Didnt make a difference. However I now changed to pin 4, with the raw pin order, which corresponds to IO2 on my board. And the reset does not occure anymore :).

Sadly, the issue i originally came here for still exist =(. So sometimes not all leds get updated.

TheAustrian, 2016-04-08:
What if you try using CRGBSet:
set it up with
CRGBSet ledSet(leds, NUM_LEDS);

and instead of 
leds.showcolor(16, 16, 16) ;
use
ledSet = CRGB(16, 16, 16);
LEDS.show();

It's kind of quicker in my sketches that way, so maybe it will help.

Lightning303, 2016-04-08:
Just tried it. No luck. Issue remains =(.

focalintent, 2016-04-08:
For the interrupt thing, try adding:

```
#define FASTLED_ALLOW_INTERRUPTS 0
```

This will disable the ability for interrupts to interrupt writing out the frame.  (Be aware that this might cause some other issues, some network libraries don't like having interrupts blocked for too long).  

This is pretty much going to be a permanent problem with ws28xx style LEDs - disable interrupts for too long, and some other libraries start having problems, allow interrupts, and you run the risk of frames being cut short.

Lightning303, 2016-04-08:
Thank you, this "fixes" my issue indeed.

TheAustrian, 2016-04-14:
I know this issue was closed but I accidentally stumbled on https://github.com/nodemcu/nodemcu-firmware/pull/951 today - maybe using UART on the ESP8266 might be something to consider for FastLed, too?

commit 0cdf707c (2016-04-08): Fix pin mode setting - also reset esp8266 pin definitions to match the Dxx definitions on the nodemcu variant boards.  Also fix #283 - if writing a frame gets cut short because of an interrupt, immediately attempt re-writing the frame to increase the odds of the entire frame getting written out.