FastLED/FastLED #285: APA102 not working with FastLED but APA102 library from Pololu

r100gs, 2016-04-08:
Hey guyes! I bought some APA102 LED stripes 
http://de.aliexpress.com/item/1m-5m-APA102-Smart-led-pixel-strip-24-36-60-96-144-leds-pixels-m-Black/32353920529.html?detailNewVersion=2

Its working fine with the APA102 library from pololu but not with FastLED.
Can you give me some hints why?

Best regards,
Stefan

focalintent, 2016-04-08:
To give you some hints I'd need to know things like:

* what version of the library are you using?
* what version of arduino are you using?
* what hardware are you building for? 
* can you upload the code that you're trying to use to gist.github.com (or make sure to put it in a code block here properly) so that I can see what code you're trying to use with it.

(and when you say not working - what do you mean?  nothing lights up?  the wrong thing lights up?  it flickers?  etc...)

r100gs, 2016-04-08:
Hello and thx for the fast replay.

Im using an arduino Uno R3 clone.  It is written  M.J. arduino on it.
FastLed is 3.1
At the moment I tried the demoreel.ino (and other examples, all the same) with pin 13 as clock and 11 as Data.  But only some of the led Light start light. I tried different pins But all the same  
So I starten searching and found plolulo apa102 library
https://github.com/pololu/apa102-arduino
I gave the rainbow.ino a try and all LEDs are working with pin 11 and 12 (standard)

In the end I try to build a soundcontrolled light for my son.

Best wishes,
Stefan

focalintent, 2016-04-08:
Can you post the code as you're trying to compile it so I can see what's going on?  I regularly drive apa102's off of UNOs with no problems.

r100gs, 2016-04-09:
```
#include "FastLED.h"

FASTLED_USING_NAMESPACE

// FastLED "100-lines-of-code" demo reel, showing just a few 
// of the kinds of animation patterns you can quickly and easily 
// compose using FastLED.  
//
// This example also shows one easy way to define multiple 
// animations patterns and have them automatically rotate.
//
// -Mark Kriegsman, December 2014

#if FASTLED_VERSION < 3001000
#error "Requires FastLED 3.1 or later; check github for latest code."
#endif

#define DATA_PIN    11
//#define CLK_PIN   12
#define LED_TYPE    APA102
#define COLOR_ORDER GRB
#define NUM_LEDS    360
CRGB leds[NUM_LEDS];

#define BRIGHTNESS          96
#define FRAMES_PER_SECOND  120

void setup() {
  delay(3000); // 3 second delay for recovery
  
  // tell FastLED about the LED strip configuration
  FastLED.addLeds<LED_TYPE,DATA_PIN,COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);
  //FastLED.addLeds<LED_TYPE,DATA_PIN,CLK_PIN,COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);

  // set master brightness control
  FastLED.setBrightness(BRIGHTNESS);
}

// List of patterns to cycle through.  Each is defined as a separate function below.
typedef void (*SimplePatternList[])();
SimplePatternList gPatterns = { rainbow, rainbowWithGlitter, confetti, sinelon, juggle, bpm };

uint8_t gCurrentPatternNumber = 0; // Index number of which pattern is current
uint8_t gHue = 0; // rotating "base color" used by many of the patterns
  
void loop()
{
  // Call the current pattern function once, updating the 'leds' array
  gPatterns[gCurrentPatternNumber]();

  // send the 'leds' array out to the actual LED strip
  FastLED.show();  
  // insert a delay to keep the framerate modest
  FastLED.delay(1000/FRAMES_PER_SECOND); 

  // do some periodic updates
  EVERY_N_MILLISECONDS( 20 ) { gHue++; } // slowly cycle the "base color" through the rainbow
  EVERY_N_SECONDS( 10 ) { nextPattern(); } // change patterns periodically
}

#define ARRAY_SIZE(A) (sizeof(A) / sizeof((A)[0]))

void nextPattern()
{
  // add one to the current pattern number, and wrap around at the end
  gCurrentPatternNumber = (gCurrentPatternNumber + 1) % ARRAY_SIZE( gPatterns);
}

void rainbow() 
{
  // FastLED's built-in rainbow generator
  fill_rainbow( leds, NUM_LEDS, gHue, 7);
}

void rainbowWithGlitter() 
{
  // built-in FastLED rainbow, plus some random sparkly glitter
  rainbow();
  addGlitter(80);
}

void addGlitter( fract8 chanceOfGlitter) 
{
  if( random8() < chanceOfGlitter) {
    leds[ random16(NUM_LEDS) ] += CRGB::White;
  }
}

void confetti() 
{
  // random colored speckles that blink in and fade smoothly
  fadeToBlackBy( leds, NUM_LEDS, 10);
  int pos = random16(NUM_LEDS);
  leds[pos] += CHSV( gHue + random8(64), 200, 255);
}

void sinelon()
{
  // a colored dot sweeping back and forth, with fading trails
  fadeToBlackBy( leds, NUM_LEDS, 20);
  int pos = beatsin16(13,0,NUM_LEDS);
  leds[pos] += CHSV( gHue, 255, 192);
}

void bpm()
{
  // colored stripes pulsing at a defined Beats-Per-Minute (BPM)
  uint8_t BeatsPerMinute = 62;
  CRGBPalette16 palette = PartyColors_p;
  uint8_t beat = beatsin8( BeatsPerMinute, 64, 255);
  for( int i = 0; i < NUM_LEDS; i++) { //9948
    leds[i] = ColorFromPalette(palette, gHue+(i*2), beat-gHue+(i*10));
  }
}

void juggle() {
  // eight colored dots, weaving in and out of sync with each other
  fadeToBlackBy( leds, NUM_LEDS, 20);
  byte dothue = 0;
  for( int i = 0; i < 8; i++) {
    leds[beatsin16(i+7,0,NUM_LEDS)] |= CHSV(dothue, 200, 255);
    dothue += 32;
  }
}

```

This is what I get after upload:

In file included from /home/stefan/Schreibtisch/DemoReel100/DemoReel100.ino:1:0:
/home/stefan/Arduino/libraries/FastLED/FastLED.h:12:2: warning: #warning FastLED version 3.001.000 (Not really a warning, just telling you here.) [-Wcpp]
 #warning FastLED version 3.001.000  (Not really a warning, just telling you here.)
  ^

Der Sketch verwendet 6.980 Bytes (21%) des Programmspeicherplatzes. Das Maximum sind 32.256 Bytes.
Globale Variablen verwenden 1.239 Bytes (60%) des dynamischen Speichers, 809 Bytes für lokale Variablen verbleiben. Das Maximum sind 2.048 Bytes.

I tried with port 13 as clk as well. Same result its not working

I installed FastLed 3.1 via the library manger.

I hope you can help me!

Best regards,
Stefan

marmilicious, 2016-04-09:
Hi r100gs.  For APA102 you need to use both data and clock but I see you have clock commented out, and it also doesn't look like you're using the addLeds line which uses data and clock.  Can you try this most basic example:
`
#include "FastLED.h"
#define DATA_PIN  11
#define CLK_PIN  13
#define LED_TYPE  APA102
#define COLOR_ORDER  GRB
#define NUM_LEDS  360
CRGB leds[NUM_LEDS];

void setup() {
  delay(3000); // 3 second delay for recovery
  FastLED.addLeds<LED_TYPE,DATA_PIN,CLK_PIN,COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);
  FastLED.setBrightness(96);
}

void loop() { 
  // Fill leds with rainbow
  fill_rainbow(leds, NUM_LEDS, millis());
  FastLED.show();
  delay(2000);

  // Turn leds off
  fill_solid(leds, NUM_LEDS, CRGB::Black);  // Set all leds black
  FastLED.show();
  delay(1000);
}
`

With 360 leds you need to be providing them proper power.  There might be too much voltage drop part way down the strip if you're not feeding it 5V every so often.

I would suggest using the FastLED G+ group when you run into trouble as you'll have a lot more people see your post to help out there.
https://plus.google.com/u/0/communities/109127054924227823508

r100gs, 2016-04-10:
So i tried again with the commented out clock pin, but still the same its not working for me

r100gs, 2016-04-25:
So, finally its working! I was too silly to read everything correct

closed 2016-04-25 by r100gs