FastLED/FastLED #286: Most recent commit seems to have broken something (via PlatformIO)

julianh2o, 2016-04-10:
I was prompted by PlatformIO to update my FastLED library.
I think it updated to this version: https://libraries.io/platformio/FastLED/c257d754e3

Anyway, my project completely stopped working, the strips stopped updating entirely. I'm using PlatformIO and uploading to an ESP-12. I'm using a custom controller to drive my LEDs.

I'm a little concerned that PlatformIO prompted me to install an unreleased new version.. I wonder if that's intentional.

focalintent, 2016-04-11:
I started putting in mappings for the pins to line up with what's on the boards for NodeMCE/WeMos D1 and others - however there's a couple of boards that there's no way for me to distinguish between yet ( see https://github.com/esp8266/Arduino/issues/1878 - hopefully they'll fix it in 0.2.2 of the esp8266 libraries and I'll support it as soon as they do).

In the meantime, if you put ```#define FASTLED_ESP8266_RAW_PIN_ORDER``` before the ```#include <FastLED.h>``` that will restore the old pin mappings for the time being (until I can determine them at compile time from the build variables).

julianh2o, 2016-04-11:
Awesome, thanks!

TheAustrian, 2016-04-11:
I use a Wemos D1 Mini and without `#define FASTLED_ESP8266_RAW_PIN_ORDER` I get no output, with it I get the following error messages:
`In file included from /Users/*/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastled_esp8266.h:5:0,
                 from /Users/*/Documents/Arduino/libraries/FastLED/platforms.h:24,
                 from /Users/*/Documents/Arduino/libraries/FastLED/FastLED.h:44,
                 from /Users/*/Documents/Arduino/ESP-Fado-Lamp/ESP-Fado-Lamp.ino:10:
/Users/*/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:41:57: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(0,0); _DEFPIN_ESP8266(1,1); _DEFPIN_8266(2,2); _DEFPIN_8266(3,3);
                                                         ^
/Users/*/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:41:76: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(0,0); _DEFPIN_ESP8266(1,1); _DEFPIN_8266(2,2); _DEFPIN_8266(3,3);
                                                                            ^
/Users/*/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:42:57: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(4,4); _DEFPIN_ESP8266(5,5); _DEFPIN_8266(6,6); _DEFPIN_8266(7,7);
                                                         ^
/Users/*/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:42:76: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(4,4); _DEFPIN_ESP8266(5,5); _DEFPIN_8266(6,6); _DEFPIN_8266(7,7);
                                                                            ^
/Users/*/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:43:57: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(8,8); _DEFPIN_ESP8266(9,9); _DEFPIN_8266(10,10); _DEFPIN_8266(11,11);
                                                         ^
/Users/*/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:43:78: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(8,8); _DEFPIN_ESP8266(9,9); _DEFPIN_8266(10,10); _DEFPIN_8266(11,11);
                                                                              ^
/Users/*/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:44:61: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(12,12); _DEFPIN_ESP8266(13,13); _DEFPIN_8266(14,14); _DEFPIN_8266(15,15);
                                                             ^
/Users/*/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:44:82: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(12,12); _DEFPIN_ESP8266(13,13); _DEFPIN_8266(14,14); _DEFPIN_8266(15,15);
                                                                                  ^
In file included from /Users/*/Documents/Arduino/libraries/FastLED/FastLED.h:40:0,
                 from /Users/*/Documents/Arduino/ESP-Fado-Lamp/ESP-Fado-Lamp.ino:10:
/Users/*/Documents/Arduino/libraries/FastLED/fastpin.h: In instantiation of 'class FastPin<15u>':
/Users/*/Documents/Arduino/libraries/FastLED/platforms/esp/8266/clockless_esp8266.h:16:49:   required from 'class ClocklessController<15, 20, 50, 30, (EOrder)66u, 0, false, 5>'
/Users/*/Documents/Arduino/libraries/FastLED/chipsets.h:401:7:   required from 'class WS2812Controller800Khz<15u, (EOrder)66u>'
/Users/*/Documents/Arduino/libraries/FastLED/FastLED.h:92:52:   required from 'class WS2812B<15u, (EOrder)66u>'
/Users/*/Documents/Arduino/libraries/FastLED/FastLED.h:273:39:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2812B; unsigned char DATA_PIN = 15u; EOrder RGB_ORDER = (EOrder)66u]'
/Users/*/Documents/Arduino/ESP-Fado-Lamp/ESP-Fado-Lamp.ino:48:51:   required from here
/Users/*/Documents/Arduino/libraries/FastLED/fastpin.h:206:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
  ^
exit status 1
Error compiling for board WeMos D1 R2 & mini.
`

TheAustrian, 2016-04-11:
Now I tried the Dxx number and it does work with that (without the #define) but only writing it as "8" not as "D8". But no-one uses them without D as far as I can tell. They use either the GPIO number or the name with D in front. Mapping the numbers like they are now might break a lot of stuff.

focalintent, 2016-04-11:
The numbers as they are now is how the pin/port mapping stuff works on pretty much every other arduino platform I work with (in part because on avr and arm environments, a digital pin is both a port identifier _and_ a number).  If you want to use the GPIO numbers instead of the Dx numbers use FASTLED_ESP8266_RAW_PIN_ORDER - I just pushed the compile error fix (and starting with, hopefully, the 2.2.0 release of esp8266 arduino - I will be able to do the correct mappings for all the platforms, once they give me a way to differentiate between them)

(Also - not every build variant in the esp8266-arduino code supplies values for the Dx pins - when i get another break from work i'll chase down a ticket for that)

TheAustrian, 2016-04-11:
Thanks a lot! The thing is that in all the tutorials or libraries I know and used so far the pin numbers used were the GPIO ones. That might be a general thing for the ESP8266/Arduino project or just for the boards I use, but should be cleared up I think.

focalintent, 2016-04-11:
Yeah - I think people were using the gpio pins because there was no one doing any mappings from board pins to gpio pins.  Hopefully that can clear up soon.  

When 2.2.x comes out i'll see if I can figure out a way to work with both the raw pin numbers and the Dx numbers.  

(GPIO to board pin mapping is a long running gripe of mine with pretty much every board maker out there - because they can be sloppy/lazy about it and just do their own mappings for it :)

TheAustrian, 2016-04-11:
Example that uses the "D": https://github.com/esp8266/Arduino/blob/5dd6acc4c20e2dfa199ae3f21b4e2bd9a8e66bb8/variants/d1_mini/pins_arduino.h

focalintent, 2016-04-11:
That's not really an example, that's a variant definition :) 

There isn't a single example under the esp8266 platform that uses the Dx convention for accessing digital pins.  For example, in https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/AdvancedWebServer/AdvancedWebServer.ino all the digitalWrite calls use a variable named led which is defined as:

```
const int led = 13;
```

and pretty much all the examples are like that.

For most of the other arduino platforms, there is a mapping between the integer pin values to the correct underlying port/pin representation (even the ESP8266 has multiple ports - technically pin 16 is on a separate gpio port from the other 15).  In fact, I don't think the other arduino platforms use the Dx convention for pins at all.  They just use the raw pin numbers (except for analog pins, which are marked as A0/A1/etc...).

Given that the majority of FastLED's users are coming from the arduino world, I do try to work/fit with that, especially now that there are esp8266 boards running around where the pin designation on the silkscreen on the board does _not_ line up to the underlying GPIO pins.

closed 2016-04-11 by focalintent