FastLED/FastLED #287: ESP8266 & P9813 not working together

SunboX, 2016-04-11:
I'm using this Module:
http://www.amazon.de/dp/B00P2KDTEC 

together with a P9813 LED strip with 90 LED's. The strip is working fine with a Arduino Uno, but it doesn't work if I flash it on the ESP8266 Module.

If I connect a simple LED to GPIO0 or GPIO2, it working fine (blinking).

So I think the issue is related to the FastLED library in combination with the ESP8266 and the P9813 LED strip.

focalintent, 2016-04-11:
When connecting a simple led, how are you changing it, with digitalWrite or with FastPin<0>::toggle();  right now the default pin numbering definitions in FastLED are for the PIN number to gpio mapping used on the nodemcu boards and clones.  It's going to require an update to the arduino esp8266 code to be able to reliably differentiate when building for different boards, in the meantime you can put the line

```
#define FASTLED_ESP8266_RAW_PIN_ORDER
```

Before you `#include "FastLED.h"` (first line of the file is best). 

See if that works.

SunboX, 2016-04-11:
Hi, and thanks for your quick response! Jep, I used digitalWrite(0) and digitalWrite(2) to blink the normal LED's.

If I add this line above:

    #define FASTLED_ESP8266_RAW_PIN_ORDER
    #include <FastLED.h>

I'm getting:

```txt
In file included from /Users/xxx/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastled_esp8266.h:5:0,
                 from /Users/xxx/Documents/Arduino/libraries/FastLED/platforms.h:24,
                 from /Users/xxx/Documents/Arduino/libraries/FastLED/FastLED.h:44,
                 from /Volumes/xxx/xxx/xxx/xxx.ino:7:
/Users/xxx/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:41:57: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(0,0); _DEFPIN_ESP8266(1,1); _DEFPIN_8266(2,2); _DEFPIN_8266(3,3);
                                                         ^
/Users/xxx/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:41:76: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(0,0); _DEFPIN_ESP8266(1,1); _DEFPIN_8266(2,2); _DEFPIN_8266(3,3);
                                                                            ^
/Users/xxx/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:42:57: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(4,4); _DEFPIN_ESP8266(5,5); _DEFPIN_8266(6,6); _DEFPIN_8266(7,7);
                                                         ^
/Users/xxx/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:42:76: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(4,4); _DEFPIN_ESP8266(5,5); _DEFPIN_8266(6,6); _DEFPIN_8266(7,7);
                                                                            ^
/Users/xxx/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:43:57: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(8,8); _DEFPIN_ESP8266(9,9); _DEFPIN_8266(10,10); _DEFPIN_8266(11,11);
                                                         ^
/Users/xxx/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:43:78: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(8,8); _DEFPIN_ESP8266(9,9); _DEFPIN_8266(10,10); _DEFPIN_8266(11,11);
                                                                              ^
/Users/xxx/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:44:61: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(12,12); _DEFPIN_ESP8266(13,13); _DEFPIN_8266(14,14); _DEFPIN_8266(15,15);
                                                             ^
/Users/xxx/Documents/Arduino/libraries/FastLED/platforms/esp/8266/fastpin_esp8266.h:44:82: error: expected constructor, destructor, or type conversion before '(' token
 _DEFPIN_ESP8266(12,12); _DEFPIN_ESP8266(13,13); _DEFPIN_8266(14,14); _DEFPIN_8266(15,15);
                                                                                  ^
In file included from /Users/xxx/Documents/Arduino/libraries/FastLED/FastLED.h:40:0,
                 from /Volumes/xxx/xxx/xxx/xxx.ino:7:
/Users/xxx/Documents/Arduino/libraries/FastLED/fastpin.h: In instantiation of 'class FastPin<2u>':
/Users/xxx/Documents/Arduino/libraries/FastLED/fastspi_bitbang.h:22:49:   required from 'class AVRSoftwareSPIOutput<2u, 0u, 6u>'
/Users/xxx/Documents/Arduino/libraries/FastLED/fastspi.h:30:7:   required from 'class SPIOutput<2u, 0u, 6u>'
/Users/xxx/Documents/Arduino/libraries/FastLED/chipsets.h:94:6:   required from 'class LPD8806Controller<2u, 0u, (EOrder)10u, 6u>'
/Users/xxx/Documents/Arduino/libraries/FastLED/FastLED.h:226:77:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = (ESPIChipsets)4u; unsigned char DATA_PIN = 2u; unsigned char CLOCK_PIN = 0u; EOrder RGB_ORDER = (EOrder)10u]'
/Volumes/xxx/xxx/xxx/xxx.ino:124:68:   required from here
/Users/xxx/Documents/Arduino/libraries/FastLED/fastpin.h:206:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
  ^
exit status 1
Fehler beim Kompilieren.
```

SunboX, 2016-04-11:
Is there an `ESP` missing in `_DEFPIN_8266`?

focalintent, 2016-04-11:
Yes - repull the code - I fixed that one earlier today

SunboX, 2016-04-15:
Works great! Thank you very much!

closed 2016-04-15 by SunboX

varinderd, 2018-02-13:
Hey guys,

I know this is an old thread, but I can't find this info anywhere and this page comes up in google first.

How are you connecting the ESP-01 to the P9813. ESP-01 has GPIO0 and GPIO2 and my question is which one is going to the P9813 CIN and DIN pins. I could do this by trial and error but I don't wana fry any of these boards.

Thanks,