FastLED/FastLED #288: How many leds?

DarkHunterer, 2016-04-12:
Forgive me if you see this question for the 100th time but I didn't find it in FAQ, neither in wiki.

So the question is, how many ws2812b leds can I control? I mean what is really the maximum length?
They are 800kbs, right? What could be the bottleneck, the reduction of signal amplitude? The voltage drop on led strip?
 Did you try when would I have to regenerate the signal?
How many meters can be powered with single source?

Thank you for answering.

electrokean, 2016-04-12:
This doesn't relate to the FastLED library, so it would be better discusedd on the FastLED Google+ group - http://fastled.io/+

That said, your biggest issue is going to be voltage drop on the strip, and that will depend on how bright you're running the LEDs and how many are on simultaneously.  You can always inject power along the strip in segments with heavy cable from a large PSU, or several "point of load" power supplies.  Just make sure to keep a common ground connection.
With the WS2812B and similar LEDs, the data signal itself is regenerated by each LED chip, so that isn't generally an issue.
With enough LEDs, your next bottleneck will be either memory capacity of your microcontroller, or the time to send each data frame (depending on your desired update rate).

Daniel-dk, 2016-04-12:
TLDR: 
it depends on a few things, generally they are : 
1. : The amount of RAM that your uC has
2. : the framerate you want to achieve
3. : the power supply capacity ( and cable capacity )

- Uno/Leonardo/micro/nano : I don't go above 512 LEDs ( 1.5kByte of SRAM reserved for LEDs ),
- Mega : Ive done 1000 LEDs but not with many other things happening. ( 3 kByte of SRAM reserved for LEDs )
-  DUE/ Zero : quite a lot of LEDs ( multiple controller examples ) never more than 1000 per "FASTLED controller" to keep framerate above 30.

my reasons:

**SRAM usage**
1.  Each LED ( or CRGB array entry ) uses 3 bytes of your SRAM.

The amount of SRAM available for LEDs depends on what else the controller is doing that also requires SRAM buffers or "chunks" such as :

- String manipulation
- Networking ( Ethernet / Wifi with OSC, OPC or MQTT protocols )
- Serial data ( DMX ), 
- Visualisations/ animations ( Fire 2012 , meteors, fireworks )
- Signal processing (audio spectrum FFT) 
- Double buffering pixel data ( if you use it as an output for a video - like Processing.org sketches)
- Stored color pallets ( interpolation happens in SRAM )
- Arduino buffers ( 64 byte reserved for Serial buffers, I2C buffers, SPI buffers if they are initialised )
- other things stored in SRAM ( Serial.print("some text "); uses SRAM to store the "some text" )

Arduino Leonardo, Micro, Uno, Nano only has 2.5 kByte of SRAM, 
Mega has 4 kByte SRAM.
Due, Teensy has oodles of SRAM, , DUE has 96kByte
Zero / MKR1000 has 32kByte SRAM

**Required framerate** - mostly to keep the animations or video frames looking smooth

- 1024 WS2812B LEDs spoken to at 800 kHz will give you 30 FPS
any more will lower your framerate
- on DUE I have done about 3000 LEDs with 3 "controllers" , each one driving less than 1024 so the framerate stayed above 30 fps

**Power supply rating ( and cabling used )**

- There is a voltage drop over the LED strips so we generally inject power every 2 strips ( 10 m ) with high capacity power cable. and with multiple supplies in parallel.

- WS series ( 2811,2812,2812B ) LEDs have output drivers that "refresh" the signal when it is forwarded to the next LED in line, you could go a few m in-between LEDs ( Ive done up to 6m which is pushing it a bit as any other cables near the signal wire corrupted the signal )

focalintent, 2016-04-12:
How many leds is a complex question - it depends on the amount of ram on the hardware that you have (e.g. about 600 on a given arduino, more like 10-15,000 on the teensy).  It also depends on what kind of frame rate that you want.  It takes 30µs to write out a single led's worth of data, so the absolute maximum number of rgb led updates per second that you can do is about 33,000 (or, if you're doing 8-way parallel output 260,000ish).  Of course, that's assuming spending 100% of your time writing led data.

Re: maximum length - there really isn't one.  Because each WS2811/WS2812 re-generates the signal, in theory you can have as long of a chain as you want (as long as you inject power regularly, see below - or every 100 or so).  Voltage drop is the reason why you need to inject power at various points - but because of the signal regeneration happening at each chip, you don't have to worry about voltage drop/signal degradation as a function of length.

DarkHunterer, 2016-04-12:
Thank you for complex your answers!

I'm going to study my needs now but I believe that what I read here is enough for my question.

Thanks again.

closed 2016-05-17 by focalintent

clemvixen, 2017-02-26:
Hello,

Thanks for informations

But I don't understand, How to calculate the number of LED WS2811 that can be put on a microcontroller ?

Can you help me for this calculation, with the size of RAM, the speed of the µC ?

And for my application, I need to receiv data with the 1.31 protocol and I find this takes place in RAM ?

thanks and soory for my level in english ..

alvin666666, 2017-04-20:
hello,we are the led strip custom factory ,for your ws2812b connect problem,it is easy.
We can supply sp105e t1000s k1000c controller for you. You also can send email to me,i can help you conenct the led strip.
Email:Alvin@witop-tech.com

Now we have a new product 12v addressable/individually breakpoint continue rgb led strip,you can learn more by the link.
https://www.alibaba.com/product-detail/High-brightness-addressable-breakpoint-continue-full_60635610880.html?spm=a2747.manage.list.11.V2sqR3

ajensen2, 2017-10-05:
Hello All,

First off, I am a complete newbie so thanks for any insight and help.

I am putting together a Christmas Light Display using Arduino's. Part 1 is 16 Solid State Relays being controlled with Vixen through an Arduino Mega. Part 2 is strips of Mukungit WS2811 Pixels. I just got the LED Strips last night and I copy and pasted some code for PixelFlex to get them to interact with Vixen and it mostly worked. I know now (after reading this thread) that one issue is that I was using an Uno because it's what I had laying around and I could tell it wasn't as responsive as I would like. Another problem is that some of the Pixels in the middle of the strip stayed lit and were not controlled with the rest of the pixels. I imagine that this is a symptom of me copying and pasting code that was not the right thing. When ran a simple chase code all of the Pixels responded appropriately so I know that the strand of pixels is fine.

Now, for the big question... Each strand has 300 Pixels and I would like to run 8 strands. Furthermore, I would like to run them all independent from each other. How many of what controllers would you recommend to control these? From this thread I am getting the feeling that a DUE might be the right answer for me but will one do it?  I'm good with power, I'm just lost when it comes to coding and the processing speed of Arduino's. I bought the book "Programming Arduino" and am working my way through that.

yes, I know I need to start small.... but that's not my style. That's why I am starting to work on my Christmas Display at the end of September. Thanks in advance!

AndyBadger, 2017-12-05:
ajensen2 - check out the controllers from SanDevices.  I started my Christmas display using Arduinos and have transitioned to the SanDevices E6804 controller.  It is awesome and has good fusing on each output for protection.  Also check out xLights software too.  I am right now switching from Vixen to xLights.  Good luck.

srinivasyadav18, 2019-02-18:
At what rate can i update 60 leds...both color and brightness...

focalintent, 2019-02-18:
@srinivasyadav18 it depends on what type of leds you are using - there’s a bunch of math examples further up in this issue.

srinivasyadav18, 2019-02-18:
Im using WS2812B  ....so at what rate I can change both colour and brightness of all leds...data  from Arduino
uno..

focalintent, 2019-02-18:
Ws2812b’s take 30us to write out data - so 60 leds would take 1.8ms to write out or a maximum of just over 550 frames per second.  However WS2812 LEDs have problems if updated more than 400 times per second, so the library will limit your update rate to 400.

jubersoudagar, 2019-07-03:
Hi,, 
I want to drive 1000 led pixels with arduino.... 
Pllzzzz help...

marmilicious, 2019-07-03:
> I want to drive 1000 led pixels with arduino....
> Pllzzzz help...
@jubersoudagar  You need to provide more information with this sort of question and show that you've made some effort to start to look into this.  Please ask this question at the FastLED Reddit group and provide way more info about what you want to do, including what type of pixels you plan to use.
FastLED.io/r

focalintent, 2019-07-04:
Also, because I've been meaning to for years and haven't yet, I've added this to the frequently asked questions page - https://github.com/FastLED/FastLED/wiki/Frequently-Asked-Questions#7-how-many-leds-can-i-drive

(short version though - if you are using an Arduino Uno or an Arduino Leonardo you can't do 1000 leds - because that would require 3000 bytes of memory for the RGB data for 1000 leds (3 bytes of data for each led) and they only have 2000 bytes and 2500 bytes of ram respectively)

KOMILJONOV, 2023-02-03:
Can we use leds without allocating to SRAM?

I think library uses standart arduino functions.

If library uses startdart functions we can use leds without allocationg to memory?

Can someone explain how this library works?

zackees, 2023-02-03:
This library requires that you allocate a buffer of CRGB which your application writes to then you call FastLED.show() to display them.

> Can we use leds without allocating to SRAM?

Not that I know of. Other libraries may allow you to stream in the LEDs. If you are on ESP32 you may be able to put the framebuffer into the PSRAM portion of your app.

KOMILJONOV, 2023-02-05:
What about neopixel library?

KOMILJONOV, 2023-02-05:
> This library requires that you allocate a buffer of CRGB which your application writes to then you call FastLED.show() to display them.
>
>
> Not that I know of. Other libraries may allow you to stream [...]

What about neopixel library?

zackees, 2023-02-06:
The neopixel library does use a framebuffer as well.

Note that the clockless LED's pretty much need to use a framebuffer because they do not handle streaming very well. The reason is because of the tight timing requirements to bitbang the leds out to the lights. If you don't feed new LED data fast enough the LED's will assume the pause is a latch signal and they'll all lock onto their current colors.

If you want to stream a Neopixel it is possible, but you'll need to run your own buffer of at least a couple of pixels and bitbang the leds using an ISR.

If you want to stream out the leds then use one of the 4 pin clocked leds like the APA102 (and the SK9822 which use the same protocol). You can totally stop the stream and resume it whenever you want. I've done this in fact to run 1800 LED's using an arduino UNO. It used an internal frame buffer of 88 pixels but then those got scaled up to 1800 using this streaming technique.