FastLED/FastLED #290: LPD1886 problems

hemske, 2016-04-19:
Guys,

We have some problems with the LPD1886 LED.
We can't find the LED in the examples, so we tried it with a statement that looks good if we read the rest of the library, but it's not working.

FastLED.addLeds<LPD1886, DATA_PIN, RGB>(leds, NUM_LEDS);

So we added the 'DATA_PIN' in the command. 
Is there someone who ever worked with this chipset?

Thank you in advance!

focalintent, 2016-04-19:
The 1886 isn't fully supported, it has some problems that I haven't finished dealing with.

hemske, 2016-04-19:
Do you know the problem, maybe we can solve it together?
Or does some know a library that this led support?

hemske, 2016-04-25:
I am still trying to fix the library for this chip, but I have one questions.
It's working quite well in the 12 bit mode, but my pcb's are made for the 8 bit mode. 
I saw that you use the variable XTRA0 for that, but if I change that variable I can't see any difference with my logic analyzer. 
Even when I give it a value of 1, 2 or 3 (the values in the switch), I can't see a difference in the protocol.

Is there another way to change this to 8 bit?

focalintent, 2016-04-25:
Ah - great to hear that it's working for 12-bit mode....  I just checked in a change that adds `LPD1886_8BIT` as an led type for the LPD1886 in 8-bit mode.  Let me know if that helps?

hemske, 2016-04-28:
PERFECT! Thank you a lot;) 
It's working like every other LED, so I think you can add this one to your supported list.

hemske, 2016-06-16:
The code worked perfect for Arduino Mega, but since we are using a DUE we have some problems.
Do you have any idea what could be the problem?

focalintent, 2016-06-16:
Are you having problems with the 8-bit or the 12-bit version of the LPD1886?  

Also - you might need a level shifter since the DUE's outputs are 3.3v and I don't know if the LPD1886 is sensitive about getting 5v signals (some chips are more sensitive about this than others).

hemske, 2016-06-17:
Problems with 8 bit version. I am using level converters.
IT looks like he is still running on 12 bit on due, in stead of 8 bit mode.
Is that Possible?

closed 2016-06-17 by focalintent

hemske, 2016-06-21:
This didn't fix the problem. 
If I upload exactly the same code to the Mega & Due for blinking 1 LED, I see 24 bits on my digital analyzer from the Mega and 36 bits from the DUE.
So maybe there is another point where something is going wrong?

Thank you in advance!

focalintent, 2016-06-21:
If you're seeing 36 bits from the due then you haven't grabbed the latest checkin of FastLED master@head - make sure that you don't have an older/second copy of the library lying around somewhere.

commit af6e00cb (2016-06-17): Fix #290 for higher MHZ platforms