FastLED/FastLED #291: support sk9822

vixns, 2016-04-20:
Is sk9822 supported ?

vendors claim they are compatible with APA102, but according http://www.szledcolor.com/newsshow.asp?sid=1&id=199, 

"So what is the difference ? you know the first signal is different , normally the first signal ,sk9822 is a base Synchronization signal, and APA102 don't"

http://www.szledcolor.com/download/SK9822%20LED.pdf

focalintent, 2016-04-21:
The difference is that the SK9822 latches a frame differently than the APA102 - I'm about to check in an SK9822 definition that does the Right Thing™ with SK9822 leds.

closed 2016-04-21 by focalintent

00fractal, 2016-07-26:
Greetings! 

Ithink you'll be happy  to  read  that  amazing article I've just found  on the web, here it is  <http://quality.greatmalaysia.com/e4hbdh>

Later, richard

78485111

commit 4624cd90 (2016-04-21): Fix #291 - while SK9822 claims APA102 compataibility it latches the led data in different ways.