FastLED/FastLED #293: Adafruit WICED Feather (M3)

vk7hse, 2016-04-23:
Hi, I've been using FastLED for a few months now and have recently  acquired an Adafruit WICED Feather development board. There's a patch for the adafruit_neopixel library (fixes a typo in CPUID) but I'd prefer to use FastLED on this board.

Unfortunately it's beyond me to hack through the code in an attempt to implement this myself, but I'm happy to test any future development if required.

Board info can be found [here](https://www.adafruit.com/products/3056)

Thanks ...

focalintent, 2016-04-23:
I'll see if i can throw something in - i don't have one of these boards, so i'm going to be limited in how much i can test it

vk7hse, 2016-08-16:
Just a thought, if you happen to have an old Arduino Due that runs with a M3 mcu. Granted they are not identical but they are of the same family. Failing that I'm happy to donate (WICED) one to you if you want?

focalintent, 2016-08-16:
I have a due but this is a wildly different chip.  The due is an Atmel Sam3x8e while this is an STM something or other.  The time in porting to a new chip has nothing to do with m0 vs m3 and everything to do with the manufacturer and hardware family.  (Even in lines - atmel'a different arm platforms are wildly different in how you access the hardware registers).

focalintent, 2016-08-16:
(Also - the due was the second arm platform I added support for after the teensy 3.0 :)

vk7hse, 2016-08-16:
Ahh ok I should have done a bit more research into the properties of each mcu! The idea only occurred to me as an example for the adafruit gps feather wing wouldn't work without drastic modification when I tried one aimed at the Due it worked! 

The offer still stands if you want a donated WICED 😉

vk7hse, 2018-09-22:
Just cleaning up old open issues raised by myself and as this one is still open after several years it's pretty fair to say that nothing will be done (and that's totally ok) so I'll close this one. However should at any stage in the future be a need, then this can always be reopened

closed 2018-09-22 by vk7hse