FastLED/FastLED #296: Error compling....

anthonyzeh, 2016-04-30:
Sorry, kind new to this and having issues compling to a Flora. any help would be great! Following is the error report: Thanks!

Arduino: 1.6.4 (Windows 7), Board: "Flora (TeensyCore), Serial, US English"

C:\Users\home\Documents\Arduino\libraries\FastLED\FastLED.cpp: In member function 'void CFastLED::delay(long unsigned int)':
C:\Users\home\Documents\Arduino\libraries\FastLED\FastLED.cpp:130:9: error: 'yield' was not declared in this scope
   yield();
         ^
Error compiling.

focalintent, 2016-04-30:
Upgrade your version of Arduino to 1.6.5 or later and make sure you've updated your board definitions as well.

anthonyzeh, 2016-04-30:
Thank you, I some how got it to work, though I seen not to know what i am doing. I was having issues with anything uploading and things seemed intermittent. All seems good at the moment. Thanks again.

closed 2016-04-30 by focalintent