FastLED/FastLED #298: APA106 support

orithena, 2016-05-01:
Hey all! :)

I'd like to have support for the APA106 LED type ([datasheet](https://cdn.sparkfun.com/datasheets/Components/LED/COM-12877.pdf)) in FastLED. They're sold sometimes as WS2812B compatible and the protocol is similar, but the timing seems to be a tad different.

I had some success simply by using the WS2812B code, but it turned out to be shaky. Using an Arduino Mega, I could control up to 18 LEDs, after that the LEDs further down the line started to flicker random values (which we pegged to power issues first). Using an Iteaduino Lite, something happened, but not even the first LED lit up properly.

Back home, I experimented with the Iteaduino Lite and 4 LEDs: I dug into chipsets.h and tried to alter the timings of the WS2811Controller800Khz chipset class. The following line worked:

`class WS2811Controller800Khz : public ClocklessController<DATA_PIN, NS(350), NS(450), NS(300), RGB_ORDER> {};`

I experimented a bit with the numbers and came up with the following ranges:
  T1: 315..374
  T2: 438..2000 (i stopped testing there)
  T3: 80..374

The attached diff file [fastled-apa106.txt](https://github.com/FastLED/FastLED/files/244366/fastled-apa106.txt) shows my additions to the library to support the APA106 (the values for the ClocklessController_Trinket are untested guesswork -- can anybody explain how to test these or how to make a better guess?). I'm going to try the patch on our big project (361 LEDs), but that might take some days... (I'm currently too far away ;)

focalintent, 2016-05-01:
(Also your diff is against an older version of FastLED).

Also - according to the datasheet - https://cdn.sparkfun.com/datasheets/Components/LED/COM-12877.pdf - the timings you have up there are a bit too fast, which will cause problems as you get into longer chains of the chips.  

Your timings come out to 1.1µs per bit - where the datasheet indicates a 1.71µs timing per bit).  

If anything, the SK6822 controller is closer to the APA106 timings from the datasheet (https://cdn.sparkfun.com/datasheets/Components/LED/COM-12877.pdf), (375ns hi time for a 0, 1.375µs hi time for a 1).  I'd say grab the most recent version of the library and try SK6822.  If that works i'll add an alias for APA106 to SK6822.

orithena, 2016-05-01:
The current FastLED library does not even compile here, lots of "impossible constraint in ‘asm’" errors and I cannot even scroll to the start of the compilation output in the Arduino IDE... might be due to the fact that I'm using a rather old version of the IDE, but newer versions don't seem to work with the Iteaduino Lite.

focalintent, 2016-05-02:
Yeah - at some point I had to stop trying to support versions of Arduino prior to 1.6.x - as it is, it is hard to keep up with testing all the platform/led combinations.

closed 2016-05-17 by focalintent

Vashiru, 2018-06-17:
Somebody has added support for the Iteaduino Light on Arduino 1.6.x here: https://github.com/udif/ITEADSW_Iteaduino-Lite-HSP. I've tried to run the FastLED library on it, but I run into the same kind of impossible contraint errors. The board has been retired though so not sure if it's still worth adding support.

commit c2e9896f (2016-05-17): Add APA106 as an alias for SK6822, fix #298