FastLED/FastLED #299: ws2801 leds not responding to commands on particle photon

loonylion, 2016-05-02:
Using fastled 3.1.5 on particle photon via the build IDE, with a chain of 25 WS2801 based leds (adafruit 25 led bullet string). I have all leds on displaying white with a hint of blue and slight flickering. Nothing changes no matter what fastled code I upload to the photon. Using code that manually bitbashed bytes down the SPI pin I was able to make it cycle colours providing there was at least 3 different colours active, and there was also a period of the flashing I see with fastled, but only in between cycles of colour. there was also occasionally an odd, uncommanded colour such as orange so I think the timing may have been off.

Below is the code I'm trying to test with:

`#ifndef STM32F2XX
#define STM32F2XX
#endif

#include "application.h"
//#define FASTLED_FORCE_SOFTWARE_SPI
#include "FastLED/FastLED.h"
FASTLED_USING_NAMESPACE;

#define NUM_LEDS 25
#define DATA_PIN A5
#define CLOCK_PIN A3
#define BRIGHTNESS 200
#define FRAMES_PER_SECOND 30

CRGB leds[NUM_LEDS];

void setup() { 
    FastLED.addLeds<WS2801, DATA_PIN, CLOCK_PIN, EOrder::RGB, DATA_RATE_MHZ(1)> (leds, NUM_LEDS);
}

void loop() {
for(int dot = 0; dot < NUM_LEDS; dot++) { 
   //leds[dot] = CRGB::Green;
   leds[dot] = CRGB::White;
 //   leds[dot] = CRGB(255,000,000);
    FastLED.show();
    // clear this led for the next time around the loop
  //leds[dot] = CRGB::Black;
  leds[dot] = CRGB::Blue;
// leds[dot] = CRGB(000,000,000);
delay(500);
}
}`

I can change whatever I want in the code; colours, delay, data rate, pins, software mode, it makes no difference.

TheAustrian, 2016-05-02:
You need to specify what type of leds you are using when adding them, like 
`FastLED.addLeds<WS2801, RGB>(leds, NUM_LEDS);`

loonylion, 2016-05-02:
It doesn't make any difference when I do.

closed 2019-01-20 by focalintent