FastLED/FastLED #305: ESP8266 - addLeds causes WDT reset on github latest

pi-ratte, 2016-05-12:
I Just grabbed the newest Github version and its causing WDT resets on my ESP8266  when adding the leds.
I used the Fastled example  ColorPallete.ino to cross check and the example wdt resets as well.
I pinned it down to the FastLED.addLeds. It looks like its taking a unusual long pause on addleds and then it resets.

focalintent, 2016-05-12:
What type of leds are you adding?

focalintent, 2016-05-12:
And what version of the esp8266 arduino libraries are you using?

mkeyno, 2016-05-12:
hi @focalintent mine have similar problem ,I'm using the lasted version of Arduino-ESP and FastLED with Arduino 1.67 , all example compile with no error , but no effect come out with any pin and for some examples ESP start to rebooting , is it necessary to install platform like SMIG or  your library can work with ESP alone

focalintent, 2016-05-12:
I've developed the library against the esp8266 arduino core version 2.1.0 and I have a dozen+ esp8266's running here without problem (and actively using wifi for sync/control).  @mkeyno try adding `#define FASTLED_ESP8266_RAW_PIN_ORDER` to the top of your .ino files to see if that fixes how the pins line up.  Also - can you tell me which examples are causing rebooting?

mkeyno, 2016-05-12:
Thanks buddy , it works now , very neat , (color pallet example ). 
part of setting 
#include <FastLED.h>
#define FASTLED_ESP8266_RAW_PIN_ORDER
#define LED_PIN     5
#define NUM_LEDS    15
#define BRIGHTNESS  64
#define LED_TYPE    APA104
#define COLOR_ORDER GRB
CRGB leds[NUM_LEDS];

#define UPDATES_PER_SECOND 100

however I was wondering if there is any topic or link which is discuss about best way to use FastLED in ESP module or  all  hints need to be considered when we use other handler in loop such as clinethandler and websocket handler

mkeyno, 2016-05-12:
@focalintent  I have question and don't know should I explain it in this issue or open another, however  I've plan  to received the the  25 frame per second buy UDP package and then give  specific amount of data to  8 each pin of ESP  and then show the frame , I don't know whether UDP  is fast enough to give data in fastLED function to handle it in proper time ,     may I have your idea about this scheme

focalintent, 2016-05-12:
@mkeyno please ask that question (about the best way to do something) on the g+ group - http://fastled.io/+ - i want to keep these issues to focusing on bugs - and specifically, I want to keep this issue focused on the original problem, namely the WDT timeouts that were originally reported here.

mkeyno, 2016-05-12:
sure, thanks any way Daniel for great fun and happy your library brought to us

pi-ratte, 2016-05-14:
Hi im adding WS2812 leds. Its a adafruit Huzzah board and im using Arduino 1.6.9,  ESP8266 core 2.1 ( just noticed theres a 2.2 out will test with that again).

pi-ratte, 2016-05-14:
just tested it with 2.2 and im still getting the same results, it resets when adding the leds im using pin 12. for testing i just conected to the serial and nothing else connected to any pins not even the leds. So i changed it to apa102 in the configuration (CLK on 13) and still the same result, I get a wdt reset. Also tryed setting it to software spi and no difference also resets.

focalintent, 2016-05-17:
There are some issues being reported here - https://plus.google.com/u/0/112709672046334148759/posts/bTnuwsUZdQy - which make me think there's something about some of the pins that attempting to use them is causing problems.  I've been travelling quite a bit lately but hoping to get to spend some time looking into this tonight.

focalintent, 2016-05-17:
There are a handful of pins that, if you attempt to use them, will cause the board to WDT reset - specifically gpio 6-11 (no matter how they're mapped).  I'm about to check in code that will both block the use of the problematic GPIO pins, and also attempt to guess at the right pin mappings to use (assuming you're using arduino-esp8266 v2.2.0 or later)

commit ae5598f7 (2016-05-17): Fix #305 - block out pins that will cause the device to WDT on use, also put in logic to chose the correct pin to gpio mapping based on the board being compiled for.
closed 2016-05-17 by focalintent