FastLED/FastLED #306: 74 LED WS2812B strip is flickering when connected to ESP8266 with Wi-Fi enabled

krzychb, 2016-05-17:
I have been testing [Fire2012.ino](https://github.com/FastLED/FastLED/blob/master/examples/Fire2012/Fire2012.ino) sketch with latest FastLED library (saved directly from GitHub) using WS2812B strip with 74 and 144 LEDs  together with ESP8266.

Fire2012.ino works as expected when Wi-Fi of ESP8266 is off.
If Wi-Fi is on, the strip is flickering :disappointed: 

Has anybody tested ESP8266 in similar scenario? Do I miss something in configuration?
ESP8266 is all about Wi-Fi and I would like to set Fire2012 parameters (brightness, cooling and sparking) wirelessly.

I do not observe flickering for Fire2012.ino sketch when implemented with Adafruit_NeoPixel and NeoPixelBus libraries.

Here are my test sketches:
•	[EspNeoPixelBus](https://github.com/krzychb/EspFire2012/tree/master/EspNeoPixelBus)
•	[EspAdafruit_NeoPixel](https://github.com/krzychb/EspFire2012/tree/master/EspAdafruit_NeoPixel)
•	[EspFastLED](https://github.com/krzychb/EspFire2012/tree/master/EspFastLED)

I really appreciate mature, feature rich and well documented code of FastLED and looking how to use it directly with ESP8266 without patching it with additional code.

Krzysztof

msurguy, 2016-05-17:
Try adding  `FastLED.delay(1000 / FRAMES_PER_SECOND);` under this line https://github.com/krzychb/EspFire2012/blob/master/EspFastLED/Fire.ino#L109 instead of doing manual delay via `millis`.

focalintent, 2016-05-17:
@msurguy the problem with that is it then blocks the handling of wifi responses in a timely fashion.

@krzychb - what happens if you add `#define FASTLED_ALLOW_INTERRUPTS 0` to the line immediately before you have `#include <FastLED.h>` - FastLED is different from the other libraries in that it allows interrupts to fire while processing WS2812 leds - however, if an interrupt takes too long to run, then it has to stop writing out the current flame.

Out of curiosity - where are you seeing flickering - is it at the beginning of the strip, the end of the strip, or across the entire strip?

(Note:  I just ran an installation using esp8266's with 150 led tm1809 (similar timing issues as ws2812) with wifi on with no problems - and i was using wifi as both a heartbeat sync and to broadcast OSC data to nearly a dozen esp8266's).

krzychb, 2016-05-18:
Thanks, @focalintent!

Sure enough, adding ```#define FASTLED_ALLOW_INTERRUPTS 0``` resolved the issue :smile: :+1: 
Flickering occurred towards the end of the strip - plese see [this video](https://www.youtube.com/watch?v=mUFssLiALFk)

Thank you again for quick help :+1: 

Krzysztof

closed 2016-05-18 by krzychb

Lightning303, 2016-05-18:
Seems like same issue as i had. Just a heads up. Not allowing interrupts will result in watchdog resets. On my project, roughly 1 reset every 5-10 minutes.

focalintent, 2016-05-18:
Ok - yeah, what's happening here is the interrupts firing are cutting off the writing out of the frame data.  

If you're hitting watchdog resets, grab the latest version of the code and try re-allowing interrupts - I did some timing adjustments on the retries (and upped the number of retry attempts).

krzychb, 2016-05-27:
@Lightning303,

I just run about 100 min testing of  [Fire2012 on ESP8266](https://github.com/krzychb/EspFire2012/tree/master/EspFastLED) with simple web server enabled (to set up fire simulation parameters) using latest FastLED code from GitHub - https://github.com/FastLED/FastLED/commit/1b546b31896dd2c4d95c16bb17c852b3d8177eb1

I am happy to confirm that I do not see any watchdog resets.

TheAustrian, 2016-05-27:
@krzychb You might want to let it run over night to be on the safe side, i had wdt resets after about 12 to 36 hours with bitbanging and wifi but haven't tried the latest version yet.

Lightning303, 2016-05-27:
I will definitely give the latest changes a try when i find some time. However you have to take into consideration that there might be a difference between having a webserver idling and a project where there is constant traffic. In mine i send 8 byte packages to my esp 24-60 times a second.

krzychb, 2016-05-27:
@TheAustrian
This is very good point :smile: 
I have just set up a [channel on ThingSpeak](https://thingspeak.com/channels/119953) to monitor [EspFastLed](https://github.com/krzychb/EspFire2012/tree/master/EspFastLED) up time 
Let's see if it will cut the mustard.

@Lightning303
My application is simple.
The more code and traffic, the more effort is required for successful debugging and stress testing.
I believe you will soon nail down the issue and identify the root cause!
Good luck :+1:

pi-ratte, 2016-05-28:
I just tested the latest github and it looks like the interupts are causing flickering for me. 
My setup is im sending artnet to the ESP8266 (huzzah). 
if i do a RGB walk over the leds: 
interupts enabled = the leds behind the one lighting up flicker. so as the RGB walk runs the flickers get less as the walk gets to the end. 
interupts disabled = no flicker, im yield()'ing right before and after FastLED.show() was running for an hour will test some more. And only calling show() when there is actually new LED info. (640 Leds configured)

pi-ratte, 2016-05-28:
Ohh forgot to wrote im using WS2812B leds.. also without interupts it reset on me :( . 
with interupts on i noticed if making fat white/black bars,  it flickers white bars the same size down the line. nut zust random colors.
if i set #define FASTLED_INTERRUPT_RETRY_COUNT 0 
the flicker is gone.

krzychb, 2016-05-28:
As suggested by @TheAustrian I let the ESP8266 run over night got the following result:

[image]

Time to do some troubleshooting :smile:

krzychb, 2016-05-30:
I run first round of testing of the three applications described in [original issue report](https://github.com/FastLED/FastLED/issues/306#issue-155272430) loading each of them on a separate ESP8266 module.

Test started on 29-May-2015 at about 11:00 CET and took about 30 hours. Each application was posting it's ``` Up Time ``` and free ``` Heap Size ``` to [Emoncms.org](https://emoncms.org/dashboard/view?id=29419). In case of self reset the ``` Up Time ``` is  automatically reset to zero.

[image]

The number or self resets over the test period is summarized below:

| EspFire 2012 Application | Number of Self Resets |
| --- | --- |
| [EspFastLED](https://github.com/krzychb/EspFire2012/tree/master/EspFastLED) | 4 |
| [EspNeoPixelBus](https://github.com/krzychb/EspFire2012/tree/master/EspNeoPixelBus) | 0 |
| [EspAdafruit_NeoPixel](https://github.com/krzychb/EspFire2012/tree/master/EspAdafruit_NeoPixel) | 1 |

The cause of resets may be s/w, h/w or both. To eliminate h/w as the source of issue, I will move the s/w around using the same three modules in sequence below.

| EspFire2012 Application | Test 1 | Test 2 | Test 3 |
| --- | --- | --- | --- |
| [EspFastLED](https://github.com/krzychb/EspFire2012/tree/master/EspFastLED) | ESP A | ESP C | ESP B |
| [EspNeoPixelBus](https://github.com/krzychb/EspFire2012/tree/master/EspNeoPixelBus) | ESP B | ESP A | ESP C |
| [EspAdafruit_NeoPixel](https://github.com/krzychb/EspFire2012/tree/master/EspAdafruit_NeoPixel) | ESP C | ESP B | ESP A |

Later today I will stop Test 1 and move to Test 2. 

Any comments are welcome.

Krzysztof

focalintent, 2016-05-30:
Is that with interrupts enabled or disabled?

focalintent, 2016-05-30:
Actually, I might have an idea of what's going on (if this is happening with interrupts enabled) - I think the resets might be getting caused by the clock count register rolling over - if it doesn't roll over at 32 bits, but instead at something lower, some of the math that I use for timing would result in really long spins.  

I can see about putting some extra guards in around that when I get home tomorrow.

krzychb, 2016-05-30:
@focalintent,

I have disabled interrupts to avoid strip flickering:
```
#define FASTLED_ALLOW_INTERRUPTS 0
#include <FastLED.h>
```

Thank you for review of his issue. 
Resets every 500 minutes or so for my application is not a problem.
My concern is if they are due to s/w or my h/w is faulty or power supply not adequate.

krzychb, 2016-05-31:
I have moved the s/w around between modules after 22:00 (10:00 p.m.) CET yesterday as visible (reset of ``` Up Time ``` to zero) on chart below.
[image]
This was to eliminate h/w as potential source of module resets.
Unfortunately the issue with resets persists for [EspFastLED](https://github.com/krzychb/EspFire2012/tree/master/EspFastLED) and [EspAdafruit_NeoPixel](https://github.com/krzychb/EspFire2012/tree/master/EspAdafruit_NeoPixel) applications.
Also, as during previous test, [EspNeoPixelBus](https://github.com/krzychb/EspFire2012/tree/master/EspNeoPixelBus) did not suffer any resets at all.

This indicated to me potential issue with s/w. Therefore I attempted to compare code of all three libraries looking for differences how low level control of WS2812B  is implemented.
I understand that FastLED (and Adafruit_NeoPixel) is using bit banging s/w control for ESP8266.
NeoPixelBus is using UART (as configured in my sketch) or DMA, so pulse timing is done by silicon instead of s/w and interrupts.

Is my understanding correct that control of WS2812B by FastLED for ESP8266 is done using bit banging? Is there UART or DMA method implemented by FastLED for ESP8266?

I will keep this test running for another 12 hours or so and then do third swap of modules. If  anybody is interested, the charts are available on-line under https://emoncms.org/dashboard/view?id=29419. 

Any comments are welcome.

focalintent, 2016-05-31:
It is done by bit banging - and I have no plans to support the dma/usart style unless I can get it to cover all the three wire chips FastLED supports - but I do know what's causing your resets and should hopefully have a fix today or tomorrow.

krzychb, 2016-05-31:
Thanks, @focalintent.
Controlling of WS2812B fully by s/w, cutting in between Wi-Fi routines, is kind of acrobatics out of my reach :+1:
I will be happy to set up a "cluster" of couple of ESP8266's to do quicker testing :smile: 
Krzysztof

focalintent, 2016-06-02:
@krzychb grab the latest master@HEAD - this should get rid of your every 500 minute or so restarts :)

krzychb, 2016-06-02:
@focalintent can't wait to test the changes once I am back home today afternoon.
Guessing your local time I see why "just say no to mornings... " :)
Thank you for your time helping me out with this issue.
Krzysztof

krzychb, 2016-06-04:
I have loaded two ESP modules with latest commit 96704ee – ESP B used in previous testing, and a brand new ESP E. I have also reloaded EspNeoPixelBus application to ESP D - another brand new module. 

The testing started at 20:40 CET on 2-Jun-2016 and took about 2100 minutes. The summary of results is below.

| EspFire 2012 Application | Number of Self Resets | Module |
| --- | --- | --- |
| [EspFastLED_1](https://github.com/krzychb/EspFire2012/tree/master/EspFastLED) | 0 | ESP B |
| [EspFastLED_2](https://github.com/krzychb/EspFire2012/tree/master/EspFastLED) | 1 + 2 by manually cycling power | ESP E (new) |
| [EspNeoPixelBus](https://github.com/krzychb/EspFire2012/tree/master/EspNeoPixelBus) | 0 | ESP D (new)  |
| [EspAdafruit_NeoPixel](https://github.com/krzychb/EspFire2012/tree/master/EspAdafruit_NeoPixel) |  3 |ESP C |

The new module ESP E behaved suspiciously. It stopped posting data on [Emoncms.org](https://emoncms.org/dashboard/view?id=29419) and I had to reset it manually two times. It used on average about 2kB more flash than ESP B loaded with identical application. I decided to remove it from testing.

[image]

The good news is the EspFastLED_1 application on module ESP B run without any resets throughout the whole test. With previous version of FastLED librarry application restarted on average 3 times during such period.

All modules are now restarted by cycling power and testing continues. Previously tested ESP A module is brought back to testing instead of suspicious ESP E.

To be continued… :smile:

unaiur, 2016-06-07:
To drive correctly a led strip in an ESP8266 module you need to use a hardware module. NeoPixelBus can use the I2S or UART1 modules. UART1 has a 128 bytes buffer, enough to store about 11 pixels. This gives enough time to the WiFi interrupt handler to keep the thing running. I2S uses Direct Memory Access so every pixel is written out asynchronously. I also contributed to NeoPixelBus an async mode for UART1, which uses interrupts to feed the UART1 asynchronously so the CPU can compute the next frame while the hardware does its job.

focalintent, 2016-06-07:
You don't _need_ to - I have multiple projects that are using esp8266's with WS2812 style leds without using hardware - partly because, for me, parallel output is more important (and parallel output + shorter strips + interrupt's being enabled allows FastLED to work with this - I need to clean up the parallel output a bit more, then I should be able to publish that for everyone to use).

unaiur, 2016-06-07:
I can reliably drive an 800 led strip with NeoPixelBus and output 30 frames per second. 2 parallel outputs is fine: you can put the ESP8266 in the middle of a led strip. But I see no point for more than 2 parallel outputs for two reasons:
 - ESP8266 modules only have 9 usable GPIO pins and, usually, you want to connect some extra hardware to it (a button, a proximity sensor, an ambient light sensor...) 
 - You need extra point-to-point cables 

PD: I just want to point you towards a solution. I don't want to start a flamewar.

Makuna, 2016-06-07:
@krzychb Great information there.  Could you include tests for using the NeoEsp8266BitBangMethod with the NeoPixelBus in the loop?  This would provide a direct comparison with the others (Adafruit was based on the original NeoPixelBus bitbang work).  

And note, I moved away from bitbang due to locking the interrupts would cause intermittent exception in the wifi core that I was told directly from company to avoid; and without them wifi traffic can cause a malformed pulse train.

krzychb, 2016-06-07:
Thank you for all the comments, good discussion and interesting information.

Below is quick “executive summary” :smile: and update on test progress:

1. H/w: four ESP8266 NodeMCU 1.0 modules, WS2812B strip with 144 LEDs connected to GPIO3
2. S/w: [Fire2012](https://github.com/FastLED/FastLED/blob/master/examples/Fire2012/Fire2012.ino) application running with [FastLED](https://github.com/FastLED/FastLED), [Adafruit_NeoPixel](https://github.com/adafruit/Adafruit_NeoPixel) and [NeoPixelBus](https://github.com/Makuna/NeoPixelBus) libraries.
3. Six test runs done so far, each for about 2000 min / 33 h, total testing time of about 8 days
4. Application s/w is loaded to next module in the row for each test run to probe for h/w as a potential source of issues
5. [EspNeoPixelBus](https://github.com/krzychb/EspFire2012/tree/master/EspNeoPixelBus) was the only application that that survived all six test runs without any issues.
6. [EspAdafruit_NeoPixel](https://github.com/krzychb/EspFire2012/tree/master/EspAdafruit_NeoPixel) and [EspFastLED](https://github.com/krzychb/EspFire2012/tree/master/EspFastLED) experienced self-resets on average 1 time per 10 hours. 
7. Commit 96704ee unfortunately did not help to eliminate self-resets 

[image]

If you are interested, please see [live results](https://emoncms.org/dashboard/view?id=29419) and check on [technical details](https://github.com/krzychb/EspFire2012):

I feel like I have tested all the s/w almost to death. To bring some fresh facts to the discussion, now I am planning to use methodology proposed by @igrr and implemented by @djoele [here](https://github.com/esp8266/Arduino/issues/1152) to capture and save the stack trace for post mortem analysis where exactly each application fails. 

Any comments are welcome.

@makuma,
This is good idea to test NeoEsp8266BitBangMethod :+1:
 I am planning to implement it later in the afternoon CET time.

Krzysztof

focalintent, 2016-06-07:
@krzychb   Are you running your 8266's at 80 or 160mhz?

krzychb, 2016-06-07:
@focalintent,
Currently I am running all 8266's at 80MHz

focalintent, 2016-06-08:
@krzychb can you run a slight variation for me?  Comment out the `#define FASTLED_ALLOW_INTERRUPTS 0` line you have in Fire.h - yes, you'll get flickering (I'm setting up a tst rig now to try to see if I have any more room to control for that) - but I want to verify that it's the disabling interrupts for 4.3ms at a time that's tripping up your device.

igrr, 2016-06-08:
@focalintent — disabling interrupts for 4.3ms is way too long. For a busy WiFi network the stack can tolerate 20-50 microseconds at most. SDK API guide recommends not to disable interrupts for more than 10 microseconds (page 14).
@krzychb — another thing regarding stack traces: you will likely not get any. If an assertion is triggered in WiFi code which runs from level 1 interrupt, device will lock up and go into hardware WDT reset after a few seconds. Exception handler will not be called.

focalintent, 2016-06-08:
@igrr I know that, which is why I asked him to comment out that line - that re-allows interrupts being handled roughly every 30µs (I have some local bug fixes that take care of some end of strip flickering issues people were seeing with interrupts enabled).

@krzychb I've got a version of your fire code running here (with some extra local diagnostic information in the web page that gets served up) that has interrupts enabled (which should take care of the random resets you were seeing) and also takes care of the end-of strip flickering.  I basically have to hit loading the webpage 10-20/s in a continual while loop for it to effect the end of the strip, and even then it seems to only have the effect of bringing the apparent frame rate down a little bit (but still north of 10fps).

I'm going to let this run overnight here and see what kind of uptime i get with it, and if it all looks good, i'll check in these changes in the morning.

krzychb, 2016-06-08:
@igrr - now I need to look for another user case to try out exception handler :smile: Thank you for supporting this discussion.

focalintent, 2016-06-08:
11 hours runtime overnight - I've commented out the `#define FASTLED_ALLOW_INTERRUPTS 0` line in your code for this run.  I'm going to leave it running all day as well to be sure (then I need to pull the hardware off into some other work in about another 8 or so hours).

Now back to RGBW related work.

krzychb, 2016-06-08:
Below is configuration I have just loaded using FastLED commit 7a6a223.
Suggestions by @focalintent and @Makuna above are included and modules swapped between.

| EspFire 2012 Application | Functionality | Module |
| --- | --- | --- |
| [EspNeoPixelBus](https://github.com/krzychb/EspFire2012/tree/master/EspNeoPixelBus) | Neo800KbpsMethod | ESP F |
| [EspAdafruit_NeoPixel](https://github.com/krzychb/EspFire2012/tree/master/EspAdafruit_NeoPixel) |  n/a |ESP A |
| [EspFastLED](https://github.com/krzychb/EspFire2012/tree/master/EspFastLED) | #define FASTLED_ALLOW_INTERRUPTS 0 | ESP B |
| [EspFastLED_1](https://github.com/krzychb/EspFire2012/tree/master/EspFastLED) | // #define FASTLED_ALLOW_INTERRUPTS 0 | ESP D |
| [EspNeoPixelBusBB](https://github.com/krzychb/EspFire2012/tree/master/EspNeoPixelBus) | NeoEsp8266BitBang800KbpsMethod | ESP C |

Live data log - http://emoncms.org/dashboard/view&id=29419.

msurguy, 2016-06-09:
@krzychb it appears there was an initial reset but after that seems like the library has been holding up pretty well so far? Thanks for the graphs!

krzychb, 2016-06-09:
@msurguy,
I am happy to see you watching this this endurance run with me :smile:
After over 1500 min run time now, out of the five players above, the following two are holding tight (did not experience any resets):

| EspFire2012 Application | Functionality | Module |
| --- | --- | --- |
| [EspNeoPixelBus](https://github.com/krzychb/EspFire2012/tree/master/EspNeoPixelBus) | Neo800KbpsMethod | ESP F |
| [EspFastLED_1](https://github.com/krzychb/EspFire2012/tree/master/EspFastLED) | // #define FASTLED_ALLOW_INTERRUPTS 0 | ESP D |

I am planning to reset all of them and swap modules in about 11 hours, after they exceed 2000 min run time as in previous tests.

Makuna, 2016-06-09:
@krzychb I forgot to ask this, but does your test sketch include WiFi activity?  The more WiFi activity, the higher chance of interference with bit banging.  This was something that caused early investigation issue for flickering.

krzychb, 2016-06-09:
@Makuna,

Yes, my test sketch includes WiFi activity. 

ESP8266 is all about Wi-Fi and I would like to set Fire2012 parameters (brightness, cooling and sparking) wirelessly. Transmitting of small packets of data (application up time and heap size) to Emoncms.org once per minute is a way to test what particular library can handle.

Basing on test results so far, statement by @igrr above, as well as according to your documentation to NeoEsp8266BitBang800KbpsMethod, it is clear that bit bang method for ESP8266 driving 144 WS2812B NeoPixels with Fire2012 application will not cut the mustard without occasional wdt resets.

Depending on particular application occasional wdt resets may not be a big deal. If they are, then the solution is using library like yours that supports UART method (or likely DMA but not tested here).

Krzysztof

Makuna, 2016-06-09:
@krzychb By default on Esp8266 with my library, the I2S DMA method is used, you have to specify either uart or bitgang to use these other methods.

krzychb, 2016-06-10:
Thanks, @Makuna - I have corrected text above.

msurguy, 2016-06-15:
@krzychb is there anymore insight into FastLED stability?

krzychb, 2016-06-18:
@msurguy here is summary for last test that I just stopped after 10 days of run time: 

| EspFire2012 Application | Functionality | Number of Self Resets |
| --- | --- | --- |
| [EspFastLED](https://github.com/krzychb/EspFire2012/tree/master/EspFastLED) | #define FASTLED_ALLOW_INTERRUPTS 0 | 22 |
| [EspFastLED_1](https://github.com/krzychb/EspFire2012/tree/master/EspFastLED) | // #define FASTLED_ALLOW_INTERRUPTS 0 | 0 |

Both application control WS2812B strip with 144 LEDs connected to GPIO3 of ESP8266. 

With disabled interrupts (EspFastLED), Fire2012 simulation runs smoothly but ESP module self-resets on random basis (on average 2 times a day).

If interrupts are enabled (EspFastLED_1) then there are no self-resets at all but Fire2012 simulation is [flickering](https://www.youtube.com/watch?v=mUFssLiALFk) towards end of the strip. 

I am planning to provide more detailed update of testing in couple of days in [EspFire2012](https://github.com/krzychb/EspFire2012) repository.

In last test run I am going to add [OTA]( https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.md) to provide automatic scheduled distribution of application to 5 test modules and verify if they deliver 60 frames/s.

focalintent, 2016-06-18:
The flickering at the end of the strip is caused by an interrupt taking too long and forcing FastLED to abandon writing the frame - FastLED can attempt to retry a frame if it detects that it was interrupted - you can control how many times it attempts to retry by adding:

```
#define FASTLED_INTERRUPT_RETRY_COUNT 3
```

before the include of FastLED.h.  I had it at 2 and wasn't seeing any flickering when running fire on an active wifi network across 180 leds with a test rig last week.

I can also see about extending the time before FastLED will treat an interrupt gap as being "too long" - though in my experience, going much above 5µs caused problems.

Also - this is a problem you can mitigate somewhat by running at 160Mhz.  At 80Mhz, an interrupt has 400 cycles to do its thing before it causes a problem for writing out WS2812 data.  At 160Mhz, that same interrupt has 800 cycles - increases the odds that the interrupt handling will fit in the gap.

(And really, if an interrupt handler is taking 400 cycles to do _anything_ then it is a very poorly written interrupt handler and should be fixed - interrupt handlers should move as little data as possible, and set flags - heavy lifting should be done _outside_ of the interrupt handler).

Are you still seeing flickering on the most recent FastLED@master with 144 leds?  I wasn't with 180.

krzychb, 2016-06-18:
Hi @focalintent,

I hope not to distract you with this issue as I understand you are in the middle of big overhaul of FastLED. I will keep quiet here unless asked for specific update.

I just did testing with:
1.	Latest FastLED library a3641f0 
2.	[EspFastLED_1](https://github.com/krzychb/EspFire2012/tree/master/EspFastLED_1) - posted on GitHub for reference
3.	esp8266 / Arduino core stable [2.2.0](https://github.com/esp8266/Arduino/releases/tag/2.2.0) as well as staging [2.3.0-rc2]( https://github.com/esp8266/Arduino/releases/tag/2.3.0-rc2)
4.	CPU speed 80MHz and 160MHz
5.	WS2812B strip with 144 LEDs as on picture below

I am still seeing flickering with:
```
#define FASTLED_ESP8266_RAW_PIN_ORDER
// #define FASTLED_ALLOW_INTERRUPTS 0
#define FASTLED_INTERRUPT_RETRY_COUNT 3
#include <FastLED.h>
```

Flickering is gone with:
```
#define FASTLED_ESP8266_RAW_PIN_ORDER
#define FASTLED_ALLOW_INTERRUPTS 0
// #define FASTLED_INTERRUPT_RETRY_COUNT 3
#include <FastLED.h>
```
[image]

focalintent, 2016-06-18:
I know that flickering goes away when you disable interrupts, what I'm interested in is trying to tune things so that when interrupts are enabled I can recover from the frame writing being interrupted by bad interrupt handlers.   (Also curious why you're seeing flickering in your setup and I wasn't here).

focalintent, 2016-06-18:
In https://github.com/FastLED/FastLED/blob/master/platforms/esp/8266/clockless_esp8266.h - if you get a chance, try adjusting the time for WAIT_TIME - currently set at 5us - that's how long of a gap that is allowed for interrupts.  If interrupts take longer than that value, then the frame writing is cut off (which causes the LEDs at the end of the strip to be updated at a much slower rate) - if you set that value too high, however, you run this risk of causing flickering at the _front_ of your strip (as the chips will reset and led data meant for LEDs near the end of the strip will instead be picked up by the LEDs at the beginning).

krzychb, 2016-06-18:
@focalintent,

I will then focus on scenario with interrupts enabled by tuning the following variables:
- [ ] CPU speed 80 MHz or 160 MHz
- [ ] WAIT_TIME around default 5us
- [ ] FASTLED_INTERRUPT_RETRY_COUNT 1 or 2 or 3 or 4

Now I thinking how to measure flickering other than by looking at it that may be subjective.
Will tracking of ```_retry_cnt``` and ``` _frame_cnt ``` over the run-time be a the way to go?

krzychb, 2016-06-19:
I have set up a test rig with five NodeMCU 1.0 modules each one with different ``` WAIT_TIME ``` ranging from 8 to 16us. Number of retries per minute (using ``` _retry_cnt ```) when writing a frame recorded so far is shown below:

[image]
It is clearly visible that number of retries is decreasing if ``` WAIT_TIME ``` is extended. I will leave this set up working as before to check how different ``` WAIT_TIME ``` is affecting stability of EspFire2012 application.

Full set of on-line charts for motioning of other variables is available on [Emoncms.org](http://emoncms.org/dashboard/view&id=34789) as before.

russp81, 2016-11-05:
@focalintent @krzychb 
I will start by saying I am nowhere near as knowledgeable on this as you guys.  I'm a self taught coder, and I have a limited grasp on the h/w and s/w relationships (do's and don't do's, etc).  I tried setting the following as mentioned by @focalintent in his last post:

dersimn, 2016-11-27:
Thanks @krzychb for the scientific analysis on the flickering issue. I just wanted to tell that I came to the same conclusion using the information you provided in this thread.

As soon as you use Wifi features on the ESP8266, the WS2812 will start to flicker. Using the `#define FASTLED_ALLOW_INTERRUPTS 0` workaround will however cause the board to make unpredictable resets (according to the serial output these resets are invoked by the watchdog). The [NeoPixel library](https://github.com/Makuna/NeoPixelBus) will work fine, with the small restriction that you won't be able to use Serial input anymore, since GPIO3 is the same pin as RXD0.

I ended up replacing just the output procedure by the NeoPixel library, since the FastLED has some pretty cool and easy features for animation.  
So just alter the output part in your `loop()` from something like this

```
FastLED.show();
FastLED.delay(1000/FRAMES_PER_SECOND);
```

to

```
static uint32_t fireTimer;
if (millis() > fireTimer + 1000 / FRAMES_PER_SECOND)
{
  fireTimer = millis();

  RgbColor pixel;
  for (int i = 0; i < NUM_LEDS; i++)
  {
    pixel = RgbColor(leds[i].r, leds[i].g, leds[i].b);
    strip.SetPixelColor(i, pixel);
  }
  strip.Show();
}
```

assuming you defined the `leds` array like this `CRGB leds[NUM_LEDS];`

marksev1, 2016-12-15:
dersimn can you please explain more in detail how you did this mashup of both libraries...Can you maybe make a repository with best parts of both libraries :D.

So what is the consensus here, resets or flickering in any case or how?

alexmoz2016, 2017-03-01:
Another side-effect of disabling the interrupts is that the "Smart Config" doesn't work anymore. Smart Config allows sending wifi essid/password from the app to ESP8266 to allow it connecting to the wifi.
wifi.beginSmartConfig()

distrakt, 2017-07-31:
Very useful thread. I'll add my anecdotal data...
• Using Wemos D1 Mini (8266's at 80mhz) boards
• Using FastLED (whose api I like **very** much! thank you!)
• FastLED 3.1.3 release
• using #define FASTLED_ALLOW_INTERRUPTS 0
• Project One has 24 LEDs, runs well, but resets every few hours (casual observation)
• Project Two has 300 LEDs. Sometimes the second half (or so) of the LED strip will lock up while animation continues on the first 140 or so LEDs. Often will "come back" after a while.
• Project Two's web API works reliably from laptop (Safari or Chrome browser), but intermittently from iPhone. Sometimes many retries/reloads are needed before it "locks in". (again, casual observation)

Any insights appreciated! Hope the anecdotal info is handy for others to see.

andgeno, 2017-11-07:
Hi, has there been any improvements in the meanwhile? Could this problem be solved by replacing an ESP8266 with an ESP32?

coryking, 2017-12-12:
For what it's worth, I've added DMA support for ESP8266 on my own fork of FastLED:
https://github.com/coryking/FastLED

I've been using it for a week now on three of my lighting projects.  One is a 300 light ws2812b strip, another is a 160 light rgbw sk6812 strip. All have been buttery smooth after switching to DMA.  Way better than the big-banging method.  The one drawback is you are forced to use GPIO pin 2 (aka the "RX" pin on a lot of dev boards).  

If you pull down my repo, to use the DMA you'll just:
`#define FASTLED_ESP8266_DMA`

Also note that while I haven't tried, I'm 99% sure this won't work with FastLED's multiple strip feature...

PS: The fork also has support for RGBW strips and actually makes use of the White LED (only esp8266 though).   For RGBW, you'll have to:
`#define FASTLED_RGBW`

lance36, 2018-01-05:
@coryking im unable to light up at all my 5 led WS2811 strip with your fork.
i installed it right into Arduino IDE removing the official FastLED

if i just comment out `//#define FASTLED_ESP8266_DMA`  it works just like before

maybe its because im not using a 3v3 <-> 5v level shifter? but should it matter on a strip so short?

coryking, 2018-01-05:
I've only tested this /w the SK6812 RGBW strips and the strip below, which
claims to be a WS2812b:
https://www.amazon.com/gp/product/B00VQ0D2TY

Could be timing?  I kinda doubt it is a level-shift issue....

lance36, 2018-01-05:
> I kinda doubt it is a level-shift issue....

yeah same, what u mean timing? i seem to be unable to even light up my strip on the blink example.

coryking, 2018-01-06:
stupid question -- are you using the right pin?

The DMA stuff can only use GPIO3, which on my boards is the "RX" pin.  In
fact, it will completely ignore whatever pin you pass into FastLED...

[20 more replies]