FastLED/FastLED #308: FastLED Error Compiling -  static_assert(validpin(), "Invalid pin specified");

dawie180, 2016-05-18:
Hi,

I've got a strip of W8212 LED's hooked up to a custom PCB running a Atmega644PA at 16Mhz, I'm using FastLED 3.1 and IDE 1.6.7.   I get a compiling error whenever i include the FastLED library.  I have the the DATA PIN connected to the PB7 pin on the 644 which translates to pin 15 in IDE.  I've used this library before on a custom PCB running a Atmega328 at 16Mhz without any issue, i do see that with the Atmega328 PB7 would just be pin 7 in IDE and with the 644 it translates a bit differently (PB7 = pin 15)... So i don't know if that has something to do with it.  

```
In file included from C:\Program Files (x86)\Arduino\libraries\FastLED-master/FastLED.h:40:0,

                 from C:\Users\feste\AppData\Local\Temp\arduino_ee0154f342bd35375a3767f897a1a60c\Blink.ino:1:

C:\Program Files (x86)\Arduino\libraries\FastLED-master/fastpin.h: In instantiation of 'class FastPin<4u>':

C:\Program Files (x86)\Arduino\libraries\FastLED-master/platforms/avr/clockless_trinket.h:96:49:   required from 'class ClocklessController<4u, 4, 10, 6, (EOrder)10u, 0, false, 10>'

C:\Program Files (x86)\Arduino\libraries\FastLED-master/chipsets.h:322:7:   required from 'class WS2812Controller800Khz<4u, (EOrder)10u>'

C:\Program Files (x86)\Arduino\libraries\FastLED-master/FastLED.h:92:52:   required from 'class WS2812B<4u, (EOrder)10u>'

C:\Program Files (x86)\Arduino\libraries\FastLED-master/FastLED.h:263:39:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2812B; unsigned char DATA_PIN = 4u; EOrder RGB_ORDER = (EOrder)10u]'

C:\Users\feste\AppData\Local\Temp\arduino_ee0154f342bd35375a3767f897a1a60c\Blink.ino:22:62:   required from here

C:\Program Files (x86)\Arduino\libraries\FastLED-master/fastpin.h:206:2: error: static assertion failed: Invalid pin specified

  static_assert(validpin(), "Invalid pin specified");

  ^

exit status 1
Error compiling.
```

TIA

focalintent, 2016-05-18:
It does - there's no pin mappings for the Atmega644PA defined, much less your custom PCB with it.  You need to take a look at platforms/avr/fastpin_avr.h and add a section defining the pin/port mappings for your 644.

closed 2016-05-18 by focalintent

dawie180, 2016-05-18:
Thank you, going to give it a go right now. That was a super humanly fast response...

dawie180, 2016-05-19:
Thanks, it worked... just a quick question... i now get 'was not declared in this scope', i did read that the workaround for this is is to define the function before it is called, is there not a simpler way?

focalintent, 2016-05-19:
Make sure you're running the most recent version of the arduino IDE - but you may still have to define the functions before they're called, this is a pretty severe/tricky bug (in reality, it's a series of bugs) in the arduino IDE.

dawie180, 2016-05-19:
Thanks, i am running the latest version... i went and edited my sketch and defined all the functions 1st, so its working 100% again. Thanks for your help, really apreciated.

ShreddinPB, 2017-08-16:
This is a bit late since this was a year ago, but I am running into this problem on the Rabid Prototypes TAU board and have no idea how to fix it. Its supposed to be Arduino Zero compatible, fastled compiles for the zero but gives me this error for the TAU.
Funny thing is, if I have the TAU set as the board in the board manager and I get board info, it reports its "BN: Arduino/Genuino Zero (Native USB Port)"
However it wont upload if I set it to a Zero, cant find board.

tonys0, 2019-10-15:
Hi Daniel,
I have the same issue
static_assert(validpin(), "Invalid pin specified");
with ESP32-WROOM-32 board. I guess the pin mapping is in 
...\libraries\FastLED\platforms\esp\32\fastpin_esp32.h
So what's wrong  this time?

marmilicious, 2019-10-17:
@tonys0  Unfortunately Daniel will not be able to get back to you.  Just wanted to make you aware of things.
https://www.reddit.com/r/FastLED/comments/czenwv/sad_news/

ktrzeciaknubisa, 2024-04-04:
The issue is most probably related to the fact, that in your code before this:
   
    #include <FastLED.h>

you have PIN constant declared:

    #define PIN 4

If you remove that, it should compile fine.

dawie180, 2024-04-04:
It was because the pin mappings, it was not included in arduino IDE because
no Arduino boards use this chip, so had to add it.