FastLED/FastLED #309: WS2812B on ESP8266: LEDs 'leaking through'

bombarie, 2016-05-29:
I know that the developments of FastLED on the ESP8266 are probably still experimental but here's something I ran into.

I'm using a Wemos D1 ESP8266 breakout in Arduino 1.6.8, using the version 2.2.0 ESP8266 board definitions. The data pin of the WS2812B strip is in port D2 of the Wemos D1 (which maps to GPIO4 on the ESP8266).

I have a 5v WS2812B strip of 30 leds but just to fool around with my program I had only 5 in use in my Arduino program. I noticed that the first 5 leds will eventually 'bleed through' to the next 5, and there is stops bleeding through. It appears to be clocked through verbatim, so led 6 will be the same as led 1, led 7 the same as led 2, etc.

Anything particular I could do to fix this? I've tried different ways of defining pin order (ie RAW_PIN_ORDER or D1_PIN_ORDER) and tried different pins on the ESP8266 too.

focalintent, 2016-05-29:
It might be a case of writing to the leds too quickly - so when writing a frame instead of leds[0] being read by the first led the data is still getting passed through the chain.  It's also possible that the code I have for catching/working woth interrupts is rewriting frames too quickly.   (The neopixel library fully blocks interrupts (which also means it doesn't bother with attempting to rewrite the frame if necessary because of an interrupt).  I'm traveling right now and won't be able to really look at this until Tuesday.

bombarie, 2016-05-29:
Well it's not blocking my project at all so no rush on my behalf. Just
posted it for the heads up. Thanks for taking the time to look at this.
Only recently found FastLED and I'm amazed at what the project has pulled
off, great work!

mkeyno, 2016-05-31:
hi @focalintent Danial , don't you suggest to  use level translator  when we use 3.3 V ESP ? I fixed most of my problem when go back to 5  volt level , I use 74HC244, an 8-bit buffer/line driver to change  level to 5 volt and its works , now all pin except D0 & D1 (16, 4) works file  and received for these two pin following error

`
system param error
01 00 00 00 01 00 00 00 01 00 00 00 01 00 00 00 01 00 00 00 01 00 00 00 01 00 00 00 01 00 00 00 
01 00 00 00 01 00 00 00 01 00 00 00 01 00 00 00 01 00 00 00 01 80 02 80 02 00 00 00 03 00 ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 29 04 00 00 
01 00 10 00 fc 70 65 3d 22 74 65 78 74 22 20 69 64 3d 22 5f 66 69 6c 65 22 20 73 74 79 6c 65 3d 
22 70 6f 73 69 74 69 6f 6e 3a 61 62 73 6f 6c 75 74 65 3b 6c 65 66 74 3a 38 32 30 70 78 3b 74 6f 
70 3a 34 34 70 78 3b 77 69 64 74 68 3a 39 38 70 78 3b 68 65 69 67 68 74 3a 32 33 70 78 3b 6c 69 
6e 65 2d 68 65 69 67 68 74 3a 32 33 70 78 3b 7a 2d 69 6e 64 65 78 3a 32 3b 22 20 6e 61 6d 65 3d 
22 45 64 69 74 62 6f 78 31 22 20 76 61 6c 75 65 3d 22 22 20 74 65 78 74 2d 61 6c 69 67 6e 3d 22 
72 69 67 68 74 22 3e 0d 0a 3c 69 6e 70 75 74 20 74 79 70 65 3d 22 74 65 78 74 22 20 69 64 3d 22 
5f 70 61 74 68 22 20 73 74 79 6c 65 3d 22 70 6f 73 69 74 69 6f 6e 3a 61 62 73 6f 6c 75 74 65 3b 
6c 65 66 74 3a 35 35 37 70 78 3b 74 6f 70 3a 34 34 70 78 3b 77 69 64 74 68 3a 39 38 70 78 3b 68 
01 80 00 00 7e 00 dc e8 00 10 00 00 01 2f 63 6f 6e 66 69 67 2e 68 74 6d 6c 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 03 00 04 00 06 00 08 00 0a 00 0c 00 0e 00 10 00 
12 00 14 00 16 00 18 00 1a 00 1c 00 1e 00 21 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
01 00 11 00 fc 65 69 67 68 74 3a 32 33 70 78 3b 6c 69 6e 65 2d 68 65 69 67 68 74 3a 32 33 70 78 
3b 7a 2d 69 6e 64 65 78 3a 33 3b 22 20 6e 61 6d 65 3d 22 45 64 69 74 62 6f 78 31 22 20 76 61 6c 
75 65 3d 22 22 20 74 65 78 74 2d 61 6c 69 67 6e 3d 22 72 69 67 68 74 22 3e 0d 0a 3c 21 2d 2d 20 
73 68 6f 77 20 6c 69 73 74 20 2d 2d 3e 0d 0a 3c 64 69 76 20 69 64 3d 22 73 68 6f 77 66 69 6c 65 
73 22 20 73 74 79 6c 65 3d 22 70 6f 73 69 74 69 6f 6e 3a 61 62 73 6f 6c 75 74 65 3b 6c 65 66 74 
3a 36 34 31 70 78 3b 74 6f 70 3a 31 37 32 70 78 3b 77 69 64 74 68 3a 31 33 33 70 78 3b 68 65 69 
67 68 74 3a 37 31 70 78 3b 7a 2d 69 6e 64 65 78 3a 34 22 3e 0d 0a 3c 2f 64 69 76 3e 0d 0a 3c 69 
6e 70 75 74 20 74 79 70 65 3d 22 62 75 74 74 6f 6e 22 20 69 64 3d 22 42 75 74 74 6f 6e 33 22 20 
01 80 00 00 7e 00 dc e8 00 11 00 00 01 2f 63 6f 6e 66 69 67 2e 68 74 6d 6c 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 03 00 04 00 06 00 08 00 0a 00 0c 00 0e 00 10 00 
12 00 14 00 16 00 18 00 1a 00 1c 00 1e 00 21 00 23 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff 
01 80 00 00 7e 00 dc e8 00 17 00 00 01 2f 63 6f 6e 66 69 67 2e 68 74 6d 6c 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 03 00 04 00 06 00 08 00 0a 00 0c 00 0e 00 10 00 
12 00 14 00 16 00 18 00 1a 00 1c 00 1e 00 21 00 23 00 25 00 27 00 29 00 2b 00 2d 00 2f 00 ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
01 00 18 00 fc 74 6f 70 3a 39 33 70 78 3b 77 69 64 74 68 3a 33 30 37 70 78 3b 68 65 69 67 68 74 
3a 34 30 70 78 3b 7a 2d 69 6e 64 65 78 3a 31 33 3b 22 3e 0d 0a 3c 66 6f 72 6d 20 6e 61 6d 65 3d 
22 6d 79 5f 66 6f 72 6d 5f 22 20 6d 65 74 68 6f 64 3d 22 70 6f 73 74 22 20 61 63 74 69 6f 6e 3d 
22 68 74 74 70 3a 2f 2f 31 39 32 2e 31 36 38 2e 34 2e 31 2f 75 70 64 61 74 65 22 20 65 6e 63 74 
79 70 65 3d 22 6d 75 6c 74 69 70 61 72 74 2f 66 6f 72 6d 2d 64 61 74 61 22 20 69 64 3d 22 46 6f 
72 6d 31 22 3e 0d 0a 3c 69 6e 70 75 74 20 74 79 70 65 3d 22 73 75 62 6d 69 74 22 20 69 64 3d 22 
75 70 6c 6f 61 64 62 75 74 74 6f 6d 22 20 6e 61 6d 65 3d 22 22 20 76 61 6c 75 65 3d 22 55 70 6c 
6f 61 64 20 46 69 6c 65 22 20 73 74 79 6c 65 3d 22 70 6f 73 69 74 69 6f 6e 3a 61 62 73 6f 6c 75 
[... cut ...]
00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 03 00 04 00 06 00 08 00 0a 00 0c 00 0e 00 10 00 
12 00 14 00 16 00 18 00 1a 00 1c 00 1e 00 21 00 23 00 25 00 27 00 29 00 2b 00 2d 00 2f 00 31 00 
33 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff f�ff ff ff ff ff ff ff 
ff ff ff ff 
03 00 03 00 00 00 03 00 00 00 03 00 00 00 03 00 00 00 03 00 00 00 03 00 00 00 03 00 

` 
but I think its related to my level translator , I have plan to  use such BSS183  part 
do you have any suggestion

focalintent, 2016-05-31:
@mkeyno _please_ keep general conversation (and requests about best practices and such) to the g+ group - this is really only for problem reporting.   (And if you're reporting a new issue - open a new issue, please stop hijacking other issues).

And yes - many people report problems improving with a level translator (the issue reported here that you've hijacked has nothing to do with a level translator).  If you continue to have problems with a specific pair of pins then file a new ticket/issue.

mkeyno, 2016-05-31:
wow , I did not know that , I thought this issue related with my experience for ESP and WS2812, however so sorry for any unintentional hijacking , I will clean this issue and open new ticket

focalintent, 2016-06-02:
@bombarie try pulling master@head - I made a couple changes that might help with the bleeding that you're seeing.

It's a side effect of work that I've done to allow interrupts - if I catch that an interrupt has run too long, then I need to re-write the entire frame, or at least attempt it.  If I'm not pausing long enough before starting that re-write, however, the new frame will be interpreted by the WS2812's as starting from the end of the last frame.  Oops :)   I've tried adjusting the timing there a little bit.  Also, you can disable the interrupt retry attempts by putting:

```
#define FASTLED_INTERRUPT_RETRY_COUNT 0 
```

before your `#include <FastLED.h>` line.

focalintent, 2016-06-08:
Ok, I've fixed the leaking through issue here.   Running an overnight test (this is part of something else) and should have the fix for this, and general esp8266 + interrupt issues, checked in tomorrow morning.

closed 2016-06-08 by focalintent

bombarie, 2016-06-10:
Thanks for fixing this issue @focalintent and apologies for not having been able to test this yet, I haven't been around my ESP8266 this week. I'll test is anyway once I do and will report if anything odd turns up.

commit 7a6a2238 (2016-06-08): Fix #309 and wrap up #306