FastLED/FastLED #311: LEDs become more yellow (instead of white) with every LED illuminated

ArduinoBasics, 2016-05-31:
Hi there,

I am using the IC Station WS2812-B LED ring:
[http://www.icstation.com/icstation-ws2812-programmable-colorful-board-p-5248.html](url)

When I set one pixel to full brightness white, it looks fine, but when I try to do the same with all 16 pixels lit at the same time - it looks more like yellow.

I have tried different ways to make it white, - HSV, setRGB, CRGB::White   etc etc... but they are all doing the same thing.

I am using an external power supply that is more than capable of powering these LEDs.

Anyone know what I can do to fix this ?

Regards
Scott

focalintent, 2016-05-31:
What board are you using, what version of arduino, what version of the library, how do you have things wired and what kind of power supply do you have wired up?

ArduinoBasics, 2016-05-31:
Board: Arduino UNO clone (IC Station)
Arduino IDE version 1.6.4
FastLED version 3.0.3
5V 4A power supply  - to rails of breadboard (have a 4700uF capacitor between pos and neg).
Pos rail to 5V of LED ring
Neg rail to GND of LED ring
Digital Pin 9 of Arduino to Din of LED ring
Pos rail to 5V line of Arduino
Neg rail to GND of Arduino

MFornander, 2016-05-31:
Sounds like a power supply issue.  I've been working with high power LEDs and usually the red needs less voltage than green or blue:
50w RGB: DC Forward Voltage (VF):Red 16-18V;Green 24-28V; Blue 24-28V

So if your power supply is overloaded in some way and the voltage drops, the green and blue will start to fade before the red.

TL;DR: Upgrade your power supply amp rating, or potentially feed the power from both ends to prevent voltage drop at the far end.

electrokean, 2016-05-31:
Could just be the wire gauge from power supply to LED ring is too thin for the current draw and length, causing enough voltage drop to start dimming the blue LEDs.

ArduinoBasics, 2016-05-31:
Hi Electrokean - I did have a long thin wire from power rail to 5V on LED ring. I replaced it with a thicker and shorter wire, and have noted significantly improved results. 
Still a little bit yellow when I get to 32 LEDs, but way better than it was... I will check the other wires and replace as necessary - but this does seem to be the key contributor to the issue. Thank you very much.

pubcoder, 2016-05-31:
I have a very similar 16-piece ring, just on a black PCB and it works fine with Arduino Nano powered on USB (5V).
These 5050 SMDs use about 0.24W each, which is 3.84W in total, hence needs about 800mA which is within reach of most USB ports.
So it must be either the wiring, (too thin wires or there is shortening), or perhaps the power supply does not meet specification.
Also it can be difficult to solder the contacts which are of "surface mount" type so one has to hold it steady when soldering, so check if the wires are soldered well.
The ring is also blindingly bright and gets warm if all leds are kept at 100% white.

ArduinoBasics, 2016-05-31:
I only had all LEDS on full white for about a second... 
The LEDS worked well individually, but only experienced the issue when ALL LEDs on the board were on/lit.
After I fixed the issue - by replacing the long thin wires with thicker short ones,I decided to see how much current was being drawn as the number of LEDs increased.

And now because of the multimeter wires, I see the problem returned. 
Interestingly, the amount of current drawn by the LED ring is not linear.

1 LED    = 70 mA
2 LEDs  = 100 mA
3 LEDs =  120 mA
16 LEDs = 160 mA
32 LEDs = 168 mA

This probably explains why it is turning yellow.
The current per LED is decreasing as the number of LEDs turned on increases..

Perhaps someone can explain what is happening here ??.

Regards
Scott

Daniel-dk, 2016-05-31:
We get this behavior on LED strip due to voltage drops ( ohms law committed on the conductors  ),
the blue LED in the RGB set goes dimmer and you end up with a yellow-ish white ( more green and red than blue )
Blue and green have higher forward voltages than red so they will be the first to dim out ( first Blue , then Green, and lastly Red )

The longer the strip, the more current is needed  and you get larger voltage drops on the conductors.
Our solution is to have multiple power injection points,
or to lower the total brightness to limit current usage ( and thus voltage drop at the end )

electrokean, 2016-05-31:
Glad the heavier wires helped - it is a common issue.  Many of these RGB LEDs don't tolerate much voltage drop from 5V.  The typical description is pinking (loss of brightness in both green and blue), but yellowing (loss of mainly blue) also happens.
Regarding the multimeter, it isn't the leads but the burden voltage of the meter when measuring current.  If you want to learn more about that do a Google search on "burden voltage eevblog".

ArduinoBasics, 2016-05-31:
Thanks again electrokean - I have never heard of burden voltage before,,, 
So the non-linearity of my measurements is likely to be function of the burden voltage ?

electrokean, 2016-05-31:
Yes, the multimeter passes the current through a resistor to measure the voltage across that (V=IR), so as the current increases so does the burden voltage of the multimeter.  This leads to less voltage getting to the LEDs, and they'll begin to draw less current as the green and blue can't light up fully.
Here's another good description of burden voltage: http://jeelabs.org/wp-content/uploads/2012/04/05/tk-burden-voltage/index.html

closed 2016-05-31 by ArduinoBasics