FastLED/FastLED #312: disappointing result using APA102 with NodeMCU rev 1.0 & 74HC244 , 8-bit buffer/line driver CMOS level

mkeyno, 2016-06-01:
hi @focalintent Danial , I've tried to run ColorPalette example of my NodeMCU module with APA102 LED strip , as you may know , I didn't get any result with direct connection between Node and Strip , even it is not lighten up, so I move to TTL converter and use  74HC244 , 8-bit buffer/line CMOS level driver 

, but whereas I could successfully run APA104 LED strip on most of NodeMCU pin (except D0, D1, D2) but for the APA102 very disappointing , it seems the big noise upon that strip even when choose the less LED num , the remain LED get fluctuated

mkeyno, 2016-06-01:
Danial , is possible to use raw SPI  to control the strip like this code
https://github.com/leonyuhanov/esp8266apa102driver/blob/master/esp8266apa102driver.ino

although I ran this code and received same problem, but I was thinking maybe 74HC244 can't follow on high speed  ESP SPI and that's the main problem, so if we use the raw SPI programming we could slow down the SPI speed

focalintent, 2016-06-01:
Have you tried changing the speed that you're writing with for the APA102 - as described here - https://github.com/FastLED/FastLED/wiki/SPI-Hardware-or-Bit-banging?  Also, are you sure that you don't have your clock and data lines backwards?  

And no, i haven't implemented hardware SPI support yet for the ESP8266, which is why it falls back to bit banging (but on the flip side, that means you can use any pair of pins that are accessible/usable for GPIO)

mkeyno, 2016-06-01:
I did Danial , I add  the  this code #define FASTLED_FORCE_SOFTWARE_SPI  at top of the sketch ,and almost change all pair of pins in NodeMcue , by this modification noticed the brightness of LED decreased and most of LED turned off , compared to last try 
also I change my borad to Arduino pro mini and remove the  #define FASTLED_FORCE_SOFTWARE_SPI  and run the  sketch again ,I noticed  the program worked but color of all palette change to red and dim green only , also it seems the speed is lesser than when I run it with APA104, 
is it possible APA102 not implemented correctly

mkeyno, 2016-06-01:
I used this line of code 
FastLED.addLeds<APA102,7,11,RGB,DATA_RATE_MHZ(10)>(leds,NUM_LEDS);
can I change the DATA_RATE_MHZ to other data?
.and my power is 5 volt -2 A (10 watt) for only 30 LED (almost 30*.2=6 watt)

mkeyno, 2016-06-01:
I've change the power  supply  to 40A 5 volt and on Arduino pro mini work as it expected , but still have problem on ESP

focalintent, 2016-06-01:
I'd say try 5.  Also be aware that a low of 5v 2A power supplies that have a usb port on them (aka, your average cell phone charger) will only put out 500mA by default without having a set of resistors across various pins on the usb cable to indicate higher charging capacities.

mkeyno, 2016-06-01:
I've check all two pair pins  and by pass the 74HC244, no luck , I'm gonna try BSS138 or any ttl translator

mkeyno, 2016-06-20:
hi @focalintent Danial , I ran the couple of test , the code only work when use 1 MHs , but not very perfectly and still some flicking and noisy appear

focalintent, 2016-06-21:
I don't know why you're having a problem - as it is working on setups that I have here.  I'm also a bit short on time right now - since your problem/question is more an advice thing than a bug report, I would strongly recommend posting your question(s) to the g+ group - http://fastled.io/+ - where there are a larger number of people who can help you with your hardware setup/wiring/power issues.

closed 2016-06-21 by focalintent