FastLED/FastLED #313: Last LED flickering on WS2801 strip with Teensy

woundedlion, 2016-06-03:
I'm continuing to improve my spherical POV display. After swapping out the Arduino Uno for a Teensy 3.2, I'm getting flickering and wrong colors on the last LED in the strip. This is consistent no matter how many LEDs of my 40 I actually address.

I'm using the latest version of FastLED (pulled last night) and hardware SPI. Hardware is a Teensy 3.2, SN74HCT245N for level shifting, and 2 20-LED WS2801 strips from adafruit wired in series.

I found previous commits that wait for data to be written out to the bus before releasing the hardware SPI pins. Looking at the code for arm_k20, it does indeed call waitFully() before release(). But, when I comment out the release() call. the whole strip paints properly. So, I'm wondering if the waitFully() implementation for arm_k20 is correct.﻿

woundedlion, 2016-06-04:
Played around with some code this morning. The below implementation works for me, but I've only tested with my code and setup. The idea is to wait for the last TX to complete, then wait for the last RX to complete. The final RX completion should guarantee that the last TX is truly done. Not sure I have it right as I don't think I understand the dynamics of the queues and what the wait() and wait1() functions do.

	static void waitFully() __attribute__((always_inline)) {
		while ((SPIX.SR & 0xF000) > 0) {
			while (!(SPIX.SR & SPI_SR_TCF));
			SPIX.SR |= SPI_SR_TCF;
		}
		while (!(SPIX.SR & SPI_SR_TCF));
		SPIX.SR |= SPI_SR_TCF;

		while ((SPIX.SR & 0xF0) > 0) {
			SPIX.POPR;
		}
	}

mkeyno, 2016-06-06:
hi @woundedlion ,is Tx & Rx use for feeding image to POV?

woundedlion, 2016-06-06:
I am using fastLED.show() to draw each column of the image (actually 2 opposite columns since it's a bike wheel spinning on its edge with LEDs on both sides). On the k20 at least, FastLED uses PUSHR (TXFIFO push) to transmit. So TX is for transmitting the pixels to the strip. The RXFIFO is probably getting garbage data since I don't even have MISO plugged in. But, I believe SPI will do both a TX and RX shift on each round, so RXCTR will increment with every TX even though the strip is not sending any response data.

mkeyno, 2016-06-06:
actually I make POV with teensy and APA102 LED strip  but , I don't dig into this level, just read the image from SC card and show it ,following this is image of running  POV , unfortunately can't provide good picture with my phone camera , but I'll upload the video and send its link

[image]

focalintent, 2016-06-07:
Ok - I just checked in a change that's a somewhat simpler form of what you have up there - it seems to work well enough here, but I no longer have any ws2801's to test with - let me know if that takes care of what you're doing.

woundedlion, 2016-06-07:
It's working on my setup with your patch. Thank you!

closed 2016-06-08 by focalintent

theaeolianmachine, 2016-08-23:
Hey @focalintent, I have a Teensy 3.2 connected to PJRC's prop shield driving some APA102's. On master, the code runs for about 10 loops before the LED light on the Teensy goes out and it stops working.

I just did a git bisect from b01f941d0964eaf486f4d050e74ae6a65a5816d7 (v3.1.0 tag, where it works fine) to master (4c32e652ce487f8bacdc9bf918317266f5e169c5), and the offending commit where it starts acting up is 4a71eeabe8ef570de676cc1f57fca599283cde28 (the commit for this issue, which would seem to make sense since it's modifying the platform code for the Teensy 3).

The crux of my code is pretty simple for testing, which is:

```
void setup() {
    pinMode(7, OUTPUT);
    digitalWrite(7, HIGH);

    // Add some entropy to initial random FastLED seed
    random16_add_entropy(generate_entropy());

    // Initialize starting patterns and palettes
    currentPattern = random8(patternsLength);
    currentHSVPalette = random8(hsvPalettesLength);
    currentRGBPalette = random8(rgbPalettesLength);

    // For Use With Explicit Clock and Data
    FastLED.addLeds<APA102, DATA_PIN, CLOCK_PIN, RGB_ORDER, DATA_RATE_MHZ(8)>(
        leds, NUM_LEDS).setCorrection(TypicalSMD5050);

    // Set the color temperature
    FastLED.setTemperature(CarbonArc);

    // Set the global brightness
    FastLED.setBrightness(DEFAULT_BRIGHTNESS);
    Serial.begin(9600);
}

void loop() {
    Serial.println("Testing...");
    patterns[currentPattern]();
    FastLED.show(); // Show the LED's
    FastLED.delay(1000 / FPS);  // Add a global delay at the frame rate.

    // Switch patterns every PATTERN_SECS
    EVERY_N_SECONDS(PATTERN_SECS) { nextPattern(); }

    // Switch palettes every PALETTE_SECS
    EVERY_N_SECONDS(PALETTE_SECS) { nextPalette(); }

    // Increment the "rainbow hue" every RAINBOW_MILLIS milli's.
    EVERY_N_MILLISECONDS(RAINBOW_MILLIS) { ++rainbowHue; }
}
```

"Testing..." only gets printed out a handful of times before it fails (on some commits in the bisect, it got through only about 3 iterations).

Anyways, I'm sticking to the 3.1 branch, but thought I should mention this so it doesn't slip away unnoticed. If you have any suggestions or thoughts, I'm all ears (and thanks as always).

focalintent, 2016-08-23:
This is likely in line with what's being tracked/talked about in #339 -- I wonder what's causing the state to lock up here.