FastLED/FastLED #314: Please make Teensy's default hardware SPI 12 Mbit/sec

PaulStoffregen, 2016-06-03:
Many LED strips can't handle 24 MHz speed well.

APA102 in particular causes the clock pulse width to shorten slightly after each LED as it "regenerates" the clock signal.  After 100 to 200 LEDs, probably often occur with 24 MHz speed.

12 MHz would give much better out-of-the-box experience.  It should also cut down on a common support request.  People who really want higher performance could always edit the code to try 24 MHz with their LEDs.

focalintent, 2016-06-03:
It's not the APA102 chip itself - it's something about the strips - i have a 768led APA102 board that I can run at 24Mhz without any problems.  (That said, I thought I did drop the default down to 12Mhz at some point)

PaulStoffregen, 2016-06-04:
Oh, maybe I've got an outdated copy in Teensyduino.

I'm updating now to the lastest.  Is the master branch currently the best for Teensy?  Previously I was pulling from a Teensy-specific branch.

commit 66ae6702 (2016-06-04): Thought I made the APA102 default to 12Mhz already - closes #314
closed 2016-06-04 by focalintent