FastLED/FastLED #315: Can't initialize or use FastLED inside object?

deltaray, 2016-06-09:
I've used FastLED before and can confirm that my physical setup is working by running the Cylon example successfully.

I'm trying to remove delay from my code by making a state machine and using an object to update the lights. I just want to test this by making the lights blink red in succession.  However its not working and according to my serial print statements, its not even initializing the object prior to running the "void setup()" statement.  It compiles and uploads successfully though. So what am I doing wrong?  Below is my code

```
#include "FastLED.h"
#define LED_PIN 7

class Lights {
       unsigned char ledPin;
       unsigned char ledCount;
       unsigned long NextOn;
       CRGB leds[13];
       char ledOn;
       int lightdelay;
 
    public:
    Lights(unsigned char pin, unsigned char ledcount) {
      ledPin = pin;
      pinMode(ledPin, OUTPUT);
      digitalWrite(ledPin, LOW);
      ledCount = ledcount;
      ledOn = 0;
      lightdelay = 100;
      NextOn = 0;
      FastLED.addLeds<NEOPIXEL, LED_PIN>(leds, ledCount); // It complains if I try to use a variable for the 7. It has to be a constant.
      for (unsigned char i = 0; i < ledCount; i++) {
        leds[i] = CRGB::White;
      }
      FastLED.show();
    }
    
    void Update() {
      unsigned long currentMillis = millis();
      
      if (currentMillis > NextOn) {
         if (ledOn >= ledCount) {
           leds[ledOn] = CRGB::Black;
           ledOn = 0;
           leds[ledOn] = CRGB::Red;
         } else if (ledOn == 0) {
           leds[ledCount - 1] = CRGB::Black;
           leds[ledOn] = CRGB::Red; // Need to put RGB to HSV converter in later.
         } else {
           leds[ledOn - 1] = CRGB::Black;
           leds[ledOn] = CRGB::Red; // Need to put RGB to HSV converter in later.
         }
         Serial.print("LED: ");
         Serial.println(ledOn);
         Serial.println(NextOn);
         FastLED.show();
         NextOn = currentMillis + lightdelay;
         ledOn++;
      } 
   }
};

Lights lightstrip1(7, 12);

void setup() {
  Serial.begin(115200);
  Serial.println("Started");
}

void loop() {
  lightstrip1.Update();  
}

```

focalintent, 2016-06-09:
I'd say try something like this:

```
Lights *lightstrip1;

void setup() { 
  Serial.begin(115200);
  Serial.println("Started");
  lightstrip1 = new Lights(7,12);
}

void loop() { 
  lightstrip1->Update();
}
```

Also - you have some problems in your code - like a buffer overflow error:

```
         if (ledOn >= ledCount) {
           leds[ledOn] = CRGB::Black;
           ledOn = 0;
```

Take out that leds[ledOn] = CRGB::Black line - you're writing outside the array there.

Ditto here:

```
         } else if (ledOn == 0) {
           leds[ledCount - 1] = CRGB::Black;
```

you're writing into memory before the start of the array.

Also, you don't need the pinMode/digitalWrite(LOW) for the pin that's used in FastLED.addLeds - that takes care of the pin initialization and setup.

Finally - my recommendation for something like this is to instead initialize FastLED in setup - and have your update object merely take a pointer to the leds array and the count of leds and then otherwise run similarly to how it is.  You don't want to create a situation where you're accidentally calling addLeds multiple times for the same pin (it will likely not work the way that you're expecting it to).

deltaray, 2016-06-13:
That worked. Thanks. And thanks for catching my off by one errors. Doh!

closed 2016-06-13 by deltaray