FastLED/FastLED #318: WeMos D1 R2 - XTENSA-architecture microcontrollers are not supported.

fotoKrille, 2016-06-19:
I tryed this and other options but got an error.
```
#define FASTLED_ESP8266_D1_PIN_ORDER

#include <FastLED.h>
```

```
Board: WeMos D1 mini
Arduino: v1.6.8
Arduino core for ESP8266: v2.2.0
FastLED: V3.1.0

```
```
ERROR: XTENSA-architecture microcontrollers are not supported.
exit status 1
Error compiling for board WeMos D1 R2 & mini.
```

krzychb, 2016-06-20:
Hi @fotoKrille,

This error shows up for FastLED installed with Library Manager.

You need to get the latest version of the library directly from githib - https://github.com/FastLED/FastLED.

Ref. https://github.com/FastLED/FastLED/issues/289, https://github.com/FastLED/FastLED/issues/264

fotoKrille, 2016-06-20:
What is the latest version? I have 3.1.0? Cant find any newer?

closed 2016-06-29 by focalintent