FastLED/FastLED #319: Wiring ESP8266 with WS2801

ToxicF3AR, 2016-06-20:
Anyone know how to do that ?

focalintent, 2016-06-21:
Please ask this on the g+ group - go to http://fastled.io/+ - I'm trying to keep the github issues limited to bug reports and feature requests.  You're far more likely to find someone who has set this up there.

closed 2016-06-21 by focalintent

MFornander, 2016-06-21:
Thanks @focalintent and the FastLED crew. There is so much noise you have to deal with here that are mostly people's hardware issues.  Could we add something to the GitGub ssiue form that asks people to try G+ first if they are unsure if it's software or hardware.  I just hate to see your valuable time being consumed with troubleshooting various setup issues when you could be coding :)