FastLED/FastLED #321: arduino due with dotstar strip breaks in fastled 3.1

brandonburr, 2016-06-23:
Hi, 
the parallel output feature in 3.1 may have a bug when using dotstars and the due, using Arduino 1.6.5.

symptom:   the first 50 or so pixels in a strip work fine, the rest blink haphazardly.

we did a git bisect, and with the same setup and code everything works fine on 3.0.3, but fails on 3.1. the commit seems to be somewhere around here (94210b7858b2d909125ab3b8f1e041bf9202cbe4)

other details:
- we are using this strip https://www.adafruit.com/products/2237
- we tried using 3.3v data/clock straight to the led, and also going through a 3.3 -> 5v step up, and both fail on 3.1 and work on 3.0.3 
- interestingly, we have an older dotstar strip (from about 2 years ago, and 60 led/m strip) that works fine. but the new strips we just got (as of june '16, and 30 led/m) both show the same problem. Only mention of a difference on the adafruit website is that the newer strips use a different color order (BRG now), but doesn't seem like that should be related. 

thanks!

Here's the code we're using:

`#include <FastLED.h>

#define DATA_PIN 3
#define CLOCK_PIN 4
#define NUM_LEDS 150

CRGB leds[NUM_LEDS];

void setup() {
    // Initialize the LED Strip.
    FastLED.addLeds<APA102, DATA_PIN, CLOCK_PIN>(
        leds, NUM_LEDS).setCorrection(TypicalSMD5050);

    // Set the color temperature
    FastLED.setTemperature(CarbonArc);
}

void loop() {
    uint8_t i = 0;
    for (i = 0; i < NUM_LEDS; ++i) {
        leds[i] = CRGB::Black;
    }
    FastLED.show();
    for (i = 0; i < NUM_LEDS; ++i) {
        leds[i] = CRGB::Red;
        FastLED.show();
        delay(100);
    }
}`

focalintent, 2016-06-23:
So the APA102 code doesn't go anywhere near the parallel output.  Can you pull the latest version of the code from github and try that (it would version number at 3.1.1, I'm overdue for doing a release, but have a bunch of loose ends to tie up).  

Also try playing with the data rate:

```
FastLED.addLeds<APA102, DATA_PIN, CLOCK_PIN, RGB, DATA_RATE_MHZ(12)>(leds, NUM_LEDS);
```

Try a variety of different values, but with some APA102 strips they don't deal well with the higher data rates.  (There's also two chips - the APA102 and the APA102C, I don't know if that ends up a difference here - but as far as I've been able to tell the higher clock rate problems have to do with the _strips_ themselves - e.g. I have a 768 led APA102 matrix that works just fine at a data rate of 24Mhz).

focalintent, 2016-06-23:
Also - actually, yeah -  definitely grab the latest code - I fixed a bunch of issues with the bitbang'd spi output recently, which is what you're using here.

brandonburr, 2016-06-23:
thanks! 
ok so we tried on master, but same problem. however, lowering the data rate to 7mhz seemed to do the trick! and yeah, we are using APA102C strips. 

Makes sense that it has to do with the strips, it was just weird that 3.0.3 worked on the same strip while 3.1 didn't. Thank you for your help and your quick response! We love fastled!

focalintent, 2016-06-23:
Part of what happened with the move to 3.1 is I made the bit banging code way faster :)

closed 2016-06-23 by focalintent