FastLED/FastLED #322: Simplex noise is discontinuous

cacktopus, 2016-06-23:
I tried pulling some data from inoise8_raw(x) running on my arduino uno using the following code:

```
void get_data() {
  uint8_t b1 = getSerial();
  uint8_t b0 = getSerial();

  uint16_t val = (((uint16_t) b1) << 8) +
                 (((uint16_t) b0) << 0);

  int8_t fx = inoise8_raw(val);
  Serial.println(fx);
}
```

The graph I get is discontinuous at integer boundaries:

[image]

I compared the above graph to one produced with the "noise" module for python (using `snoise2(x, 0)`):

[image]

I'm wondering:
  - Are the discontinuities are expected?
  - Do I have a bug in my graph generation code?
  - Is this is a bug in the FastLED noise code?

closed 2016-06-23 by cacktopus

kriegsman, 2016-06-23:
Hi- I see you closed the issue (great report by the way, thank you)... Did you figure out what was causing what you were seeing?

marmilicious, 2016-06-23:
I'd also be curious to know if you discovered anything here.