FastLED/FastLED #323: Using FastLED with MPU6050

batata004, 2016-06-27:
Hi, I am having lots of problems using FastLED with MPU6050 on arduino. It's surelly because of the interrupts and because of the really tight timing that FASTLED has to run to accomplish it's goal.

So I have this question: I want my LED stripe to be all red. So I use FASTLED code to make it read and everytime at the bottom of the loop() function I call:

fastled.show();

Do I really need to do this? Can I call this function only once and never more call it? Of course, if I need to change my LED stripe to blue I would call fastled.show(); again, but do I really need to call fastled.show(); everytime inside my loop function even if I dont want to change LEDs color?

kriegsman, 2016-06-27:
You only have to call "FastLED.show()" when you want to _update_ the strip with new data, new colors.  The strip will keep the same color it had before until you call FastLED.show() again with new data.

For general help with FastLED, please use our Google Plus discussion group: http://fastled.io/+
We try to reserve the github tickets specifically for bugs in the library itself; general help questions get handled on the Google Plus group.

closed 2016-06-27 by kriegsman

batata004, 2016-06-27:
@kriegsman I am really sorry for opening this issue here, but I didnt know where to ask this question. Since you know really well this library could you kindly answer this question: after I call FastLED.show() will fastled keep doing any job/routine related to interrupts? If I stay 1 hour without executin "FastLED.show()" will my arduino be completelly free to do any other thing? Or fastled will still be using/firing interrupts (or preventing them from running from time to time) while ".show" is not executed anymore?

kriegsman, 2016-06-27:
FastLED does nothing unless you call it explicitly: no interrupt usage, no background processing, nothing.   Each one time that you call "FastLED.show()", data is sent to the LED strip one time, and that's it.

FastLED does _disable_ interrupts _while_ it is doing FastLED.show() if you are using WS2811-type LED strips, but once it returns to you, interrupts are re-enabled, and work normally.  FastLED does no background processing.

batata004, 2016-06-27:
That's good news @kriegsman  ! Other libraries that I tried kept triggering interrupts in the backgroudn even when it's not being used (like the adafruit library). I removed the .show() from my loop code and called it only once when I set the color of the stripe to red and it worked much smothier! Great library, congrats!

kriegsman, 2016-06-27:
Happy to help, and welcome to the community.  Check out our Google Plus discussion group, and make sure to post pictures / video of your project!  Here's a shortcut to the group: http://fastled.io/+

batata004, 2016-06-27:
Amazing, I will surellly do that as soon as I finish my project :)