FastLED/FastLED #324: very slow frame rate for ESP & APA102

mkeyno, 2016-06-27:
hi @focalintent , I've tried to implement my POV project  with ESP, but frame rate is so slow for 144 APA102 pixel,my idea is to read pixel data from SPIFF file without closing it , but result beyond my expectation very slow even with 160Mhz clock 
following is  the serial output and loop code

i=0 New Image=1.bin duration=4
[frame]1076598
[frame]1043446
[frame]1044003
[frame]1043555
 i=1 New Image=2.bin duration=4
[frame]1057822
[frame]1052909
[frame]1052453
 i=2 New Image=3.bin duration=4
[frame]1067164
[frame]1042669
[frame]1043549
[frame]1042468
 i=3 New Image=4.bin duration=4
[frame]1059210
[frame]1044297
[frame]1044701
[frame]1044550
 i=4 New Image=5.bin duration=4
[frame]1059668
[frame]1042478
[frame]1043678
[frame]1043824
 i=5 New Image= duration=0
FILE NOT OPEN
 i=0 New Image=1.bin duration=4
[frame]1075868
[frame]1044087
[frame]1044103
[frame]1044578
 i=1 New Image=2.bin duration=4
[frame]1059307
[frame]1052474
[frame]1052181
 i=2 New Image=3.bin duration=4
[frame]1067527
[frame]1042757
[frame]1042843
[frame]1043076
 i=3 New Image=4.bin duration=4
[frame]1058051
[frame]1045418
[frame]1046048
[frame]1045034
 i=4 New Image=5.bin duration=4
[frame]1059883
[frame]1043249
[frame]1043334
[frame]1043710
 i=5 New Image= duration=0
FILE NOT OPEN
 i=0 New Image=1.bin duration=4

loop code
void loop() 
{
 
 if(!SHOW)   server.handleClient();  
  webSocket.loop();
 
  if(SHOW)
      {
        if( (millis()- OpenlastTime) > DURATION[image_index]*1000L )
        {
          FILe.close();
          image_index++; 
          if(image_index>IMAGE_NUM)    image_index=0;
          
           Serial.print(F(" i="));  Serial.print(image_index); 
          Serial.print(F(" New Image="));  Serial.print(IMAGES[image_index]);
          Serial.print(F(" duration="));  Serial.println(DURATION[image_index]);
           
           FILe=SPIFFS.open("/SHOW/"+IMAGES[image_index], "r");
          if(!FILe) Serial.println("FILE NOT OPEN");
          OpenlastTime=millis();
        }
        if(!Start_new_round && (micros()-lastLineShow)> lineInterval)
                          {
                          int i=0,j=0;
                          byte rgb=0,r,g,b; 
                          while(FILe.available()&& i< (NUM_LEDS*3))
                                                                    {
                                                                    byte c=FILe.read();
                                                                         if(rgb==0) r=byte(c);
                                                                    else if(rgb==1) g=byte(c);
                                                                    else if(rgb==2) b=byte(c);
                                                                    else           { rgb=0;
                                                                                     leds[j] = CRGB(r,g,b);
                                                                                     j++; 
                                                                                     }
                                                                    rgb++;            
                                                                    i++;
                                                                    }
                          FastLED.show(); lastLineShow=micros();                          
                          Current_imageLine++;                                        
                          }
       FastLED.show(); lastLineShow=micros(); }
          if(Current_imageLine >= IMAGES_LINES)   
          { 
            Serial.print("[frame]"); Serial.println((micros()-lastFrameShow));
            Current_imageLine = 0;
            lastFrameShow=micros(); 
            Start_new_round=false; 
            FILe.close();
            FILe=SPIFFS.open("/SHOW/"+IMAGES[image_index], "r");
           } 
         
      }
}
`void loop() 
{
 
 if(!SHOW)   server.handleClient();  
  webSocket.loop();
 
  if(SHOW)
      {
        if( (millis()- OpenlastTime) > DURATION[image_index]*1000L )
        {
          FILe.close();
          image_index++; 
          if(image_index>IMAGE_NUM)    image_index=0;
          
           Serial.print(F(" i="));  Serial.print(image_index); 
          Serial.print(F(" New Image="));  Serial.print(IMAGES[image_index]);
          Serial.print(F(" duration="));  Serial.println(DURATION[image_index]);
           
           FILe=SPIFFS.open("/SHOW/"+IMAGES[image_index], "r");
          if(!FILe) Serial.println("FILE NOT OPEN");
          OpenlastTime=millis();
        }
        if(!Start_new_round && (micros()-lastLineShow)> lineInterval)
                          {
                          int i=0,j=0;
                          byte rgb=0,r,g,b; 
                          while(FILe.available()&& i< (NUM_LEDS*3))
                                                                    {
                                                                    byte c=FILe.read();
                                                                         if(rgb==0) r=byte(c);
                                                                    else if(rgb==1) g=byte(c);
                                                                    else if(rgb==2) b=byte(c);
                                                                    else           { rgb=0;
                                                                                     leds[j] = CRGB(r,g,b);
                                                                                     j++; 
                                                                                     }
                                                                    rgb++;            
                                                                    i++;
                                                                    }
                          FastLED.show(); lastLineShow=micros();                          
                          Current_imageLine++;                                        
                          }
       FastLED.show(); lastLineShow=micros(); }
          if(Current_imageLine >= IMAGES_LINES)   
          { 
            Serial.print("[frame]"); Serial.println((micros()-lastFrameShow));
            Current_imageLine = 0;
            lastFrameShow=micros(); 
            Start_new_round=false; 
            FILe.close();
            FILe=SPIFFS.open("/SHOW/"+IMAGES[image_index], "r");
           } 
         
      }
}`

kriegsman, 2016-06-28:
Hi, @mkeyno.  Looks like it'll be a cool project once you get the I/O speed worked out.

Could you take this question over to our Google Plus discussion group?  Here's a link: http://fastled.io/+
We like to reserve github tickets for actual bugs in the FastLED library itself, and we use the Google Plus discussion group for general troubleshooting and performance help.

focalintent, 2016-06-28:
Ok, a couple of things - first of all, you're calling FastLED.show() _way_ too often.  Instead of just calling it every time you read a new line in, you're reading it every time through the loop (or you pasted your code in badly).  

The second is the way that you're reading data in, one byte at a time, is going to be wildly inefficient - you can just read the bytes directly into the CRGB array:

```
  if(!Start_new_round && (micros()-lastLineShow)> lineInterval) {
    int i = 0; 

    while( (i < NUM_LEDS*3) && FIle.available()) { 
      i += FIle.readBytes( ((byte*)leds) + i, (NUM_LEDS*3)-i);
    }
    FastLED.show();
    lastLineShow(micros());
  } 
```

Also, to get a better feel for your timing numbers, it'd be useful to know how many _lines_ of output are being written in the ~1s/"frame" that you're running.  

Also, instead of closing/re-opening the file, instead use the seek method to jump back to the beginning of it - `FIle.seek(0, SeekSet);`

kriegsman, 2016-06-28:
Thanks, Dan.
Yeah, I think this is a "performance optimization" conversation; let's take it over to Google Plus.  I'm closing the ticket here.  

Shortcut to G+ group: http://fastled.io/+

closed 2016-06-28 by kriegsman

mkeyno, 2016-06-28:
thanks @kriegsman for your concern , but believe me I've never had any good experience to get the right technical answer from public forum like G+ and thanks very much to github which is help me and direct me to correct way very fast. most of time my topic get lost in place like google group and G+, however I will post it there either

thanks dear @focalintent  for your swift reply , I might add to read image in size of 300 line * 144 pixel * 3 byte ( about 127 Kbyte) , I  open the file  then read one by one and add it to LED array and then show it as soon as fill one 144*3 chuck of  bytes  , there is three condition in my loop along with wifi service websocket ( other service like webserver disable in case of show action )

the first condition check duration show ,whether is time for close old file and open the new image file
` if( (millis()- OpenlastTime) > DURATION[image_index]*1000L )
        {
          FILe.close();
          image_index++; 
          if(image_index>IMAGE_NUM)    image_index=0;
           FILe=SPIFFS.open("/SHOW/"+IMAGES[image_index], "r");
          if(!FILe) Serial.println("FILE NOT OPEN");
          OpenlastTime=millis();
        }`

the second condition check new revolution is start and time to show new line
`if(Start_new_round && (micros()-lastLineShow)> lineInterval)`
read in air byte and put in the array
`byte c=FILe.read();
if(rgb==0) r=byte(c);
else if(rgb==1) g=byte(c);
else if(rgb==2) b=byte(c);
else   
{
  rgb=0;
  leds[j] = CRGB(r,g,b);
  j++; 
}
rgb++;            
i++;
}
FastLED.show(); 
lastLineShow=micros();                         
Current_imageLine++;  `
 the second condition check  whether all line of one frame has been fed or not  , in case of reading all line , I close the file and reopen it so the file position get back to beginning the file 
`Current_imageLine = 0;
 lastFrameShow=micros(); 
 Start_new_round=false; 
 FILe.close();
 FILe=SPIFFS.open("/SHOW/"+IMAGES[image_index], "r");`
I though it would be fastest way to read byte one by one instead of read one line and then fed to the LED array

focalintent, 2016-06-28:
As mark said - we prefer for discussions and requests for _help_ to be kept to the g+ group - not here.  (Its even worse when people hijack a ticket for one thing with a request for help with something entirely different)

The code you wrote to read a single byte and put it into an array means that for _every_ byte you are doing multiple function calls, at least 5 comparisons, etc... That's a lot of wasted time there - calling readBytes allows the underlying code to do more interesting things with larger transfers not to mention completely bypassing the chain of extra comparisons you have in there. 

If you want to discuss this more, please take it to the g+ group.  I'm starting to lock issues/threads that either shouldn't be here or get derailed off topic.