FastLED/FastLED #329: APA102 sticking in SPI mode

hydronics2, 2016-07-26:
Hello, any ideas would be much appreciated!

I tried to drive 4 strips of LEDs from a teensy3.2 using 4 SPI pins as suggested here... [https://github.com/FastLED/FastLED/wiki/SPI-Hardware-or-Bit-banging](url)

It works for a few seconds and then freezes.  It works all day long if I un-comment //#define FASTLED_FORCE_SOFTWARE_SPI

  FastLED.addLeds<APA102, 7, 14, RGB>(colors1, NUM_LEDS);
  FastLED.addLeds<APA102, 7, 13, RGB>(colors2, NUM_LEDS);
  FastLED.addLeds<APA102, 11, 13, RGB>(colors3, NUM_LEDS);
  FastLED.addLeds<APA102, 11, 14, RGB>(colors4, NUM_LEDS);

I tried different teensy speeds and different strip speeds to no avail.

I'm using teensy3.2, teensyduino 1.29, arduino IDE 1.6.9, octoshield, the latest fastLED.h, windows 7.

```
//#define FASTLED_FORCE_SOFTWARE_SPI
#include "FastLED.h"

// How many leds in your strip?
#define NUM_LEDS 32

// Define the array of leds
CRGB colors1[NUM_LEDS];
//CRGB colors2[NUM_LEDS];
//CRGB colors3[NUM_LEDS];
//CRGB colors4[NUM_LEDS];

void setup() { 

   //FastLED.addLeds<APA102, RGB>(leds, NUM_LEDS); //any and all combinations cause the program to freeze after a few seconds

   FastLED.addLeds<APA102, 7,14, RGB>(colors1, NUM_LEDS);
   //FastLED.addLeds<APA102, 7,13, RGB>(colors2, NUM_LEDS);
   //FastLED.addLeds<APA102, 11,13, RGB>(colors3, NUM_LEDS);
   //FastLED.addLeds<APA102, 11,14, RGB>(colors4, NUM_LEDS);

}

void loop() { 
  // Turn the LED on, then pause
  for(int i=0; i<NUM_LEDS; i++){
  colors1[i] = CRGB::Red;
  }
  FastLED.show();
  
  delay(100);

  // Now turn the LED off, then pause
  for(int i=0; i<NUM_LEDS; i++){
  colors1[i] = CRGB::Black;
  }
  FastLED.show();
  
  delay(100);
}
```

hydronics2, 2016-07-26:
thanks for the quick reply!  The IDE errors when I try the latest FastLed library. I'm posting to PJRC.com. Not sure what version is packaged with teensyduino right now.

```
C:\Arduino\libraries\FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.001

 #    pragma message "FastLED version 3.001.001"
                     ^
In file included from D:\test.ino:2:0:

C:\Arduino\libraries\FastLED/FastLED.h: In static member function 'static CLEDController& CFastLED::addLeds(CRGB*, int, int)':

C:\Arduino\libraries\FastLED/FastLED.h:437:47: error: expected type-specifier before 'SixteenWayInlineBlockClocklessCOntroller'

         case WS2813_PORTC: return addLeds(new SixteenWayInlineBlockClocklessCOntroller<NUM_LANES, NS(320), NS(320), NS(640), RGB_ORDER, 0, false, 300>(), data, nLedsOrOffset, nLedsIfOffset);

                                               ^

C:\Arduino\libraries\FastLED/FastLED.h:437:152: error: expected primary-expression before ')' token

         case WS2813_PORTC: return addLeds(new SixteenWayInlineBlockClocklessCOntroller<NUM_LANES, NS(320), NS(320), NS(640), RGB_ORDER, 0, false, 300>(), data, nLedsOrOffset, nLedsIfOffset);
```

                                                                                                                                                        ^

focalintent, 2016-07-26:
I checked in a fix for the compile error last night.

closed 2016-08-16 by focalintent