FastLED/FastLED #331: Glitch with more than 189 WS2813

alex-Arc, 2016-08-01:
When running more than 189 WS2813 LED's  there is some none periodic glitches
tested on a Teensy LC with this code:
`#include <FastLED.h>

#define LED_PIN     2
#define NUM_LEDS    288
#define BRIGHTNESS  255
#define LED_TYPE    WS2813
#define COLOR_ORDER GRB
CRGB leds[NUM_LEDS];

CRGBPalette16 currentPalette;
TBlendType    currentBlending;

extern CRGBPalette16 myRedWhiteBluePalette;
extern const TProgmemPalette16 myRedWhiteBluePalette_p PROGMEM;

void setup() {
    delay( 3000 ); // power-up safety delay
    FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
    FastLED.setBrightness(  BRIGHTNESS );
    
    currentPalette = RainbowColors_p;
    currentBlending = LINEARBLEND;
}

void loop()
{
    
    static uint8_t startIndex = 0;
    startIndex = startIndex + 1; /* motion speed */
    
    FillLEDsFromPaletteColors( startIndex);
    
    FastLED.show();
    FastLED.delay(10);
}

void FillLEDsFromPaletteColors( uint8_t colorIndex)
{
    uint8_t brightness = 255;
    
    for( int i = 0; i < NUM_LEDS; i++) {
        leds[i] = ColorFromPalette( currentPalette, colorIndex, brightness, currentBlending);
        colorIndex += 3;
    }
}`

focalintent, 2016-08-01:
Can you describe what the glitching is that you're seeing?   where on the strip is it, etc?

Can you also try adding the line:
```
#define FASTLED_ALLOW_INTERRUPTS 0
```

before the `#include <FastLED.h>` and let me know if that helps the glitching (helps me narrow down where the problem may be)

alex-Arc, 2016-08-01:
it is mostly in the end but so sporadic and infrequent i'm having trouble determining anything. the glitch is a random collection of pixels flashing white. i will try the `FASTLED_ALLOW_INTERRUPTS`

alex-Arc, 2016-08-01:
here is a video of the glitch [https://www.dropbox.com/sh/dwi6282wusguqb3/AACdmLFwcvFml47-qdmI1Wk-a?dl=0](url)

alex-Arc, 2016-08-01:
`#define FASTLED_ALLOW_INTERRUPTS 0` 'do not solve it :(

alex-Arc, 2016-08-16:
Look like it is a power problem, adding power to the other end and haven't seen anything glitch yet

closed 2016-08-16 by focalintent