FastLED/FastLED #338: ESP8266 + APA102: Invalid Pin specifed

sebastius, 2016-08-14:
Trying to hook up APA102 LEDs on GPIO 2 and GPIO12

`#define DATA_PIN    2
#define CLK_PIN   12
#define LED_TYPE    APA102

void setup() {
  FastLED.addLeds<LED_TYPE,DATA_PIN,COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);
`

I get the following error:

`In file included from /var/folders/hq/p5tk9qkd4tj75ykjv97p607r0000gn/T/arduino_modified_sketch_508062/DemoReel100.ino:2:0:
/Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.001
 #    pragma message "FastLED version 3.001.001"
                     ^
In file included from /Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/FastLED.h:65:0,
                 from /var/folders/hq/p5tk9qkd4tj75ykjv97p607r0000gn/T/arduino_modified_sketch_508062/DemoReel100.ino:2:
/Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
In file included from /Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/FastLED.h:48:0,
                 from /var/folders/hq/p5tk9qkd4tj75ykjv97p607r0000gn/T/arduino_modified_sketch_508062/DemoReel100.ino:2:
/Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/fastpin.h: In instantiation of 'class FastPin<66u>':
/Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/fastspi_bitbang.h:23:50:   required from 'class AVRSoftwareSPIOutput<2u, 66u, 6u>'
/Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/fastspi.h:30:7:   required from 'class SPIOutput<2u, 66u, 6u>'
/Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/chipsets.h:94:6:   required from 'class LPD8806Controller<2u, 66u, (EOrder)10u, 6u>'
/Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/FastLED.h:233:66:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = (ESPIChipsets)5u; unsigned char DATA_PIN = 2u; unsigned char CLOCK_PIN = 66u]'
/var/folders/hq/p5tk9qkd4tj75ykjv97p607r0000gn/T/arduino_modified_sketch_508062/DemoReel100.ino:33:64:   required from here
/Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/fastpin.h:206:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
  ^
/Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/fastpin.h: In instantiation of 'static void FastPin<PIN>::fastset(FastPin<PIN>::port_ptr_t, port_t) [with unsigned char PIN = 66u; FastPin<PIN>::port_ptr_t = volatile unsigned int*; RwReg = volatile unsigned int; FastPin<PIN>::port_t = volatile unsigned int]':
/Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/fastspi_bitbang.h:158:49:   required from 'static void AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::writeBit(uint8_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::clock_ptr_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::data_ptr_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::data_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::data_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::clock_t, AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::clock_t) [with unsigned char BIT = 0u; unsigned char DATA_PIN = 2u; unsigned char CLOCK_PIN = 66u; unsigned char SPI_SPEED = 3u; uint8_t = unsigned char; AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::clock_ptr_t = volatile unsigned int*; AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::data_ptr_t = volatile unsigned int*; AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::data_t = unsigned int; AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::clock_t = volatile unsigned int]'
/Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/fastspi_bitbang.h:327:14:   required from 'void AVRSoftwareSPIOutput<DATA_PIN, CLOCK_PIN, SPI_SPEED>::writePixels(PixelController<RGB_ORDER>) [with unsigned char FLAGS = 0u; D = DATA_NOP; EOrder RGB_ORDER = (EOrder)10u; unsigned char DATA_PIN = 2u; unsigned char CLOCK_PIN = 66u; unsigned char SPI_SPEED = 3u]'
/Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/chipsets.h:138:3:   required from 'void WS2801Controller<DATA_PIN, CLOCK_PIN, RGB_ORDER, SPI_SPEED>::showPixels(PixelController<RGB_ORDER>&) [with unsigned char DATA_PIN = 2u; unsigned char CLOCK_PIN = 66u; EOrder RGB_ORDER = (EOrder)10u; unsigned char SPI_SPEED = 3u]'
/var/folders/hq/p5tk9qkd4tj75ykjv97p607r0000gn/T/arduino_modified_sketch_508062/DemoReel100.ino:126:1:   required from here
/Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/fastpin.h:228:21: warning: unused parameter 'port' [-Wunused-parameter]
  inline static void fastset(register port_ptr_t port, register port_t val) __attribute__ ((always_inline)) { }
                     ^
/Users/sebastianoort/Documents/Arduino/libraries/FastLED-master/fastpin.h:228:21: warning: unused parameter 'val' [-Wunused-parameter]
Bibliotheek FastLED-master op versie 3.1.1 in map: /Users/sebastianoort/Documents/Arduino/libraries/FastLED-master  wordt gebruikt
exit status 1
Fout bij het compileren van board NodeMCU 1.0 (ESP-12E Module)
`

Also happens on any other pin i specify. Am i doing something wrong?

mkeyno, 2016-08-14:
have you choose the proper board ? cause if you select any generic ESP board it will not throw such error  , although , not using pre compiler ' the raw pin order ' the selected pin actually 4 and undefined  pin

sebastius, 2016-08-14:
I have tried 'generic ESP', NodeMCU and several others. All of them throw the same error.

focalintent, 2016-08-14:
What version of the arduino ide and what version of the arduino esp8266 platform are you using?

sebastius, 2016-08-14:
Latest Arduino .10 and latest release ESP.

focalintent, 2016-08-15:
Ah - you had a bad code format tag so I couldn't see your code - you aren't specifying the clock pin in your add leds line - it should be:

```
FastLED.addLeds<LED_TYPE,DATA_PIN,CLK_PIN,COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);
```

closed 2016-08-15 by focalintent

sebastius, 2016-08-16:
Damn, so it was a PEBCAK... So sorry!