FastLED/FastLED #339: Teensy31 stops running APA102 strip

bryanculbertson, 2016-08-18:
My Teensy 3.1 stops running after a couple minutes of running a single APA102 strip. This always happens after a couple minutes, but not at exactly the same time. This smells of a memory leak.

In case it matters, I am using data pin 7 and clock pin 14. I have the lastest version of FastLED (4c32e652ce) downloaded via PlatformIO.

Let me know if you there is other information that would be helpful.

focalintent, 2016-08-18:
Seeing the code you're trying to run would be helpful for attempting to recreate the problem here.

bryanculbertson, 2016-08-18:
It doesn't matter what code is running, as long as it calls FastLED.show() lots of times. I think it crashes sooner the higher NUM_LEDS is, but I didn't measure it it enough times to be sure it wasn't just coincidental.

```
#include <FastLED.h>

#define LED_PIN     7
#define CLOCK_PIN   14
#define COLOR_ORDER BGR
#define CHIPSET     APA102
#define NUM_LEDS    300

#define BRIGHTNESS  200

CRGB leds[NUM_LEDS];

void setup() {
  delay(3000);

  FastLED.addLeds<CHIPSET, LED_PIN, CLOCK_PIN, COLOR_ORDER>(leds, NUM_LEDS);
  FastLED.setBrightness(BRIGHTNESS);

  for (int i=0; i<NUM_LEDS; ++i) {
    leds[i] = CRGB::Black;
  }
}

void loop() {
  for (int i=0; i<NUM_LEDS; ++i) {
    leds[i] = CRGB::Red;
  }
  FastLED.show();
  delay(500);

  for (int i=0; i<NUM_LEDS; ++i) {
    leds[i] = CRGB::Green;
  }
  FastLED.show();
  delay(500);

  for (int i=0; i<NUM_LEDS; ++i) {
    leds[i] = CRGB::Blue;
  }
  FastLED.show();
  delay(500);
}
```

reloc4, 2016-08-19:
I have got similar problems with my teensy 3.2. I have found a workaround by disabling USB. ( Setting usb Type to " no USB" in the Menu. )
Disconnecting USB seems to help too, but eventually the Program stoops as well.

marmilicious, 2016-08-23:
I'm experiencing something similar with things "locking up" so I thought I'd share what I'm seeing here too.  I'm using a Teensy 3.0 and LPD8806 strip (32 pixels).  This was also happening for me with an APA102 strip (38 pixels).  This evening was testing with a slightly modified version of the FastLED cylon example, http://pastebin.com/wRX2U00B but this "locking up" happens to me with other sketches also.

I did some tests experimenting with how often FastLED.show() is called and it seemed like things would "lock up" sooner when it was called very often.  But it's not consistent for me as to how quickly it locks up.  For example if I call show() every 15 milliseconds it might lockup after 20 seconds, or it might run for several minutes.  If there wasn't any delay at all though, and show() was called every single loop, then things would usually just stop running immediately right after sketch upload.  I could watch the Teensy's onboard led blink from the update, then maybe the first few pixels might light up, and then it would be locked up.

After I upload a sketch and it locks up, if I unplug the MCU from the computer and just power it with a battery it seems to run fine without ever "locking up".   I also tried reloc4's suggestion of setting USB type to "no USB" and that seemed to work fine (but I didn't let it run for a really long time).

I don't think I ever experienced this "locking up" in the past with Arduino IDE 1.6.8, but I think this was happening with 1.6.9.  Can't remember if I ever tried out 1.6.10.   Currently I have the new 1.6.11 installed and am testing out Teensyduino 1.30 beta2.

focalintent, 2016-08-23:
Hmmm - I wonder if there's some conflict happening between the USB peripheral and the SPI peripheral?  Out of curiosity - what happens if you put `#define FASTLED_FORCE_SOFTWARE_SPI 1` before the `#include <FastLED.h>`?  

If the USB peripheral is leaving one of the status registers that are being checked by the SPI code in a weird state (or, if I'm checking the SPI status register incorrectly) that could cause things to lock up.

marmilicious, 2016-08-23:
Adding `#define FASTLED_FORCE_SOFTWARE_SPI 1` above the FastLED.h include does seem to fix the locking up, but (unexpectedly!) it also made all the pixels display as nearly white (as if saturation was set to a very low value).

focalintent, 2016-08-23:
Huh - that's a particularly strange reaction - what types of LEDs are you driving?  (If they're something like WS2801's, the software SPI on the teensy 3.x may be driving things too fast!)

focalintent, 2016-08-23:
(Also, it feels like a thing that happens when clock/data pins are backwards)

marmilicious, 2016-08-23:
My setups use Data on pin 11, and Clock on 13.  Haven't changed anything there.

With Teensy 3.0 and LPD8806 all pixels went nearly white with FASTLED_FORCE_SOFTWARE_SPI 1.  Without force software spi colors are correct, but then getting the lockup.

Just tried that same Teensy 3.0 with an APA102 strip and that is running fine color wise and not locking up when using FASTLED_FORCE_SOFTWARE_SPI 1 (and powered via the USB from computer).

And now I tested with _a different Teensy 3.0 and a different_ LPD8806 strip (32 pixels) on a _different USB plug_ (thus a different COM port) and it locks up without the FASTLED_FORCE_SOFTWARE_SPI 1.  If I add force software spi it's not locking up, but I do see odd flickering, even more so in the first few pixels.  I just tried unplugged the USB and powering it with wall power supply and still seeing some odd flickering in some pixels.  So that doesn't seem like it's USB related then.  Meh, this is totally confusing me.

I'm going to uninstall arduino IDE and go back to 1.6.8 again.  Thank you for your help and suggestions Daniel.

focalintent, 2016-08-23:
(Also - make sure that you are not using the FastLED version that comes w/teensyduino - I've done a bunch of fixes since then)

marmilicious, 2016-08-23:
Yes, I never install FastLED that way, always grab the the latest here at github.  :)

dimitre, 2016-08-28:
I'm using Teensy 3.2 and APA102 and have freezing issues too, both with Teensyduino and Github latest version. It seems to be USB related because it runs well on separate 5V font.
this is the minimal code to reproduce here:
```c++
#include "FastLED.h"
#define NUM_LEDS 80
#define DATA_PIN 7
#define CLOCK_PIN 14
CRGB leds[NUM_LEDS];

void setup() {
  LEDS.addLeds<APA102, DATA_PIN,  CLOCK_PIN, BGR>(leds, NUM_LEDS);
  LEDS.setBrightness(65); //100
}

void loop()
{
  float r =  sin(millis() / 30.0) * 127.0 + 127;
  for (int a = 0; a < NUM_LEDS; a++) {
    leds[a] = CRGB(0, r, 255 - r);
  }
  FastLED.show();
}
```
Thanks

reloc4, 2016-08-28:
Seems like you don't need to change the colour at all and the number of calls till the freeze is different every time. I have measured 176 to up to 5700 calls with is code:

```
#include "FastLED.h"
#define NUM_LEDS 144
CRGB leds[NUM_LEDS];

void setup() {
  delay(2000);
  FastLED.addLeds<APA102, BGR>(leds, NUM_LEDS);
  pinMode(7, OUTPUT);
  digitalWrite(7, HIGH);
  FastLED.setBrightness(2.55 * 1.0); // 1% Helligkeit
  FastLED.setMaxPowerInVoltsAndMilliamps(5,2000); 
  Serial.begin(9600);
}

void loop()
{
      FastLED.show();
      static long d = 0;
      d++;
      Serial.println(d);
      //delay(5);
}
```

marmilicious, 2016-08-28:
Just to follow up, I reverted back to using Arduino IDE 1.6.8 and Teensyduino 1.28 (which I don't believe I ever had any troubles with in the past) but I'm still getting the locking up.
Since the locking up only happens when the controller is plugged into USB, I wondered if something had gone wacky with my USB drivers, so I uninstalled all USB drivers, but still no luck solving this.  I'm not sure how to continue to try to figure this out, or what to test?

reloc4, 2016-08-28:
I just reverted back to fast LED Version "12 Apr " now I have got flickering LED's but no freeze.

dimitre, 2016-08-28:
In a complete setup I'm using an ethernet port too (which proves itself not being the reason for the freeze) but strangely the ethernet chip heats up really fast when FastLED freezes, using USB.
It uses pins 10, 11, 12, 13 of the teensy 3.2.
Thanks

reloc4, 2016-08-28:
I have just checked the code for the waitFully() in platforms/arm/k20/fastspi_arm_k20.h suggesed by @woundedlion in issue  #313 
this is freezing as well.

```
static void waitFully() __attribute__((always_inline)) {
    while ((SPIX.SR & 0xF000) > 0) { // SPI_SR_TXCTR
        while (!(SPIX.SR & SPI_SR_TCF));
        SPIX.SR |= SPI_SR_TCF;
    }
    while (!(SPIX.SR & SPI_SR_TCF));
    SPIX.SR |= SPI_SR_TCF;

    while ((SPIX.SR & 0xF0) > 0) { // SPI_SR_RXCTR
        SPIX.POPR;
		}
	}
```

As a work around you can go back to the old waitFully(): in platforms/arm/k20/fastspi_arm_k20.h

```
static void waitFully() __attribute__((always_inline)) {
		while( (SPIX.SR & 0xF000) > 0);
		// wait for the TCF flag to get set
		while (!(SPIX.SR & SPI_SR_TCF));
		SPIX.SR |= (SPI_SR_TCF | SPI_SR_EOQF);
	}
```

but you may get errors as described in issue #313

focalintent, 2016-08-30:
Ok - I'm thinking there's something going on here with this status register when the USB port is also being used.  I don't know if the USB port is using the same port or what's going on - I think I need to revisit the patch from #313 and see if I can come up with a better way to do it.  (The problem with the old code is that it returns too soon.  This isn't a problem when you're driving, say, the APA102 and pushing out bytes as fast as you can, but when you have something that's pushing out data slowly, like say the WS2801, we end up releasing the SPI hardware before we've finished writing out the last few bytes of data....  whoops :) 

i'm going to read through the data sheet a bit tonight and tighten up my understanding of what's going on with these registers and will try to get a test rig up and running over the next few days (and then I go on vacation for a week where I will be mostly offline, but hopefully I can nail this down in the next couple of days)

focalintent, 2016-08-30:
Yeah - there's a race condition in here - if interrupts fire at just the right point, the TCF flag will get cleared too late, after the last byte has already been written out - which will cause the loop to spin forever waiting for TCF to get set...

focalintent, 2016-08-30:
If someone wants to try the bugs/339 branch mentioned above and let me know if it works - otherwise i'll try it against my own hardware setup in the next couple of days.

reloc4, 2016-08-30:
I have tested the bugs/339 branch (fe32cf1320bd25e5094693f8e3ab4ec6e064d3b3) with Teensy 3.2 and APA102 seems to work.

marmilicious, 2016-08-30:
Just tried bugs/339 branch (fe32cf1) with a Teensy 3.0 and LPD8806 and it's
freezing on me.  Using IDE 1.68 and Teenysduino 1.28.

focalintent, 2016-08-30:
Ok - good to know, it was a long shot of a fix, but something I could try before getting a chance to set up a full set of hardware.  (I don't want to roll back the fix for #313 - I'd rather have both issues fixed)

reloc4, 2016-08-30:
I was using using IDE 1.6.8 and Teenysduino **1.29**  if there is anything else to try I can give it a shot.

hydronics2, 2016-08-31:
I uploaded using the 339 branch above. I'm using Arduino 1.69, Teensyduino 1.29 and Teensy3.2 and working fine for a couple of hours now.

I'm running 4 strips of about 600 APA102 pixels in SPI mode using an assortment of hardware SPI pins (7,11,13,14) without freezing. thanks!

PaulStoffregen, 2016-09-05:
I'm preparing to post Teensyduino 1.30-beta4, and (hopefully very soon) turn this into a stable 1.30 release.

I've pulled in the 339 branch for 1.30-beta4.  Please let me know if this is ok for a final 1.30 release.  Would be nice to have a good, updated FastLED with the installer.

hydronics2, 2016-09-06:
still working well for me.

I'm having to run the teensy at 16mhz to eliminate flickering but I'm
guessing that's an unrelated issue(?).

dimitre, 2016-09-07:
I can confirm APA102, Teensy 3.2 running OK with Teensyduino 1.30 final (339 branch)

jhps, 2016-11-30:
Also having this problem on a APA102s on Teensy 3.1, with Teensyduino 1.31 and also with FastLED d99001d, but not with b01f941 as per https://forum.pjrc.com/threads/35684-FastLED-freezing-with-prop-shield.

jhps, 2016-12-04:
More or less undoing fe32cf1 "attempt a fix for #339 - guard waitfully against a race condition ..." seems to fix it for me. I do not know what the theory of that commit was, or how to test the problem it addressed.

Karniv4, 2016-12-05:
No real help here, just confirming removing the fix solved the hanging problem for me. Raspberry Pi 3 pushing tpm2 to the Teensy 3.2 over usb serial (Using Pixelcontroller).

Plasmarobo, 2017-01-27:
I'm definitely having this issue with master driving APA102s via SPI with a Teensy 3.2.
[b01f941](https://github.com/FastLED/FastLED/commit/b01f941d0964eaf486f4d050e74ae6a65a5816d7) corrects it for me, but none of the other attempts at fixes do.
My USB is still in serial mode, and has no effect on the issue either way.
I am also read a lot of data over I2C as well. Other commits seem to destabilize the I2C hardware (sometimes I2C locks up, sometimes SPI locks up).

alvin666666, 2017-04-20:
we are the led strip custom factory,our sk9822 almost can replace apa102 led strip and more cheaper.
Maybe you can change other controller,like t1000s sp105e or others..
Email:Alvin@witop-tech.com

florianlink, 2017-05-14:
The current master branch fixes my hangs that I experienced with FastLED + FFT + Serial1 on Teensy 3.2, see the bug mentioned above!

PaulStoffregen, 2017-05-15:
> see the bug mentioned above!

Could you please be more specific.  I do not see any code above using Serial1.  Did I miss it?

Just so you know, I ignore any bug report which does not have a complete program I can copy into Arduino and run on a board.

florianlink, 2017-05-15:
It was not a bug report, I just said, that 

https://github.com/FastLED/FastLED/issues/419

fixes the problems described in this bug, so I think this bug can be closed.
Before trying the current master of FastLED, I also tried your 339 fix-branch and it did not fix my hangs.
But the current master, which includes the fix for 419 fixes the hangs.

hydronics2, 2017-06-23:
The current version of Teensyduino 1.37 that includes a newer version of FastLed works for me..! No locking up.

I have 600+ APA102 LEDs and I still have to run the LEDs at DATA_RATE_MHZ(12) and run the teensy at 16MHZ to completely remove the flickering.

`FastLED.addLeds<LED_TYPE,DATA_PIN1,CLOCK_PIN1,COLOR_ORDER, DATA_RATE_MHZ(12)>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);`

closed 2019-01-20 by focalintent