FastLED/FastLED #341: SK9822 strip red flickering

kasperkamperman, 2016-09-03:
- Arduino 1.6.9 / TeensyLoader 1.29 / Teensy 3.0
- FastLed bug #339 branch (I thought it might be safer to test with this one). 

I'm doing a test with a 5m strip of SK9822 leds and the FirstLight example. They work perfect when declaring them as APA102:
`fastLED.addLeds<APA102, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS);`
However when declaring them as SK9822:
`FastLED.addLeds<SK9822, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS)` the strip starts flickering red after led 60-70 (in total i have 160 leds) and after a while also green (more at the end of the strip). 

The white dot also doesn't reach the end of the strip. 

Video: https://www.youtube.com/watch?v=MFps7mLp-aw

I think it's not by accident an APA102 strip, because the led looks different (chip inside the APA102 is a bit bigger then in the SK9822). I bought them at http://www.szledcolor.com/, which seems a reliable manufacturer regarding other users in the G+ FastLED group.

focalintent, 2016-09-08:
This is most likely because APA102 defaults to DATA_RATE_MHZ(12) while SK9822 defaults to DATA_RATE_MHZ(24) - try setting DATA_RATE_MHZ(12) for the SK9822 line to see if that takes care of the corruption (I wouldn't be surprised if some SK9822 strips have similar issues to the APA102 at higher data rates).

closed 2016-09-08 by focalintent

kasperkamperman, 2016-09-09:
Yes, this worked. Do you know what could be the issue? The strips or the led itself? I assumed szledcolor.com was a reliable supplier.

focalintent, 2016-09-09:
This has been a long running issue with apa102's - and it has something to do with how the strips are laid out.  I can run a 768 led apa102 board at 24mhz (and higher!) but with strips, the longer the strip the lower the max clock rate I can use.