FastLED/FastLED #344: Best chip for FastLed

atomicus, 2016-09-13:
Hi!

It's not an issue, just my question regarding a choice of IC chip to control leds.

Originally I was planning to controll leds directly by Arduino PWM's->FET->LED, but while working on color correction and lightning intensity issues I found out this library.

Digging a bit I found out I can use any of ws**** chip or other to drive my FET's, while saving some PWM outputs for other stuff. 

What I want to ask is, what would be the best chip to put into my design in terms of:

-best support by FastLEd
-possibly 12bits or higher per channel resolution (docs of WS do not state it clearly, or I'm blind).
-support for 4'rd channel (used as white, but I need sometimes to dim it, and run all at once (RGB+W)

If anyone could give me a hint, I would be really gratefull.

BTW - Kudos for this code. 
It's great to a point I need to really think what it does and where, which does not happen often.

BTW - WS**** docs really need someone who can write in english.

focalintent, 2016-09-13:
Right now only 8-bit RGB chipsets are supported (in in the middle of a major rewrite to add support for RGBW as well as 12/16-bit chipsets - but life has been getting in the way of getting that out the door.

atomicus, 2016-09-13:
Ok, thanks. Just read other issues about RGB+W, that's not a problem. 
Still, which IC would you recommend for now, assuming 8bit only and no W channel?

electrokean, 2016-09-13:
For PWM driver chips, the WS2811 or LPD8806 are both pretty readily available, the latter giving you twice as many channels and higher data rate.  Datasheets for both can be found at Adafruit and many other places.

closed 2016-10-30 by focalintent