FastLED/FastLED #345: Problem with IC SM16703P

vecio88, 2016-09-14:
Hi, 
i have a problem with my strep led with IC SM16703P. THis IC not is between the supported chipsets. I have tried to use other chipset but not work well. For example, if i run the example "FirstLight" on the strip , the led turn on in random mode and not with a sequence how is wrote in the code.

how can i solve?

thanks

focalintent, 2016-10-30:
The SM16703P is not a supported chipset, I will have to add support for it at some point in the future.

makenb, 2016-12-05:
+1 for sm16703, please, please, please
Thanks

kdauzickas, 2017-01-25:
+1 seems to be same as ws with a bit different timing

CCERocks, 2017-02-28:
+1 for sm16703

genkagen, 2017-06-01:
+1 for sm16703 also sm16723P(Iout:12mA), thanks!

Agent-3P1C, 2017-10-22:
Has the chip been added since?
If not: are there other libaries that can be used?
Thanks

focalintent, 2017-10-22:
It has not been added yet - I don’t know if there are other libraries for it or not.

makenb, 2017-12-21:
SM16703 is supposed to be compatible with ws2811 which is supported, how ever has slightly different timing requirements. here is a link to a discussion about the differences that may be helpful: 
[http://falconchristmas.com/forum/index.php?topic=5336.0](url)

shd7899, 2018-03-11:
Could not find any working SM16703 Arduino library. It's rather a chip in demand. No idea why some far east LED manufacturers claim that its protocol is compatible with ws2811.

Talamis, 2018-06-16:
+1 for sm16703

surik00, 2018-09-26:
Wow, it's been more than 2 years and still no support for SM16703

focalintent, 2018-09-26:
That’s because major work on the library stopped about two years ago because of this - https://plus.google.com/102282558639672545743/posts/JQKcujjaXXC - and no one has submitted a PR for 16703 support.

closed 2019-01-20 by focalintent

galgametentacle, 2019-05-17:
My 16703 is workable with WS2811 settings. It says 'CC16703 184508' on chip. Maybe from a China Factory.

kingnoss, 2019-11-22:
I have the cc16703 chip and my work around is using Adafruit Neopixel at 400KHz, and it works like a charm. I got a RF controller with my strip, but where is the fun in that.

NilsNippe, 2021-12-21:
For anyone looking for a solution: use the WS2811_400 led type.

FastLED.addLeds<WS2811_400, DATA_PIN, RGB>(leds, NUM_LEDS);

ahmetcemturan, 2024-05-14:
HAs anyone been able to run a strip with the SM16703 chipset yet?  It just does not work..

commit 87da428d (2019-01-20): Fix #345 - Add SM16703 timings