FastLED/FastLED #348: Can't get Example Code to compile

cogzoid, 2016-09-25:
I'm getting started on a FastLED/OctoWS2811 LED project.  I opened up the Example sketch in the FastLED folder called OctoWS2811Demo.  I thought that'd be a great place to get going.  The example code, however, does not compile.  I did a fresh install of Arduino 1.6.12 and Teensyduino, but the problem remains.  The error looking for OctoWS2813 seems like there's a typo in the library or something?  Anyway, here's the error msg:

In file included from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED/examples/Multiple/OctoWS2811Demo/OctoWS2811Demo.ino:3:0:
/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED/FastLED.h: In static member function 'static CLEDController& CFastLED::addLeds(CRGB*, int, int)':
/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED/FastLED.h:359:12: error: 'OCTOWS2813' was not declared in this scope
       case OCTOWS2813: { static COctoWS2811Controller<RGB_ORDER,WS2813_800kHz> controller; return addLeds(&controller, data, nLedsOrOffset, nLedsIfOffset); }
            ^
Error compiling for board Teensy 3.2 / 3.1.

focalintent, 2016-09-25:
This has already been fixed - pull the most recent version of the library.  (See #346 for the pull request where I took the fix)

closed 2016-09-25 by focalintent

cogzoid, 2016-09-25:
Thanks, that did it!  I mistakenly thought that a fresh install of Teensyduino would've started me with the most up to date libraries.  Still learning all the ins and outs.  Thanks for the fast reply.