FastLED/FastLED #350: Unable to Compile for Teensy 3.2

davidbitton, 2016-09-28:
I've trying to compile some code, and I'm receiving the following build errors:

```
In file included from /Users/bittondb/Documents/Arduino/libraries/FastLED/bitswap.h:4:0,
                 from /var/folders/nk/n29krwvj22s9xfk72mk0tt7h0000gn/T/arduino_build_369312/sketch/FrSkySPort.cpp:1:
/Users/bittondb/Documents/Arduino/libraries/FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.003
 #    pragma message "FastLED version 3.001.003"
                     ^
In file included from /Users/bittondb/Documents/Arduino/libraries/FastLED/platforms/arm/k20/fastled_arm_k20.h:12:0,
                 from /Users/bittondb/Documents/Arduino/libraries/FastLED/platforms.h:12,
                 from /Users/bittondb/Documents/Arduino/libraries/FastLED/FastLED.h:52,
                 from /Users/bittondb/Documents/Arduino/libraries/FastLED/bitswap.h:4,
                 from /var/folders/nk/n29krwvj22s9xfk72mk0tt7h0000gn/T/arduino_build_369312/sketch/FrSkySPort.cpp:1:
/Users/bittondb/Documents/Arduino/libraries/FastLED/platforms/arm/k20/clockless_block_arm_k20.h: In static member function 'static void InlineBlockClocklessController<LANES, FIRST_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::writeBits(uint32_t&, InlineBlockClocklessController<LANES, FIRST_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::Lines&, PixelController<RGB_ORDER, LANES, (((1 << LANES) - 1) & ((FIRST_PIN == 2) ? 255 : 4095))>&)':
/Users/bittondb/Documents/Arduino/libraries/FastLED/platforms/arm/k20/clockless_block_arm_k20.h:88:4: error: 'transpose8' was not declared in this scope
    transpose8<1,2>(b.bytes,b2.bytes);
    ^
Multiple libraries were found for "bitswap.h"
 Used: /Users/bittondb/Documents/Arduino/libraries/FastLED
 Not used: /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED
 Not used: /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED
 Not used: /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED
 Not used: /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED
Using library FastLED at version 3.1.3 in folder: /Users/bittondb/Documents/Arduino/libraries/FastLED 
Using library GCS_MAVLink in folder: /Users/bittondb/Documents/Arduino/libraries/GCS_MAVLink (legacy)
Using library EEPROM at version 2.0 in folder: /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/EEPROM 
Error compiling for board Teensy 3.2 / 3.1.
```

Any idea why this would be? Thanks!

focalintent, 2016-10-30:
Please grab the most recent version of the library (you should see FastLED version 3.001.005 in the compiler output) and try building again.  Also it would be helpful to see the code your'e trying to compile, beyond that.

davidbitton, 2016-10-30:
@focalintent i'm trying to compile https://github.com/scottflys/mavsky

davidbitton, 2016-10-30:
Here's the compile output:

```
In file included from /Users/bittondb/Documents/Arduino/libraries/FastLED/bitswap.h:4:0,
                 from /var/folders/nk/n29krwvj22s9xfk72mk0tt7h0000gn/T/arduino_build_395151/sketch/FrSkySPort.cpp:1:
/Users/bittondb/Documents/Arduino/libraries/FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.005
 #    pragma message "FastLED version 3.001.005"
                     ^
In file included from /Users/bittondb/Documents/Arduino/libraries/FastLED/platforms/arm/k20/fastled_arm_k20.h:12:0,
                 from /Users/bittondb/Documents/Arduino/libraries/FastLED/platforms.h:12,
                 from /Users/bittondb/Documents/Arduino/libraries/FastLED/FastLED.h:52,
                 from /Users/bittondb/Documents/Arduino/libraries/FastLED/bitswap.h:4,
                 from /var/folders/nk/n29krwvj22s9xfk72mk0tt7h0000gn/T/arduino_build_395151/sketch/FrSkySPort.cpp:1:
/Users/bittondb/Documents/Arduino/libraries/FastLED/platforms/arm/k20/clockless_block_arm_k20.h: In static member function 'static void InlineBlockClocklessController<LANES, FIRST_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::writeBits(uint32_t&, InlineBlockClocklessController<LANES, FIRST_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::Lines&, PixelController<RGB_ORDER, LANES, (((1 << LANES) - 1) & ((FIRST_PIN == 2) ? 255 : 4095))>&)':
/Users/bittondb/Documents/Arduino/libraries/FastLED/platforms/arm/k20/clockless_block_arm_k20.h:88:4: error: 'transpose8' was not declared in this scope
    transpose8<1,2>(b.bytes,b2.bytes);
    ^
Multiple libraries were found for "bitswap.h"
 Used: /Users/bittondb/Documents/Arduino/libraries/FastLED
 Not used: /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED
 Not used: /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED
 Not used: /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED
 Not used: /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED
Using library FastLED at version 3.1.5 in folder: /Users/bittondb/Documents/Arduino/libraries/FastLED 
Using library GCS_MAVLink in folder: /Users/bittondb/Documents/Arduino/libraries/GCS_MAVLink (legacy)
Using library EEPROM at version 2.0 in folder: /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/EEPROM 
Error compiling for board Teensy 3.2 / 3.1.
```

Still no-go. Thanks.

focalintent, 2016-10-30:
What version of arduino/teensyduino are you Ian g to compile?

davidbitton, 2016-10-30:
Arduno IDE 1.6.12 and Teensyduino 1.31-beta1. Thanks.

KarlAustin, 2017-03-19:
Same issue with 3.1.5 (pulled down master branch about 60s ago) and Teensyduino 1.35 on Teensy 3.2.

focalintent, 2017-03-20:
@KarlAustin what code are you attempting to build?  Also make sure that FastLED.h is the only header file you're pulling in (some versions of arduino like to include a whole bunch of the other header files and that can mess with things)

focalintent, 2017-08-16:
I changed how bitswap is being included which should take care of cases like this where library search paths are set up oddly.  (Alternatively, you might need to remove the teensyduino version of FastLED from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED)

closed 2017-08-16 by focalintent

JohnCoker, 2017-12-14:
I'm seeing the same problem with Arduino 1.8.5 and Teensyduino 1.40 (FastLED 3.1.6 is included). Note that this is a fresh install on Mac OS X (Sierra).

```
Arduino: 1.8.5 (Mac OS X), TD: 1.40, Board: "Teensy 3.6, Serial, 180 MHz, Faster, US English"

In file included from /Users/john/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED/bitswap.h:4:0,
                 from /Users/john/Documents/Arduino/LightShow2/Rainbow/Rainbow.ino:1:
/Users/john/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.006
 #    pragma message "FastLED version 3.001.006"
                     ^
In file included from /Users/john/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED/platforms/arm/k66/fastled_arm_k66.h:10:0,
                 from /Users/john/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED/platforms.h:15,
                 from /Users/john/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED/FastLED.h:55,
                 from /Users/john/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED/bitswap.h:4,
                 from /Users/john/Documents/Arduino/LightShow2/Rainbow/Rainbow.ino:1:
/Users/john/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED/platforms/arm/k66/clockless_block_arm_k66.h: In static member function 'static void InlineBlockClocklessController<LANES, FIRST_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::writeBits(uint32_t&, InlineBlockClocklessController<LANES, FIRST_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::Lines&, PixelController<RGB_ORDER, LANES, (((1 << LANES) - 1) & ((FIRST_PIN == 2) ? 255 : 4095))>&)':
/Users/john/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/FastLED/platforms/arm/k66/clockless_block_arm_k66.h:102:4: error: 'transpose8' was not declared in this scope
    transpose8<1,2>(b.bytes,b2.bytes);
    ^
Error compiling for board Teensy 3.6.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```