FastLED/FastLED #351: howAnalogRGB with "inverting" LED-driver

MHz000, 2016-09-28:
for power LEDs or for non pixel addressable RGB-stripes there are really cheap drivers called LED-amplifier aviable. However those "amplifiers" act as inverter. Driven by a PWM pin means a pure red fieres green and blue leaving red dark etc. Maybe there is a function/setting to invert the PWM duty cycles I don't know of ...
or it can be implemented
Lookk for: LED Strip Amplifer **DC5-24V12A** Mini LED Amplifier for LED Strip Power Repeater Console Controller. Less than 2 $

Daniel-dk, 2016-09-28:
when I directly control LEDs thru these devices I "invert " the signal by subtracting it form 255 ( assuming its 8 bit)  so that Low == LED on, high == LED off.

analogWrite(redPin,redVal); becomes analogWrite(redPin, 255-redVal);

there are probably much better ways of doing this.

MHz000, 2016-09-28:
Hi Daniel this sounds good, thank you!!!
Btw. I've replaced the zenerdiode on the "amplifier" PCB by an R equal to
that from Vcc to Zenerdiode (to ground)  Vcc---R---Z----| (4k7 ? i don't
remember) to lower the treshold.
Thank you for the tip