FastLED/FastLED #352: Compiler error: error: 'calculate_max_brightness_for_power_mW' was not declared in this scope

tgiphil, 2016-10-02:
FastLED.h: In member function void CFastLED::setMaxPowerInMilliWatts(uint32_t)
 
FastLED.h: 466:77: error: 'calculate_max_brightness_for_power_mW' was not declared in this scope
   inline void setMaxPowerInMilliWatts(uint32_t milliwatts) { m_pPowerFunc = &calculate_max_brightness_for_power_mW; m_nPowerData = milliwatts; }
Error compiling project sources

focalintent, 2016-10-02:
What version of arduino are you using, what version of the library do you have, and what hardware are you compiling for, and what is the code you are trying to compile?

tgiphil, 2016-10-02:
I figured out the issue. When add a library with Arduino IDE for Visual Studio, it adds includes for all the header files when only FastLED.h is necessary.

The resolution was to remove all the references to the other header files. 

Btw. Great project! Keep up the great work!

closed 2016-10-02 by tgiphil