FastLED/FastLED #353: Three LEDs turn on at a time

RickMeasham, 2016-10-09:
I recently purchased a [WS2811 LED Strip from AliExpress](https://www.aliexpress.com/item/new-hot-selling-best-price-5m-DC12V-ws2811ic-5050-RGB-SMD-addressable-ws2811-led-pixels-strip/32621311780.html). (The sales page includes no detailed specs and the package is similarly lacking any specs. So let's just assume it really is a a WS2811 strip.)

I followed the [Basic Usage](https://github.com/FastLED/FastLED/wiki/Basic-usage) guide after connecting it to my Arduino Nano, but noticed that when I ran the first piece of code (to turn the first LED to red) the first three LEDs turned on. 

I then went on to the 'Moving' section and noticed that it lights the first three, then the next three LEDs right down the line. If I say there are 60 LEDs, then it lights the first 180 - 3 at a time - before starting over.

I'm unable to find any other Issue or Google result that describes this problem. Am I missing something?

(The color order was also wrong, but that was an easy fix using the RGBCalibrate process)

focalintent, 2016-10-09:
If you look closely st the strip that you brought, you'll see that there is one ws2811 IC per group of 3 leds, so, for better or for worse, this is going to be the expected behavior, as there is not going to be any way to address each of those leds individually.  

Sadly, this arrangement seems to be pretty standard for the 12v ws2811 strips.

closed 2016-10-09 by focalintent

jasoncoon, 2016-10-09:
The key part in the product description: LED Type: 1 WS2811 IC control 3 LED Chip.  I usually avoid WS2811 unless I want the 8mm node style.

RickMeasham, 2016-10-09:
Thanks for the help. It's clear in the description when you're looking for it or have been to this rodeo before! Luckily for this project, controlling them in batches of three will be fine.