FastLED/FastLED #354: Compile under Arduino CREATE fails.

BallscrewBob, 2016-10-11:
Has anyone tried these libs under CREATE the online Arduino IDE ?

`Using library FastLED-3.1.3 at version 3.1.3 in folder: /tmp/336652572/custom/FastLED-3.1.3

In file included from /tmp/336652572/AnalogOutput/AnalogOutput.ino:1:0:

/tmp/336652572/custom/FastLED-3.1.3/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.003

# pragma message "FastLED version 3.001.003"

^`

Tried a few versions and error as above.

focalintent, 2016-10-11:
Weird - that shouldn't be getting treated as an error - it's just a message.  Are there other error messages as well, or is it actually failing to compile because of this?

Needless to say, CREATE isn't  an ide that I've done any testing with. For that matter, i didn't even know it existed until just now :)

BallscrewBob, 2016-10-11:
Its actually failing to compile or upload but does both in the regular IDE's
From what I understand the libs are a little more defined for CREATE so
maybe its something pretty minor.

I also have access to create DEV side as a tester if you want anything
trying out.

focalintent, 2016-10-11:
Is the only error message the pragma?  Or are there other messages? (Can you toss the full build log in a gust?)

BallscrewBob, 2016-10-11:
Full build from CREATE. I can if you wish put the Chrome output here too if it helps.
Adafruits libs compile fine as do Polus for LED's

> arduino-builder/arduino-builder -compile -core-api-version 10611 -build-path /tmp/780737267 -hardware arduino-builder/hardware -hardware arduino-builder/packages/cores -tools arduino-builder/tools -tools arduino-builder/packages/tools -built-in-libraries arduino-builder/latest -libraries /tmp/071885028/pinned -libraries /tmp/071885028/custom -fqbn arduino:avr:uno -verbose=true /tmp/071885028/AnalogOutput

Using board 'uno' from platform in folder: /home/admin/builder/arduino-builder/packages/cores/arduino/avr

Using core 'arduino' from platform in folder: /home/admin/builder/arduino-builder/packages/cores/arduino/avr

Detecting libraries used...

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/standard" "/tmp/780737267/sketch/AnalogOutput.ino.cpp" -o "/dev/null"

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/standard" "-I/tmp/071885028/custom/FastLED-3.1.3" "/tmp/780737267/sketch/AnalogOutput.ino.cpp" -o "/dev/null"

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/standard" "-I/tmp/071885028/custom/FastLED-3.1.3" "/tmp/071885028/custom/FastLED-3.1.3/FastLED.cpp" -o "/dev/null"

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/standard" "-I/tmp/071885028/custom/FastLED-3.1.3" "/tmp/071885028/custom/FastLED-3.1.3/bitswap.cpp" -o "/dev/null"

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/standard" "-I/tmp/071885028/custom/FastLED-3.1.3" "/tmp/071885028/custom/FastLED-3.1.3/colorpalettes.cpp" -o "/dev/null"

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/standard" "-I/tmp/071885028/custom/FastLED-3.1.3" "/tmp/071885028/custom/FastLED-3.1.3/colorutils.cpp" -o "/dev/null"

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/standard" "-I/tmp/071885028/custom/FastLED-3.1.3" "/tmp/071885028/custom/FastLED-3.1.3/hsv2rgb.cpp" -o "/dev/null"

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/standard" "-I/tmp/071885028/custom/FastLED-3.1.3" "/tmp/071885028/custom/FastLED-3.1.3/ [... cut ...]
"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-objcopy" -O ihex -R .eeprom "/tmp/780737267/AnalogOutput.ino.elf" "/tmp/780737267/AnalogOutput.ino.hex"

Using library FastLED-3.1.3 at version 3.1.3 in folder: /tmp/071885028/custom/FastLED-3.1.3

In file included from /tmp/071885028/AnalogOutput/AnalogOutput.ino:1:0:

/tmp/071885028/custom/FastLED-3.1.3/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.003

# pragma message "FastLED version 3.001.003"

^

focalintent, 2016-10-11:
So according to that build log it fully and correctly built everything (you wouldn't have a .hex file at the end of it all if it hadn't).  If the presence of a #pragma message is keeping CREATE from being able to then flash that hex file, then that's a bug in CREATE - but you can work around that in the meantime by commenting out line 17 in FastLED.h

BallscrewBob, 2016-10-11:
Agreed with CREATE being issue and have passed along your comment.
They agree also with issue thier end and will look at a fix for it.
meanwhile going to try your suggestion and thank you so much for taking the time to help with this.

Can close this if you wish please

closed 2019-01-20 by focalintent