FastLED/FastLED #355: Does not work with latest version (2.3.0) of the ESP8266 Arduino library

gozerdgozerian, 2016-10-13:
First, thanks for the awesome work on this library and all the support! So I tried this out on a ESP8266 (NodeMCU) and followed the instructions in the note regarding the same. I was using version 2.3.0 (the latest) of the ESP8266 Arduino library and the latest version of FastLED (3.1.3). Everything compiled fine, but I couldn't get any LEDs to light up (using an Adafruit NeoPixel ring for what it's worth). I noticed something in the G+ forum about using version 2.1 of the ESP8266 library, so I rolled back to that and now everything is working fine. Not sure if this is on FastLED or the latest ESP8266 library, but thought I would let you know. Thanks

focalintent, 2016-10-13:
I wonder if they did something with 2.3 that breaks the pin ordering (there's a couple different mappings of PIN number on the board and physical gpio) - 2.3.0 works with the wemos d1 mini - there's defines you can set to force which mapping it uses, when I get back to a computer I can look them up if you haven't found them yet.

SciLor, 2016-10-29:
I also have the problem with my NodeMCU and WS2801 Pixels.
I am on Arduino 1.6.12 + FastLED 3.1.3 + ESP8266 2.3.0
Using Adafruit_WS2801 it works fine.

Delphi, 2016-11-16:
Unfortunately I too am experiencing issues getting any response from the LEDs using a LinkNode D1
http://linksprite.com/wiki/index.php5?title=LinkNode_D1

I have tried the following combinations:-
Arduino 1.6.12 + FastLED 3.1.5 + ESP8266 2.3.0
Arduino 1.6.12 + FastLED 3.1.3 + ESP8266 2.3.0
Arduino 1.6.12 + FastLED 3.1.5 + ESP8266 2.2.0
Arduino 1.6.12 + FastLED 3.1.3 + ESP8266 2.2.0
Arduino 1.6.12 + FastLED 3.1.5 + ESP8266 2.1.0
Arduino 1.6.8 + FastLED 3.1.5 + ESP8266 2.3.0
Arduino 1.6.8 + FastLED 3.1.5 + ESP8266 2.2.0

I have #define FASTLED_ESP8266_RAW_PIN_ORDER before #include FastLED.h and have set pin 13 (D7) as the data pin.

I have confirmed that pin 13 is D7 by wiring up a normal LED between this and ground which would flash as 13 is set HIGH or LOW.

Code compiles without error.
Code and LED strips work on Arduino Uno R3, tho sadly not on this ESP8266.

kloot, 2017-01-27:
I can confirm the observation of gozerdgozerian - fail with esp8266 versions 2.3.0 and 2.2.0 with all combinations of three different NodeMCU boards, two different WS2812B devices, and two different power setups (NodeMCU and WS on same PSU, different PSUs w GND connected). Also tried TXB level converter and 220R in data line without success. All these setups worked perfectly fine when running the same sketch on arduino uno, but none produced any output on NodeMCU (all output pins tested with scope)
Conversely, all combinations work after downgrading esp8266 to 2.1.0.
My best conclusion is that the problems were software-related (esp8266) and not caused by power issues.
Used FASTLed v 3.1 arduino 1.8.1, with #define DATA_PIN 5 (NodeMCU D1) for all tests.
Hope this helps, many thanks for the great library!

misan, 2017-04-17:
I have had to revert to version 2.0.0 as 2.2.0 and 2.3.0 do cause a wdt when using timer0. Not sure if that is due to some change on how these newer versions handle it. So my code is happily working with 2.0.0 but causes wdt all the time with newer versions.

closed 2019-01-20 by focalintent