FastLED/FastLED #356: Parallel Output Due

hemske, 2016-10-17:
Dear,

We have some issues with using 2 or more outputs with an Arduino Due.
The first output is working quite well, but the second output is flickering to totally different colors. 
Is this a common problem and are there some solutions available?

Thank you in advance.

focalintent, 2016-10-17:
Please upload your code to gist.girhub.com so I can see what you're doing, code wise.  

Also, what version of arduino are you using? 

What type of leds are you using?

Are you using a level shifter on the Due's outputs to go from 3.3v to 5v?

hemske, 2016-10-17:
I am using 1.6.9, LED's are LPD1886 and i am using level shifters.
I will upload my code.

focalintent, 2016-10-17:
What happens if you run some demo code that writes out to the two led strips that _doesn't_ involve DMX reading.  It's possible that the issue you're running into has more to do with interrupts stepping on each other - either the DMX reading interrupts messing with the LED output or the LED output writing messing with reading DMX data.

hemske, 2016-10-24:
I am still having some problems with receiving ArtNet and sending with Fastled. Do you have a working example for that?

focalintent, 2016-10-24:
See https://github.com/FastLED/FastLED/wiki/Interrupt-problems for some discussion about interruotndriven protocols and the 3-wire chipsets.  I personally don't use anything like artnet or dmx with most of my projects - you would have better luck finding other folks who have on the g+ group at http://FastLED.io/+

closed 2016-10-30 by focalintent