FastLED/FastLED #360: Problem with WS2801

pstama, 2016-10-22:
Hello.

I have tried many examples from the FASTLED library But no one is working correctly.
(Sometimes working  sometimes not).

As i have searched there is a issue with this chipset. Any recommendations ?

I have the arduino UNO.

focalintent, 2016-10-22:
What version of the library and what version of arduino are you using?  Also - what pins are you using for clock and data? 

(I'll have to see if I still have any ws2801 chips around here to work with).  

And when you say not working - how do you mean?  No leds turn on?  The colors are wrong and random?  Or something else?

Could you also post one of the programs that isn't working, as you're trying to build it - so I can see if there's anything slipping in there that might be a problem.

pstama, 2016-10-25:
Hello

I have just fixed the problem. I have input  directly at the command that
specifies what kind of controller is used the data pin and the clock pin. (
without using the #define command ). Now its working perfect.

Thank you again for the support.

The problem was random colors and sometimes only the first led was flashing
. I am using the latest library.

focalintent, 2016-10-25:
Ah - that sounds like having the clock and data signals wired backwards.

closed 2016-10-25 by focalintent