FastLED/FastLED #361: Teensy 3.5, static_assert(validpin(), "Invalid pin specified");

tehuster, 2016-10-25:
When i try to use pin 35/36 on the teensy 3.5 i get the error: static_assert(validpin(), "Invalid pin specified");

If i understand it correctly i need to define a pinmapping for the teensy in the fastpin.h? 
The problem is, I don't know how to do that. If you point me in the right direction I would be very gratefull. 

Thanks

jasoncoon, 2016-10-25:
Teensy 3.5 and 3.6 aren't supported by FastLED yet.  Daniel has said he has the hardware, but hasn't had the time yet: https://plus.google.com/u/0/+BrianLewis515/posts/e9PN9heiWcR

Side note, I noticed Paul has added Teensy 3.5 and 3.6 support to the OctoWS2811 library: https://github.com/PaulStoffregen/OctoWS2811/commit/d51e6a65c63e1d8b4539c7a93fc3c1c42ccdbd4b

tehuster, 2016-10-25:
Thanks for the info Jason. Guess i will just wait on Daniel or figure out how i need to pinmap it myself.

robertcedwards, 2016-11-16:
Ah shoot, that's the problem! I was having a bunch of funky issues but using 3.0.1 seems to have done the trick for our Dotstars, along with with slowing down our clock as per Paul's recommendation:
https://plus.google.com/111022735393886581158/posts/JcCkB7j66aj?sfc=true

Mad-Onion, 2016-11-23:
@robertcedwards can yout tell me what you did to get the Dotstars running on T3.5? I'm using arduino 1.6.12, teensyduino 1.3.1-beta3 and FastLED 3.1.3 but still get the error mentioned in the title.

robertcedwards, 2016-11-23:
I'm running 1.6.10 with FastLED 3.1.0, that FastLED library version was key to getting it to run properly.

Mad-Onion, 2016-11-23:
@robertcedwards thank you for the info, I'm testing Arduino 1.6.12 with FastLED 3.1.0 and WS2812Bs now and when I try to compile the blink-example I run into lots of asm-constraint and macro-expansion-errors.
Any ideas?

robertcedwards, 2016-11-23:
@Mad-Onion We had to roll back to Arduino 1.6.10 as well. There was definitely a lot of testing and multiple software setups before we got it working. I'm able to compile it currently with the IDE, so that should be correct.

kcouck, 2017-01-17:
If you'd like, I can add 3.5 support to my branch for 3.6: https://github.com/kcouck/FastLED/tree/Teensy_3.6
I can't test it, I don't have the hardware, only a 3.6...
But the pin mapping etc should be identical, so adding support should be trivial...

oleiman, 2017-01-22:
@kcouck Yes, that would be super helpful. I've just started a prototype for a large project, and I planned to use 3.5 and FastLED. Didn't think to check whether the library supports the new hardware. There's a lesson here, I'm sure. 

I'm going to have a peek into your branch right now to see whether I can get it going. Thanks!

henryflorence, 2017-02-14:
Teensyduino does not currently support Arduino 1.6.10. Supported versions are: 1.0.6, 1.6.5-r5, 1.6.9, 1.6.12, 1.6.13, 1.8.0. I am getting the error: 'WS2811_PORTDC' was not declared in this scope when trying to compile for teensy 3.6 using Arduino 1.8.

kcouck, 2017-02-15:
Hello Henry, (and others)

this issue has nothing to do with the Arduino orr Teensyduino version, but with the fact that the stable branch of FastLED just doesn't support the Teensy 3.5/3.6 yet.

henryflorence, 2017-02-16:
Thanks for your reply kcouck, I found a solution, after pulling your fork `git checkout -b Teensy_3.6` makes your alterations appear, managed to pull changes from the main FastLED source also. Did you have any luck getting more than 16 lanes working? I'm currently using WS2811_PORTDC

kcouck, 2017-02-16:
You can use PORTB as well, it's on pins 0,1,29,30,43,46,44,45 (some of them are on the backside of the Teensy)
But you'll need to use it by adding a second 8-way parallel output; I didn't create a 24-way controller (yet)
Waiting for @focalintent to merge it :)

closed 2019-01-20 by focalintent