FastLED/FastLED #362: Wemos D1 R2 resetting

gavspav, 2016-10-26:
My board is resetting. 
My sketch causes frequent resets.
At first I thought it was Ram or Power issues but I have ruled them out.
Code works with Teensy
Apologies if it is not an issue with the library.

`include "FastLED.h"

#define NUM_LEDS 8 //number of leds in strip length on one side
#define DATA_PIN D2//7 = second hardware spi data
#define CLOCK_PIN D3//14 = second hardware spi clock
#define COLOR_ORDER BGR
CRGB leds[NUM_LEDS];
int numberOfSlices = 8;

void PoiSonic(unsigned long time, const unsigned int array[]){
unsigned long currentTime = millis();
 while (millis()< currentTime + (time)) {

int f= numberOfSlices;
int z; //a counter
int j=NUM_LEDS;

    for (int x=0;x<f;x++){
     for(z=NUM_LEDS;z>0;z--){
       leds[z-1]=array[x+((j-z)*f)];
       
       }
    
     FastLED.show();
     delayMicroseconds(1200); //may need to increase / decrease depending on spin rate
     }       
    delayMicroseconds(1000); //may need to increase / decrease depending on spin rate
   }
 }`

focalintent, 2016-10-26:
You can't go for 20 seconds without exiting the loop function on the esp8266 - that will likely cause the watchdog timer to reset, also - how many items are in array1 - you'll want to make sure that you aren't accessing outside of array1 - reading from an invalid address is also something that might cause a reset on the esp8266.

gavspav, 2016-10-26:
Ah ok. Thanks - I threw a delay(0) in there and the resets stopped.
Sorry for the false flag.

closed 2016-10-26 by gavspav