FastLED/FastLED #364: Teensy 3.5 ports

dimitre, 2016-10-28:
It would be great to support Teensy 3.5 ports >= 24.
I'm testing with Adafruit_DotStar and they are already working there.
Thank you!

Mad-Onion, 2016-11-23:
@dimitre how did you get Teensy 3.5 with FastLED working? When using FastLED 3.1.5, Teensyduino 1.3.1 and Arduino 1.6.12 I get different errors in the blink-example:
- No pin/port mappings found
- no matching function for call to 'CFastLED::addLeds(CRGB [1], int)'

dimitre, 2016-11-23:
@Mad-Onion it worked using branch 339, as you can follow on this thread:
https://github.com/FastLED/FastLED/issues/339

Mad-Onion, 2016-11-23:
@dimitre thank you!

Is there a roadmap for a T3.5 port?

focalintent, 2016-11-23:
No, everything is on an indefinite hold right now while @kriegsman and I deal with a death in the family.

kcouck, 2017-01-17:
Isn't this the same issue as #361?

focalintent, 2019-01-20:
Someone submitted 3.5/3.6 support.