FastLED/FastLED #365: BRG Color Order causes compile error (Feather M0, APA102, Arduino 1.6.11, FastLED 3.1.5)

happyinmotion, 2016-10-30:
I'm setting up an APA102 strip to run from a Feather M0 WiFi (ATSAMD21 + ATWINC1500), using the 1.6.11 IDE and FastLED 3.1.5, with a MOSFET in there to control power to the strip. First job after getting the LEDs to light is check the colour order.

RGBCalibrate runs just fine with the RGB colour order. The LED strip shows 1 Blue, 2 Green, 3 Red. Ok, so set it to BGR. And the code fails to compile. Error is:

`Arduino: 1.6.11 (Windows 10), TD: 1.30, Board: "Adafruit Feather M0 (Native USB Port)"

In file included from C:\Users\jezwe\Documents\Making\workspace\TTOE_Testing_RGBCalibrate\TTOE_Testing_RGBCalibrate.ino:1:0:

C:\Users\jezwe\Documents\Making\workspace\libraries\FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.005

 #    pragma message "FastLED version 3.001.005"

                     ^

In file included from C:\Users\jezwe\Documents\Making\workspace\libraries\FastLED/FastLED.h:48:0,

                 from C:\Users\jezwe\Documents\Making\workspace\TTOE_Testing_RGBCalibrate\TTOE_Testing_RGBCalibrate.ino:1:

C:\Users\jezwe\Documents\Making\workspace\libraries\FastLED/fastpin.h: In instantiation of 'class FastPin<136u>':

C:\Users\jezwe\Documents\Making\workspace\libraries\FastLED/fastspi_bitbang.h:23:50:   required from 'class AVRSoftwareSPIOutput<9u, 136u, 4u>'

C:\Users\jezwe\Documents\Making\workspace\libraries\FastLED/fastspi.h:30:7:   required from 'class SPIOutput<9u, 136u, 4u>'

C:\Users\jezwe\Documents\Making\workspace\libraries\FastLED/chipsets.h:94:6:   required from 'class LPD8806Controller<9u, 136u, (EOrder)10u, 4u>'

C:\Users\jezwe\Documents\Making\workspace\libraries\FastLED/FastLED.h:233:66:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = (ESPIChipsets)5u; unsigned char DATA_PIN = 9u; unsigned char CLOCK_PIN = 136u]'

C:\Users\jezwe\Documents\Making\workspace\TTOE_Testing_RGBCalibrate\TTOE_Testing_RGBCalibrate.ino:44:64:   required from here

C:\Users\jezwe\Documents\Making\workspace\libraries\FastLED/fastpin.h:206:2: error: static assertion failed: Invalid pin specified

  static_assert(validpin(), "Invalid pin specified");

  ^

exit status 1
Error compiling for board Adafruit Feather M0 (Native USB Port).
Invalid library found in C:\Users\jezwe\Documents\Making\workspace\libraries\adafruit_guide_fft: C:\Users\jezwe\Documents\Making\workspace\libraries\adafruit_guide_fft
Invalid library found in C:\Users\jezwe\Documents\Making\workspace\libraries\MPU-9250: C:\Users\jezwe\Documents\Making\workspace\libraries\MPU-9250
Invalid library found in C:\Users\jezwe\Documents\Making\workspace\libraries\adafruit_guide_fft: C:\Users\jezwe\Documents\Making\workspace\libraries\adafruit_guide_fft
Invalid library found in C:\Users\jezwe\Documents\Making\workspace\libraries\MPU-9250: C:\Users\jezwe\Documents\Making\workspace\libraries\MPU-9250

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`
Odd.

Checking other colour orders gives:
RBG - compiles without complaint, but LEDs don't light.
GRB - same compile error as above.
GBR - same compile error as above.
BRG - same compile error as above.

The code I'm running is RGBCalibrate with a minor change to turn on the MOSFET:
`#include "FastLED.h"

////////////////////////////////////////////////////////////////////////////////////////////////////
//
// RGB Calibration code
//
// Use this sketch to determine what the RGB ordering for your chipset should be.  Steps for setting up to use:

// * Uncomment the line in setup that corresponds to the LED chipset that you are using.  (Note that they
//   all explicitly specify the RGB order as RGB)
// * Define DATA_PIN to the pin that data is connected to.
// * (Optional) if using software SPI for chipsets that are SPI based, define CLOCK_PIN to the clock pin
// * Compile/upload/run the sketch 

// You should see six leds on.  If the RGB ordering is correct, you should see 1 red led, 2 green 
// leds, and 3 blue leds.  If you see different colors, the count of each color tells you what the 
// position for that color in the rgb orering should be.  So, for example, if you see 1 Blue, and 2
// Red, and 3 Green leds then the rgb ordering should be BRG (Blue, Red, Green).  

// You can then test this ordering by setting the RGB ordering in the addLeds line below to the new ordering
// and it should come out correctly, 1 red, 2 green, and 3 blue.
//
//////////////////////////////////////////////////

#define NUM_LEDS 6

const uint8_t LEDenablePowerPin = 12;
#define DATA_PIN    9
#define CLK_PIN     10
#define LED_TYPE    APA102
#define COLOR_ORDER BRG
#define NUM_LEDS    16
CRGB leds[NUM_LEDS];

void setup() {
	// sanity check delay - allows reprogramming if accidently blowing power w/leds
  delay(200);

  pinMode(LEDenablePowerPin, OUTPUT);
  digitalWrite(LEDenablePowerPin, LOW);  // Turn ON power to LED strips

//  FastLED.addLeds<LED_TYPE,DATA_PIN,COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);
  FastLED.addLeds<LED_TYPE,DATA_PIN,COLOR_ORDER>(leds, NUM_LEDS);
}

void loop() {
   leds[0] = CRGB(255,0,0); 
   leds[1] = CRGB(0,255,0);
   leds[2] = CRGB(0,255,0);
   leds[3] = CRGB(0,0,255);
   leds[4] = CRGB(0,0,255);
   leds[5] = CRGB(0,0,255);
   FastLED.show();
   delay(1000);
}`

(Also, trivially, this is FastLED version 3.1.5 and library.properties has "version=3.1.5" but FastLED.h has the message "pragma message "FastLED version 3.001.005". Is there a reason for that or is that just an oversight?)

focalintent, 2016-10-30:
The problem is that APA102 requires a clock _and_ data pin:

```FastLED.addLeds<LED_TYPE,DATA_PIN,COLOR_ORDER>(leds, NUM_LEDS);```

The compiler is interpreting the value that you put in for COLOR_ORDER as the pin for the clock - some of the integer values for color orders are high enough that they would show off as an invalid pin.

As for the pragma output, the leading zeros is just a printing convention on output - it's still 3.1.5 in all the places.

closed 2016-10-30 by focalintent

happyinmotion, 2016-10-30:
Ah, cheers for spotting that.

Putting:
    FastLED.addLeds<LED_TYPE,DATA_PIN,CLK_PIN,COLOR_ORDER>(leds, NUM_LEDS);
Works just fine and BGR it is.