FastLED/FastLED #366: Invalid duration of first delay() after FastLED.show() on Particle Photon

aboyett, 2016-10-30:
I'm having trouble with FastLED 3.1.5 on a Particle Photon (firmware version 0.5.3) with a 12 LED Neopixel (WS2812) ring. This may be the same or a similar issue to #299.

It looks like the first `delay()` call after a call to `FastLED.show()` ends up being cut short (or skipped entirely) but every so often will last for the requested duration. A second `delay()` after the first runs for the requested duration.

This short sample program is able to reproduce the issue consistently for me and has been tested against two different Photons.

```
// This #include statement was automatically added by the Particle IDE.
#include "FastLED/FastLED.h"
FASTLED_USING_NAMESPACE;

// How many leds in your strip?
#define NUM_LEDS 12
#define DATA_PIN 3

// Define the array of leds
CRGB leds[NUM_LEDS];

int count = 0;
String msg;

void pauseAndLog(String color, int numPause = 1, int duration = 1000) {
    msg = String(count) + "  " + String(millis()) + "  " + Time.timeStr();
    Serial.println(color + " " + msg);
    for(int i = 0; i < numPause; i++) {delay(duration);}
}

void setup() {
    Serial.begin(115200);
    delay(15000); // pause allow attaching to console before messages are printed
    FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS);
    Serial.println("Starting firmware.");
}

void loop() {
    leds[0] = CRGB::Red;
    FastLED.show();
    pauseAndLog("red  ");

    leds[0] = CRGB::Black;
    FastLED.show();
    pauseAndLog("black");
    count++;
}
```

With all `FastLED.show()` calls commented it pauses for 1000ms between each serial message as requested:
```
Starting firmware.
red   0  19723  Sun Oct 30 06:29:28 2016
black 0  20723  Sun Oct 30 06:29:29 2016
red   1  21724  Sun Oct 30 06:29:31 2016
black 1  22724  Sun Oct 30 06:29:32 2016
red   2  23725  Sun Oct 30 06:29:33 2016
black 2  24725  Sun Oct 30 06:29:34 2016
red   3  25726  Sun Oct 30 06:29:35 2016
black 3  26726  Sun Oct 30 06:29:36 2016
red   4  27727  Sun Oct 30 06:29:37 2016
black 4  28727  Sun Oct 30 06:29:38 2016
red   5  29728  Sun Oct 30 06:29:39 2016
black 5  30728  Sun Oct 30 06:29:40 2016
```

With `FastLED.show()` uncommented, I see brief flickers of the LED, which corresponds with the 3ms between most black to red transitions. Of note, every so often I do see the LED lit for a full second, these correspond with the instances where the `millis()` output goes backwards (visible in loops 0, 6, and 22)
```
Starting firmware.
red   0  35790  Sun Oct 30 06:32:01 2016
black 0  55870  Sun Oct 30 06:32:02 2016
red   1  55873  Sun Oct 30 06:32:03 2016
black 1  56874  Sun Oct 30 06:32:03 2016
[... 150 lines cut ...]
red   14  55971  Sun Oct 30 06:48:26 2016
black 14  20185  Sun Oct 30 06:48:26 2016
red   15  55976  Sun Oct 30 06:48:26 2016
black 15  20190  Sun Oct 30 06:48:26 2016
red   16  55981  Sun Oct 30 06:48:26 2016
black 16  20195  Sun Oct 30 06:48:26 2016
red   17  55986  Sun Oct 30 06:48:26 2016
black 17  20200  Sun Oct 30 06:48:26 2016
red   18  55991  Sun Oct 30 06:48:26 2016
black 18  20205  Sun Oct 30 06:48:26 2016
red   19  55996  Sun Oct 30 06:48:26 2016
black 19  20210  Sun Oct 30 06:48:26 2016
red   20  56001  Sun Oct 30 06:48:26 2016
black 20  20215  Sun Oct 30 06:48:26 2016
red   21  56006  Sun Oct 30 06:48:26 2016
black 21  20220  Sun Oct 30 06:48:26 2016
red   22  56011  Sun Oct 30 06:48:26 2016
black 22  20225  Sun Oct 30 06:48:26 2016
red   23  56016  Sun Oct 30 06:48:26 2016
black 23  20230  Sun Oct 30 06:48:26 2016
red   24  56021  Sun Oct 30 06:48:26 2016
black 24  20235  Sun Oct 30 06:48:26 2016
red   25  56026  Sun Oct 30 06:48:26 2016
black 25  20240  Sun Oct 30 06:48:26 2016
```

focalintent, 2016-10-30:
The problem is that writing out WS2812 data requires disabling interrupts -and the clock/timer is advanced by interrupts.  FastLED will try to adjust the timer appropriately - but it's approximate, and not always exact.

Adding the line:

```
#define FASTLED_ALLOW_INTERRUPTS 1
```

before you include FastLED.h may help - as it will allow the SYSTICK interrupts to fire when expected, advancing the clock appropriately.

aboyett, 2016-10-30:
Thanks for the quick response. Unfortunately, adding that definition at the top of the program doesn't appear to help. I'm still seeing the issue.

To drill down deeper, the value of `millis()` changes after `FastLED.show()` in every iteration of `loop()`

I'm now using an event loop rather than calls to `delay()`. See https://gist.github.com/aboyett/23ccea0b89fe5feef1fe330f43c04840 for the code and output both with FastLED enabled and disabled.

 The code prints an output line both before and after each call to FastLED.show(). In `Output_without_FastLED.show()`, the value of `millis()` (column four) closely matches the `millis()` value at the beginning of the loop (called "ls:" in the output) in both the before and after cases. Compare that to `Output_with_FastLED.show()` where the value of `millis()` before a call to `FastLED.show()` matches the loopstart value, but afterwards it varies. After the first call (setting the LED to red), millis() - loopstart is 16122. After the second call (setting the LED to black) and for every call there after, millis() - loopstart = 34792

focalintent, 2016-10-31:
Hmm - you know what, I wonder if the sparkcore firmware is using the _CYCCNT debug cycle count register to do its clock/millis timing (if so, that'd be really stupid) - as I use that register for cycle accurate counting for getting the output timings correct.  I'll have to get a hold of particle's firmware code to check on how, exactly, they're doing their millis timing.

focalintent, 2016-10-31:
Yup - that's the problem - they're using DWT->CYCCNT to try to harden up against interrupts disabling the SYSTICK timer from counting/incrementing - https://github.com/spark/firmware/blob/ef92286e147acdc3730b7752737e1503059f1d2c/platform/MCU/shared/STM32/src/hw_ticks.c  - the problem is, that's a debug register that I also use for the timing control, and part of that involves resetting the value of the timer (to remove the complexity that comes with dealing with/worrying about overflow).  I can work on rewriting that at some point (though  at the moment my time available for working on this stuff is very limited - and the particle is a pain in the ass to work with as a library developer because of other fuckery with their build environment that I've been waiting years for them to fix).  This is the first time that i've seen platform code use the debug registers in a way that expects user code to not use them.

focalintent, 2016-10-31:
(ah - this was a change they made this past april - which would explain why I didn't see this issue at all when I was initially working on/writing this code)

focalintent, 2016-10-31:
And I think I have a way to fix - depending on how work goes today I should be able to get it in sometime during the day today, if not then hopefully tonight after work.

aboyett, 2016-11-01:
No need to prioritize this. For the projects I was working on the event-loop approach is a better choice, I just was trying to take the lazy approach initially by using `delay()`.

It looks like these changes to `millis()` and `micros()` were merged in the latest patch release ([0.5.3](https://github.com/spark/firmware/releases/tag/v0.5.3)) put out less than a month ago. I'll try testing both 0.5.2 and one of the 0.6 release candidates and report back. I suspect the issue won't be present in 0.5.2 as it is lacking the changes you identified (from https://github.com/spark/firmware/pull/943).

jassmith, 2016-11-01:
I hope you dont mind my asking aboyett but how are you managing to run FastLED master on a particle photon? Any time I try to build I get: led_sysdefs.h:27:43: fatal error: platforms/avr/led_sysdefs_avr.h: No such file or directory

I have googled and found references to people saying FASTLED_USING_NAMESPACE; fixes it for them, but that doesn't seem to work for me (I am already doing that). I am using the FastLED-Sparkcore repo in the mean time, but I would very much like to be running the tip of fastled instead.

focalintent, 2016-11-01:
Tell the particle folks to make it less of a pain in the ass to manage libraries in their system :p. Between not functioning with the same directory layout as arduino libraries and requiring librar directory structures to be flat, it's enough of a pain that I'm not planning on doing the next push to spark core until I finish the rgbw work.

focalintent, 2016-11-01:
(That's why I have the separate sparkcore repo)

jassmith, 2016-11-01:
@focalintent if you are happy enough to explain to me how to do it I am more than happy to maintain your port. I looked at what you did and it seems you *mostly* flattened it out, however there are some other tweaks (especially around the math library) I don't understand why you did them.

jogleasonjr, 2016-12-06:
I'm with @jassmith  and would contribute. It's not immediately clear to me what, at a high level, is different in the Sparkcore variant since the current one has many more commits. 

I also experience the missing references in `led_sysdefs_avr.h`. I'm sure the particle compiler has something to do with it, but it'd be great to have a Particle fork in relative lockstep with this repo, possibly using automated migration if there's a pattern of things to do.

closed 2019-01-20 by focalintent