FastLED/FastLED #368: LED array crash due to wrong input in beatsin16() in the examples

danlsk, 2016-11-08:
They are 2 examples sinelon() and jungle() which calls for beatsin16() function. 
`uint16_t pos = beatsin16(13,0,NUM_LEDS, 0 , 0);`
It should be :
`uint16_t pos = beatsin16(13,0,NUM_LEDS - 1, 0 , 0);`
Otherwise, the array calls for leds[NUM_LEDS] which crashes the array. Call only up to leds[NUM_LEDS -1].

focalintent, 2016-11-08:
This isn't a bug in the examples - it's a bug in beatsin16 that is fallout from a scaling fix we put in. @kriegsman can you confirm?

kriegsman, 2016-11-08:
Yep. Actual bug. I'll try to get a fix committed as soon as I can.

kriegsman, 2017-01-08:
I'm taking a look at this now, and actually, I'm not seeing any overflow or out-of-range results from beatsin16 or any of the related support functions.  beatsin16( BPM, low, high) is designed to return a result >= low and < high, which is what I'm seeing in my testing.

Were you actually seeing crashes / hangs from this, and if so, what value for NUM_LEDS?  Or were you mostly just concerned about how the code appeared to look in the example (but not seeing any actual crashes or hangs) ?

mikeburg, 2017-01-08:
I think I tracked it down.

scale16 on non-Arduino/AVR platforms implements the FASTLED_SCALE8_FIXED. On Arduino/AVR, the fix is *NOT* implemented in the inline assembly.

As a test example: 

On Teensy 3.1 (ARM), and Linux (AMD64):

scale16(65149, 64) = 64

On Arduino Uno:

scale16(65149, 64) = 63

---
I saw this problem when porting the DemoReel100 example to the Raspberry PI. The program was crashing due a variable being corrupted because he led array  was being over indexed by beatsin16().

kriegsman, 2017-01-09:
Ah thanks yes. I'll check it on a 32bit platform. Much appreciated.

commit 94e46e39 (2017-01-15): Updated scale16 and scale16by8 for AVR when FASTLED_SCALE8_FIXED is 1.  scale16(foo,65535) now equals foo.  Updated DemoReel100 to stay within range of NUM_LEDS on non-AVR platforms. Fixes #368.
closed 2017-01-15 by kriegsman