FastLED/FastLED #370: fill_gradient with CRGBArray no matching function

marmilicious, 2016-11-11:
When using CRGBArray, **fill_gradient_RGB** seems to work, but **fill_gradient** doesn't compile and gives the error:
no matching function for call to 'fill_gradient(CRGBArray<12>&, int, CHSV, int, CHSV)'

This works:
  fill_gradient_RGB(leds, 0, CRGB(255,0,0), NUM_LEDS-1, CRGB(0,255,0) );

This won't compile:
  fill_gradient(leds, 0, CHSV(96,255,255), NUM_LEDS-1, CHSV(150,255,255));

Here's my simple example sketch:
http://pastebin.com/jaHjFJzV

Testing with Arduino 1.6.11, FastLED 3.1.5, and an UNO

bradyjoh, 2016-12-25:
I too am having this issue.

AlainGourves, 2018-10-03:
I just found the solution looking at the "RGBSet Reference" documentation (RTFM ! ), where there is an example with `fill_rainbow()`. The correct syntax would be :

```
leds.fill_gradient(CHSV(64,255,255), CHSV(160,255,255));
leds(0,NUM_LEDS/2).fill_gradient(CHSV(64,255,255), CHSV(160,255,255));
```

closed 2019-01-20 by focalintent