FastLED/FastLED #371: MKR1000 with external power supply

knewson, 2016-11-18:
Hey guys, I'm having an issue with MKR1000. Seems to work on a WS2812 line that runs 5v when I power it via the board. but when it's a 12v board powered seperately with a common ground I get nothing. Same setup works fine with due

focalintent, 2016-11-18:
Please ask on the g+ community - http://fastled.io/+ - you're more likely to come across folks that can help with this setup/wiring issue.  (I'd like to keep the issues here to library specific issues).

closed 2016-11-18 by focalintent

knewson, 2016-11-18:
I don't disagree. but this isn't a setup or wiring issue. It's an issue with the library specific to the MKR1000. I had hoped to have specified that by pointing out that the setup / wiring is good and known to work with a similar board (due)

focalintent, 2016-11-19:
But you also said that it worked with this board and 5v leds.  

I'm taking an extended break from things because of a death in the family - there are also far more people watching the g+ group - which is a part of why i suggested posting there.